ros-controls/gazebo_ros2_control #6: [Meta-ticket] Port gazebo_ros2_control to ROS 2

ahcorde, 2020-06-03:
Not all functionality from ROS 1 has been ported to ROS 2 yet. 

> For ROS 1 see http://gazebosim.org/tutorials?tut=ros_control

This package provides a plugin which uses the ros2_control API to control joints inside Gazebo simulations.

The gazebo_ros_control <plugin> tag also has the following optional child elements:

 - **robotNamespace**: The ROS namespace to be used for this instance of the plugin, defaults to robot name in URDF/SDF
 - **controlPeriod**: The period of the controller update (in seconds), defaults to Gazebo's period
 - **robotParam**: The location of the robot_description (URDF) on the parameter server.
    - Parameters in ROS 2 are inside each node. By default the plugin check this parameter in a node called  `robot_state_publisher` and the parameter is  called `robot_description`
 - **robotSimType**: The pluginlib name of a custom robot sim interface to be used (see below for more details), defaults to 'DefaultRobotHWSim'

# TODO

 - [ ] Extend parameters:
   - Define the node to get the robotParam.  https://github.com/ros-simulation/gazebo_ros2_control/issues/4
 - Add demos for all ported functionality.
 
# Already added

 - ROS 2's style guidelines and linters.
 - Add demos for all ported functionality.

# Not implemented yet

 - [ ] PID https://github.com/ros-simulation/gazebo_ros2_control/issues/5

# Status

| Package            |                                  | Status |
| ------------------ | -------------------------------- | ------ |
| gazebo_ros_control |                                  |  |
|                    | default_robot_hw_sim             | :factory_worker: partially ported |
|                    | gazebo_ros_control_plugin        | :factory_worker:  partially ported |
|                    | robot_hw_sim                     | :factory_worker:  partially ported |

bmagyar, 2020-06-03:
Hi @ahcorde , here's a github project on the ros-controls org side collecting tickets for making ROS1 functionality available: https://github.com/orgs/ros-controls/projects/5

feel free to request new ones to be added

chapulina, 2020-06-03:
@bmagyar , if it would make it easier to transfer this repo to `ros-controls` so it is better integrated with the board, just let us know.

ahcorde, 2020-06-04:
- Define the node to get the robotParam https://github.com/ros-simulation/gazebo_ros2_control/pull/8

mkhansenbot, 2020-11-20:
I'm looking for a status, not clear to me what's left to be done for porting gazebo_ros2_controls? @chapulina or @ahcorde can you update?

chapulina, 2020-11-20:
@mkhansenbot , we've paused work on this task until the upstream `ros2_control` package is released into Foxy. Our understanding is that there are upcoming API changes to `ros2_control` that will require a refactoring of the code that we currently have in `gazebo_ros2_control`.

We've been tracking upstream's changes in the hope that providing simulation would speed up their development. However, the latest I heard, the `ros2_control` developers aren't making use of `gazebo_ros2_control`, so there isn't much benefit in us continuing to track their development so closely. Once `ros2_control` is stable (i.e. released), we'll restart work and update what's needed here.

mkhansenbot, 2020-11-20:
Thanks for the update, I'll check with the ros2_control maintainers

chapulina, 2021-02-03:
All the basic functionality is available since #44 . We're planning to make a bloom release into Foxy soon.

ahcorde, 2021-02-15:
Package was released and master branch build on top of Foxy packages thanks to this PR https://github.com/ros-simulation/gazebo_ros2_control/pull/44

closed 2021-02-15 by ahcorde