ros-controls/gazebo_ros2_control #33: About Gazebo crashing due to segmentation fault in libdefault_robot_hw_sim.so

Nyarn-WTF, 2020-08-25:
Hello.
Do you have a solution to the problem I had with ROS2 Foxy?

## Computer specs
- ThinkPad T14
- CPU: Ryzen5 4650U
- RAM: 16GB
- OS: Linux Mint
- kernel: 5.7.7-050707-generic

## ros2 log
```
[rviz2-1] qt5ct: using qt5ct plugin
[gazebo-2] Gazebo multi-robot simulator, version 11.0.0
[gazebo-2] Copyright (C) 2012 Open Source Robotics Foundation.
[gazebo-2] Released under the Apache 2 License.
[gazebo-2] http://gazebosim.org
[gazebo-2] 
[gazebo-2] Gazebo multi-robot simulator, version 11.0.0
[gazebo-2] Copyright (C) 2012 Open Source Robotics Foundation.
[gazebo-2] Released under the Apache 2 License.
[gazebo-2] http://gazebosim.org
[gazebo-2] 
[spawn_entity.py-3] [INFO] [1598339504.954850274] [urdf_spawner]: Spawn Entity started
[spawn_entity.py-3] [INFO] [1598339504.955676288] [urdf_spawner]: Loading entity XML from file /home/nyarn/Workspace/takao_x/install/takao_x_description/share/takao_x_description/urdf/takao_x.urdf
[spawn_entity.py-3] [INFO] [1598339504.960860964] [urdf_spawner]: Waiting for service /spawn_entity, timeout = 5
[spawn_entity.py-3] [INFO] [1598339504.961485346] [urdf_spawner]: Waiting for service /spawn_entity
[rviz2-1] [INFO] [1598339504.975179010] [rviz2]: Stereo is NOT SUPPORTED
[rviz2-1] [INFO] [1598339504.975658681] [rviz2]: OpenGl version: 4.6 (GLSL 4.6)
[rviz2-1] qt5ct: D-Bus global menu: no
[rviz2-1] [INFO] [1598339505.014769026] [rviz2]: Stereo is NOT SUPPORTED
[gazebo-2] [INFO] [1598339505.480245536] [gazebo_ros_node]: ROS was initialized without arguments.
[spawn_entity.py-3] [INFO] [1598339505.720582710] [urdf_spawner]: Calling service /spawn_entity
[spawn_entity.py-3] [INFO] [1598339506.710534881] [urdf_spawner]: Spawn status: SpawnEntity: Successfully spawned entity [base_link]
[INFO] [spawn_entity.py-3]: process has finished cleanly [pid 34955]
[gazebo-2] [INFO] [1598339506.838296451] [gazebo_ros2_control]: Loading gazebo_ros2_control plugin
[gazebo-2] [INFO] [1598339506.848054761] [takao_x.gazebo_ros2_control]: Starting gazebo_ros2_control plugin in namespace: /takao_x
[gazebo-2] [INFO] [1598339506.848379663] [takao_x.gazebo_ros2_control]: Starting gazebo_ros2_control plugin in ros 2 node: gazebo_ros2_control
[gazebo-2] [ERROR] [1598339506.858212214] [takao_x.gazebo_ros2_control]: connected to service!! /takao_x/robot_state_publisher
[gazebo-2] [ERROR] [1598339506.858283103] [takao_x.gazebo_ros2_control]: param_name robot_description
[gazebo-2] [ERROR] [1598339506.860119723] [takao_x.gazebo_ros2_control]: Recieved urdf from param server, parsing...
[gazebo-2] Parsing robot urdf xml string.
[gazebo-2] [Msg] Waiting for master.
[gazebo-2] [Msg] Connected to gazebo master @ http://127.0.0.1:11345
[gazebo-2] [Msg] Publicized address: 172.18.34.232
[gazebo-2] Loading joint 'right_wheel_joint' of type 'hardware_interface/VelocityJointInterface'
[gazebo-2] [INFO] [1598339506.890981589] [takao_x.gazebo_ros2_control]: joint right_wheel_joint is configured in VELOCITY mode
[gazebo-2] [Msg] Waiting for master.
[gazebo-2] [Msg] Connected to gazebo master @ http://127.0.0.1:11345
[gazebo-2] [Msg] Publicized address: 172.18.34.232
[gazebo-2] [Dbg] [GuiIface.cc:111] using qt5ct plugin
[gazebo-2] [Dbg] [GuiIface.cc:111] custom style sheet is disabled
[gazebo-2] [Dbg] [GuiIface.cc:111] D-Bus global menu: no
[ERROR] [gazebo-2]: process has died [pid 34953, exit code 255, cmd 'gazebo --verbose -s libgazebo_ros_factory.so'].
[gazebo-2] 
[gazebo-2] 

```

## syslog

```
Aug 25 16:11:46 nyarn kernel: [16718.163267] gzserver[35137]: segfault at 0 ip 00007f53880a108a sp 00007f5335fef2f0 error 4 in libdefault_robot_hw_sim.so[7f538808e000+35000]
Aug 25 16:11:46 nyarn kernel: [16718.163278] Code: e8 2b 5e ff ff 84 c0 74 07 c6 85 7f ff ff ff 01 48 8d 45 80 48 89 c7 e8 64 4f ff ff 48 05 c8 00 00 00 48 89 c7 e8 a6 8a ff ff <f2> 0f 10 00 f2 0f 11 45 b0 48 8d 45 80 48 89 c7 e8 41 4f ff ff 48
Aug 25 16:11:46 nyarn systemd[1]: Started Process Core Dump (PID 35222/UID 0).
Aug 25 16:11:49 nyarn systemd-coredump[35223]: Process 34989 (gzserver) of user 1000 dumped core.#012#012Stack trace of thread 35137:#012#0  0x00007f53880a108a n/a (/home/nyarn/Workspace/ros2_ws/build/gazebo_ros2_control/libdefault_robot_hw_sim.so + 0x6808a)#012#1  0x00007f538809ded9 n/a (/home/nyarn/Workspace/ros2_ws/build/gazebo_ros2_control/libdefault_robot_hw_sim.so + 0x64ed9)#012#2  0x00007f53883256e5 n/a (/home/nyarn/Workspace/ros2_ws/build/gazebo_ros2_control/libgazebo_ros2_control.so + 0x2456e5)#012#3  0x00007f545d36eda7 _ZN6gazebo7physics5Model10LoadPluginESt10shared_ptrIN3sdf2v97ElementEE (libgazebo_physics.so.11 + 0x269da7)#012#4  0x00007f545d370e10 _ZN6gazebo7physics5Model11LoadPluginsEv (libgazebo_physics.so.11 + 0x26be10)#012#5  0x00007f545d3c085a _ZN6gazebo7physics5World18ProcessFactoryMsgsEv (libgazebo_physics.so.11 + 0x2bb85a)#012#6  0x00007f545d3cb078 _ZN6gazebo7physics5World15ProcessMessagesEv (libgazebo_physics.so.11 + 0x2c6078)#012#7  0x00007f545d3cb5ee _ZN6gazebo7physics5World4StepEv (libgazebo_physics.so.11 + 0x2c65ee)#012#8  0x00007f545d3ce9b5 _ZN6gazebo7physics5World7RunLoopEv (libgazebo_physics.so.11 + 0x2c99b5)#012#9  0x00007f545db6ccb4 n/a (libstdc++.so.6 + 0xd6cb4)#012#10 0x00007f545d4c9609 start_thread (libpthread.so.0 + 0x9609)#012#11 0x00007f545d9ab103 __clone (libc.so.6 + 0x122103)#012#012Stack trace of thread 35063:#012#0  0x00007f545d4d0376 futex_wait_cancelable (libpthread.so.0 + 0x10376)#012#1  0x00007f543edfe62b n/a (radeonsi_dri.so + 0x47c62b)#012#2  0x00007f543edfe23b n/a (radeonsi_dri.so + 0x47c23b)#012#3  0x00007f545d4c9609 start_thread (libpthread.so.0 + 0x9609)#012#4  0x00007f545d9ab103 __clone (libc.so.6 + 0x122103)#012#012Stack trace of thread 35068:#012#0  0x00007f545d4d0376 futex_wait_cancelable (libpthread.so.0 + 0x10376)#012#1  0x00007f543edfe62b n/a (radeonsi_dri.so + 0x47c62b)#012#2  0x00007f543edfe23b n/a (radeonsi_dri.so + 0x47c23b)#012#3  0x00007f545d4c9609 start_thread (libpthread.so.0 + 0x9609)#012#4  0x00007f545d9ab103 __clone (libc.so.6 + 0x122103)#012#012Stack trace of thread 35023:#012#0  0x00007f545d969361 __GI___clock_nanosleep (libc.so.6 + 0xe0361)#012#1  0x00007f545df344fc _ZN6gazebo6common4Time5SleepERKS1_ (libgazebo_common.so.11 + 0xf64fc)#012#2  0x00007f545df3468c _ZN6gazebo6common4Time6MSleepEj (libgazebo_common.so.11 + 0xf668c)#012#3  0x00007f545e609185 _ZN6gazebo6Master3RunEv (libgazebo.so.11 + 0x48185)#012#4  0x00007f545db6ccb4 n/a (libstdc++.so.6 + 0xd6cb4)#012#5  0x00007f545d4c9609 start_thread (libpthread.so.0 + 0x9609)#012#6  0x00007f545d9ab103 __clone (libc.so.6 + 0x122103)#012#012Stack trace of thread 35022:#012#0  0x00007f545d9ab43e epoll_wait (libc.so.6 + 0x12243e)#012#1  0x00007f545d0a221b n/a (libgazebo_transport.so.11 + 0x4b21b)#012#2  0x00007f545d0a4611 n/a (libgazebo_transport.so.11 + 0x4d611)#012#3  0x00007f545d0bd4e2 n/a (libgazebo_transport.so.11 + 0x664e2)#012#4  0x00007f545c31443b n/a (libboost_thread.so.1.71.0 + 0x1143b)#012#5  0x00007f545d4c9609 start_thread (libpthread.so.0 + 0x9609)#012#6  0x00007f545d9ab103 __clone (libc.so.6 + 0x122103)#012#012Stack trace of thread 35059:#012#0  0x00007f545d4d0376 futex_wait_cancelable (libpthread.so.0 + 0x10376)#012#1  0x00007f543edfe62b n/a (radeonsi_dri.so + 0x47c62b)#012#2  0x00007f543edfe23b n/a (radeonsi_dri.so + 0x47c23b)#012#3  0x00007f545d4c9609 start_thread (libpthread.so.0 + 0x9609)#012#4  0x00007f545d9ab103 __clone (libc.so.6 + 0x122103)#012#012Stack trace of thread 35049:#012#0  0x00007f545d4d07b1 futex_abstimed_wait_cancelable (libpthread.so.0 + 0x107b1)#012#1  0x00007f545d0b557e _ZN6gazebo9transport17ConnectionManager3RunEv (libgazebo_transport.so.11 + 0x5e57e)#012#2  0x00007f545c31443b n/a (libboost_thread.so.1.71.0 + 0x1143b)#012#3  0x00007f545d4c9609 start_thread (libpthread.so.0 + 0x9609)#012#4  0x00007f545d9ab103 __clone (libc.so.6 + 0x122103)#012#012Stack trace of thread 35002:#012#0  0x00007f545d4d0376 futex_wait_cancelable (libpthread.so.0 + 0x10376)#012#1  0x00007f545deff7db n/a (libgazebo_common.so.11 + 0xc17db)#012#2  0x00007f545def5f52 _ZN6gazebo6common13ModelDatabase16UpdateModelCacheEb (libgazebo_common.so.11 + 0xb7f52)#012#3  0x00007f545c31443b n/a (libboost_thread.so.1.71.0 + 0x1143b)#012#4  0x00007f545d4c9609 start_thread (libpthread.so.0 + 0x9609)#012#5  0x00007f545d9ab103 __clone (libc.so.6 + 0x122103)#012#012Stack trace of thread 35057:#012#0  0x00007f545d4d0376 futex_wait_cancelable (libpthread.so.0 + 0x10376)#012#1  0x00007f543edfe62b n/a (radeonsi_dri.so + 0x47c62b)#012#2  0x00007f543edfe23b n/a (rad [... cut ...] 07f545d4c9609 start_thread (libpthread.so.0 + 0x9609)#012#6  0x00007f545d9ab103 __clone (libc.so.6 + 0x122103)#012#012Stack trace of thread 35060:#012#0  0x00007f545d4d0376 futex_wait_cancelable (libpthread.so.0 + 0x10376)#012#1  0x00007f543edfe62b n/a (radeonsi_dri.so + 0x47c62b)#012#2  0x00007f543edfe23b n/a (radeonsi_dri.so + 0x47c23b)#012#3  0x00007f545d4c9609 start_thread (libpthread.so.0 + 0x9609)#012#4 
Aug 25 16:11:49 nyarn systemd[1]: systemd-coredump@27-35222-0.service: Succeeded.

```

ahcorde, 2020-08-26:
This repository is still under development. There are two packages that are already a simple port from ROS (like transmission and joint_limit).

Anyhow, can you provide more details? 
 - maybe a repo? 
 - just share your configuration files?
 - backtrace from `gdb` ?

Nyarn-WTF, 2020-08-26:
I built gazebo_ros2_control by following the instructions here https://github.com/ros-simulation/gazebo_ros2_control/issues/3#issue-626344752.
Then I started gazebo_ros2_control from my URDF as follows.

## URDF (excerpt)
```
  <transmission name="right_wheel_trans">
    <type>transmission_interface/SimpleTransmission</type>
    <joint name="right_wheel_joint">
      <hardwareInterface>hardware_interface/VelocityJointInterface</hardwareInterface>
    </joint>
    <actuator name="right_wheel_motor">
      <hardwareInterface>hardware_interface/VelocityJointInterface</hardwareInterface>
    </actuator>
  </transmission>
  <transmission name="left_wheel_trans">
    <type>transmission_interface/SimpleTransmission</type>
    <joint name="left_wheel_joint">
      <hardwareInterface>hardware_interface/VelocityJointInterface</hardwareInterface>
    </joint>
    <actuator name="left_wheel_motor">
      <hardwareInterface>hardware_interface/VelocityJointInterface</hardwareInterface>
    </actuator>
  </transmission>
  <gazebo reference="base_link">
    <mu1>0.05</mu1>
    <mu2>0.05</mu2>
  </gazebo>
  <gazebo reference="right_wheel">
    <mu1>1.0</mu1>
    <mu2>1.0</mu2>
  </gazebo>
  <gazebo reference="left_wheel">
    <mu1>1.0</mu1>
    <mu2>1.0</mu2>
  </gazebo>
  <gazebo reference="caster_link">
    <mu1>0.0</mu1>
    <mu2>0.0</mu2>
  </gazebo>
  <gazebo>
    <plugin filename="libgazebo_ros2_control.so" name="gazebo_ros2_control">
      <namespace>takao_x</namespace>
      <robot_sim_type>gazebo_ros2_control/DefaultRobotHWSim</robot_sim_type>
      <robot_param>robot_description</robot_param>
      <robot_param_node>/takao_x/robot_state_publisher</robot_param_node>
      <parameters>/home/nyarn/Workspace/takao_x/install/takao_x_description/share/takao_x_description/config/controller.yaml</parameters>
    </plugin>
  </gazebo>
```

## controller.yaml
```
takao_x:
    joint_state_controller:
        ros__parameters:
            type: joint_state_controller/JointStateController
                publish_rate: 50
    base_controller:
        ros__parameters:
            type        : "diff_drive_controller/DiffDriveController"
            left_wheel  : 'left_wheel_joint'
            right_wheel : 'right_wheel_joint'
            publish_rate: 50.0               # default: 50
            pose_covariance_diagonal : [0.001, 0.001, 1000000.0, 1000000.0, 1000000.0, 1000.0]
            twist_covariance_diagonal: [0.001, 0.001, 1000000.0, 1000000.0, 1000000.0, 1000.0]
            # Wheel separation and diameter. These are both optional.
            # diff_drive_controller will attempt to read either one or both from the
            # URDF if not specified as a parameter
            wheel_separation : 0.55
            wheel_radius : 0.1
            # Wheel separation and radius multipliers
            wheel_separation_multiplier: 1.0 # default: 1.0
            wheel_radius_multiplier    : 1.0 # default: 1.0
            # Velocity commands timeout [s], default 0.5
            cmd_vel_timeout: 0.25
            # Base frame_id
            base_frame_id: base_footprint #default: base_link
            # Velocity and acceleration limits
            # Whenever a min_* is unspecified, default to -max_*
            linear:
              x:
                has_velocity_limits    : true
                max_velocity           : 1.0  # m/s
                min_velocity           : -0.5 # m/s
                has_acceleration_limits: true
                max_acceleration       : 0.8  # m/s^2
                min_acceleration       : -0.4 # m/s^2
                has_jerk_limits        : true
                max_jerk               : 5.0  # m/s^3
            angular:
              z:
                has_velocity_limits    : true
                max_velocity           : 1.7  # rad/s
                has_acceleration_limits: true
                max_acceleration       : 1.5  # rad/s^2
                has_jerk_limits        : true
                max_jerk               : 2.5  # rad/s^3
```

Nyarn-WTF, 2020-08-26:
And this is the output of GDB.

## GDB
```
Thread 76 "gzserver" received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0x7ffed6ffd700 (LWP 21346)]
gazebo_ros2_control::DefaultRobotHWSim::registerJointLimits (this=0x7ffea10c4130, joint_name="right_wheel_joint",
    joint_handle=..., ctrl_method=gazebo_ros2_control::DefaultRobotHWSim::VELOCITY, joint_limit_nh=
    std::shared_ptr<rclcpp::Node> (use count 2, weak count 1) = {...}, urdf_model=0x7ffed6ffa310,
    joint_type=0x7ffea10c5c00, lower_limit=0x7ffea10be070, upper_limit=0x7ffea10d9700, effort_limit=0x7ffea10d8f00,
    vel_limit=0x7ffea10d89d0)
    at /home/nyarn/Workspace/ros2_ws/src/gazebo_ros2_control/gazebo_ros2_control/src/default_robot_hw_sim.cpp:519
519	      limits.min_position = urdf_joint->limits->lower;
```

Nyarn-WTF, 2020-09-01:
In addition, I looked at the values of the variables. As a result, urdf_joint->limits->lower was set to 0x0. I think it crashed because it couldn't refer to the variable.

```
(gdb) p urdf_joint->limits->lower
Cannot access memory at address 0x0
(gdb) p urdf_joint->limits
$1 = std::shared_ptr<urdf::JointLimits> (empty) = {get() = 0x0}
(gdb) p urdf_joint
$2 = std::shared_ptr<const urdf::Joint> (use count 4, weak count 0) = {get() = 0x7ffe9ce5d810}
```

Nyarn-WTF, 2020-09-01:
Finally, the cause of the bug was that we had written the "limit" in the wrong place.
Thank you.

closed 2020-09-01 by Nyarn-WTF