ros-controls/gazebo_ros2_control #39: Can't build on Foxy distro

kong2983, 2020-10-27:
I tried to build this package. However, it is impossible to build because the package compatibility is not appropriate. If you try to build by cloning the latest version of the ros2_control package, you cannot build it because there is no joint_state_handle package. 
Is there any way to solve this problem?

ahcorde, 2020-10-27:
Hi @kong2983,

Some new changes are happening right now in `ros2_control`. 

For now you can compile the current `gazebo_ros2_control`, `ros2_control` and `ros2_controller` master branch with the branch of this PR https://github.com/ros-simulation/gazebo_ros2_control/pull/34#issuecomment-716975698 (`ahcorde/update/ros2_control`).

Let me know if you have some issue compiling this branch.

kong2983, 2020-11-02:
I have confirmed that the example works normally.
Thank you for the reply!

closed 2020-11-02 by kong2983

heike380, 2020-11-02:
Hi @kong2983, 

how did you build the "ahcorde/update/ros2_control"-branch? 
I built the "ahcorde/update/ros2_control" with docker (docker build -t gazebo_ros2_control .), but still got the error that "JointStateHandle" wasn't found. 

Thanks.

kong2983, 2020-11-02:
@heike380 

The "ahcorde/update/ros2_control"-branch does not use JointStateHandle.
It seems that the master branch is installed while building with docker.
I didn't build with docker. Try again with the command below.

git clone -b ahcorde/update/ros2_control --single-branch https://github.com/ros-simulation/gazebo_ros2_control
git clone https://github.com/ros-controls/ros2_control
git clone https://github.com/ros-controls/ros2_controllers

heike380, 2020-11-02:
@kong2983 

Thanks! Perfect, now it builds.

I just adapted "git clone https://github.com/ros-simulation/gazebo_ros2_control" to "git clone -b ahcorde/update/ros2_control --single-branch https://github.com/ros-simulation/gazebo_ros2_control" in the Dockerfile.