ros-controls/gazebo_ros2_control #42: rosdep dependency problems when trying to install

farbod-new, 2020-12-18:
Hello,

I am trying to follow the directions from the "docker file" in the following link:

https://github.com/ros-simulation/gazebo_ros2_control/pull/34/files

When I use the following line to install ros dependencies:

rosdep install --from-paths ./ -i -y --rosdistro foxy --ignore-src

I get the following:

ERROR: the following packages/stacks could not have their rosdep keys
resolved
to system dependencies:
gazebo_ros2_control: Cannot locate rosdep definition for
[joint_limits_interface]
gazebo_ros2_control_demos: Cannot locate rosdep definition for
[velocity_controllers]
ros2_control: Cannot locate rosdep definition for
[controller_parameter_server]

So, I can't proceed correctly. If I do proceed with the build, I get

Starting >>> hardware_interface
Starting >>> controller_manager_msgs
Starting >>> control_toolbox
Starting >>> transmission_interface
Finished <<< transmission_interface [0.23s]
Finished <<< hardware_interface [0.31s]
Starting >>> controller_interface
Starting >>> test_robot_hardware
Finished <<< control_toolbox [0.31s]
Finished <<< controller_interface [0.21s]
Finished <<< test_robot_hardware [0.23s]
Finished <<< controller_manager_msgs [0.72s]
Starting >>> controller_manager
Finished <<< controller_manager [0.35s]
Starting >>> gazebo_ros2_control
Starting >>> joint_state_controller
Starting >>> joint_trajectory_controller
Finished <<< joint_state_controller [0.27s]
Finished <<< joint_trajectory_controller [0.31s]
--- stderr: gazebo_ros2_control
CMake Error at CMakeLists.txt:11 (find_package):
   By not providing "Findjoint_limits_interface.cmake" in CMAKE_MODULE_PATH
   this project has asked CMake to find a package configuration file
provided
   by "joint_limits_interface", but CMake did not find one.

   Could not find a package configuration file provided by
   "joint_limits_interface" with any of the following names:

     joint_limits_interfaceConfig.cmake
     joint_limits_interface-config.cmake

   Add the installation prefix of "joint_limits_interface" to
   CMAKE_PREFIX_PATH or set "joint_limits_interface_DIR" to a directory
   containing one of the above files.  If "joint_limits_interface"
provides a
   separate development package or SDK, be sure it has been installed.

---
Failed   <<< gazebo_ros2_control [1.99s, exited with code 1]

Summary: 9 packages finished [3.20s]
   1 package failed: gazebo_ros2_control
   1 package had stderr output: gazebo_ros2_control
   1 package not processed

Any help is appreciated.

Thanks so much,

Farbod

ahcorde, 2020-12-18:
Hi @farbod-new, 

there are some packages that are not currently available in rosdep for `foxy` or `rolling`. `ros2_control` is under heavy development, you may need to install these packages from sources.

farbod-new, 2020-12-18:
> Hi @farbod-new,
> 
> there are some packages that are not currently available in rosdep for `foxy` or `rolling`. `ros2_control` is under heavy development, you may need to install these packages from sources.

Thanks for the reply.

As I understand it joint_limits_interface is part of ros2_control (or even ros_control) package and is not an independent package. In other words, how can one install joint_limits_interface from source separately when its source is already included in the ros2_control package?

In the line:
cp -r ros2_control_ddengster/joint_limits_interface ros2_control_ddengster/transmission_interface  ros2_control

joint_limits_interface and transmission_interface are copied from another source to ros2_control (ros2_control already includes those folders). While building the package transmission_interface is built with no problem, but joint_limits_interface fails. Why is that?

Thanks,
Farbod

farbod-new, 2020-12-21:
> Thanks for the reply.
>
> As I understand it joint_limits_interface is part of ros2_control (or even ros_control) package and is not an independent package. In other words, how can one install [...]

Hi everyone,

I am still having trouble with the joint_limits_interface dependency, although the code is included in the ros2_control package. Do I need a ROS1-Bridge to solve this dependency? Or do I have to compile those dependencies from source with ros2? How do I make this compilation work? Do I need ROS1-Bridge?

Any help is appreciated.

Thanks,
Farbod

ahcorde, 2020-12-22:
I checked the ros2_control repository. It seems that right now `joint_limit_interface` it's not available https://github.com/ros-controls/ros2_control/tree/master/joint_limits_interface. It has a `COLCON_IGNORE` inside the folder which means this package will not compile. 

I will have a look as soon as possible

farbod-new, 2020-12-22:
> I checked the ros2_control repository. It seems that right now `joint_limit_interface` it's not available https://github.com/ros-controls/ros2_control/tree/master/joint_limits_interface. It has a [...]

Thank you so much!

An extra information that might help is that I successfully built the following repository (with no errors):

https://github.com/ros-controls/ros2_control

farbod@SavantzDev:~/ros2_control_ws$ colcon build
Starting >>> control_msgs
Starting >>> controller_manager_msgs
Starting >>> realtime_tools
Starting >>> angles
Finished <<< angles [0.18s]                                                                                                                              
Finished <<< realtime_tools [0.29s]                                                                                                          
Finished <<< controller_manager_msgs [0.81s]                                                                   
Starting >>> ros2controlcli
Finished <<< control_msgs [1.04s]                                                       
Starting >>> hardware_interface
Finished <<< hardware_interface [0.28s]                                                          
Starting >>> controller_interface
Starting >>> test_robot_hardware
Starting >>> transmission_interface
Finished <<< ros2controlcli [0.54s]
Finished <<< test_robot_hardware [0.23s]                                                                                                                
Finished <<< controller_interface [0.25s]
Starting >>> controller_manager
Finished <<< transmission_interface [0.30s]                                                           
Finished <<< controller_manager [0.37s]                      
Starting >>> forward_command_controller
Starting >>> joint_state_controller
Starting >>> joint_trajectory_controller
Starting >>> ros2_control
Finished <<< ros2_control [0.17s]                                                                                              
Finished <<< joint_state_controller [0.30s]                                                                                
Finished <<< forward_command_controller [0.34s]
Finished <<< joint_trajectory_controller [0.35s]                    

Summary: 14 packages finished [2.48s]

farbod-new, 2020-12-23:
Finally, I was able to build "joint_limits_interface". But now, a new error comes up in "default_robot_hw_sim.hpp" 

Please see below. Was there any corrections to this file, or the missing ".hpp"?

Thanks,
Farbod

farbod@SavantzDev:~/ros2_ws$ colcon build
Starting >>> hardware_interface
Starting >>> controller_manager_msgs
Finished <<< hardware_interface [0.34s]                                                                             
Starting >>> controller_interface
Starting >>> joint_limits_interface
Starting >>> transmission_interface
Starting >>> test_robot_hardware
Finished <<< joint_limits_interface [0.19s]                                                                             
Finished <<< controller_interface [0.20s]
Finished <<< test_robot_hardware [0.29s]                                                                             
Finished <<< controller_manager_msgs [0.94s]                                                                     
Starting >>> controller_manager
Starting >>> ros2controlcli
Finished <<< controller_manager [0.37s]                                                                                                
Starting >>> joint_state_controller
Starting >>> joint_trajectory_controller
Starting >>> forward_command_controller
Finished <<< ros2controlcli [0.54s]                                                                                                       
Starting >>> ros2_control
Finished <<< joint_state_controller [0.28s]                                                                            
Finished <<< forward_command_controller [0.28s]                                                                          
Finished <<< joint_trajectory_controller [0.31s]
Finished <<< ros2_control [0.19s]
Finished <<< transmission_interface [6.02s]                       
Starting >>> gazebo_ros2_control
--- stderr: gazebo_ros2_control                                
In file included from /home/farbod/ros2_ws/src/gazebo_ros2_control/gazebo_ros2_control/include/gazebo_ros2_control/default_robot_hw_sim.hpp:46,
                 from /home/farbod/ros2_ws/src/gazebo_ros2_control/gazebo_ros2_control/src/default_robot_hw_sim.cpp:39:
/home/farbod/ros2_ws/install/joint_limits_interface/include/joint_limits_interface/joint_limits_interface.hpp:41:10: fatal error: hardware_interface/joint_state_handle.hpp: No such file or directory
   41 | #include <hardware_interface/joint_state_handle.hpp>
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [CMakeFiles/default_robot_hw_sim.dir/build.make:63: CMakeFiles/default_robot_hw_sim.dir/src/default_robot_hw_sim.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:80: CMakeFiles/default_robot_hw_sim.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
In file included from /home/farbod/ros2_ws/src/gazebo_ros2_control/gazebo_ros2_control/include/gazebo_ros2_control/gazebo_ros2_control_plugin.hpp:53,
                 from /home/farbod/ros2_ws/src/gazebo_ros2_control/gazebo_ros2_control/src/gazebo_ros2_control_plugin.cpp:41:
/home/farbod/ros2_ws/src/gazebo_ros2_control/gazebo_ros2_control/include/gazebo_ros2_control/robot_hw_sim.hpp:42:10: fatal error: hardware_interface/robot_hardware.hpp: No such file or directory
   42 | #include "hardware_interface/robot_hardware.hpp"
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [CMakeFiles/gazebo_ros2_control.dir/build.make:63: CMakeFiles/gazebo_ros2_control.dir/src/gazebo_ros2_control_plugin.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:107: CMakeFiles/gazebo_ros2_control.dir/all] Error 2
make: *** [Makefile:141: all] Error 2
---
Failed   <<< gazebo_ros2_control [3.75s, exited with code 2]
                                   
Summary: 12 packages finished [10.3s]
  1 package failed: gazebo_ros2_control
  1 package had stderr output: gazebo_ros2_control
  1 package not processed

farbod-new, 2020-12-23:
Also, I want to mention that I used the original folder for "ros2_control_transmission_interface" from the original ros2_control repository, rather than "ros2_control_ddengster/transmission_interface" The former did not build.

But I did use "ros2_control_ddengster/joint_limits_interface" rather than "ros2_control/joint_limits_interface".

I hope this makes sense.

ar0usel, 2020-12-30:
Hi @farbod-new,

Do you manage to build it?

ahcorde, 2020-12-30:
Hi @ar0usel ,

Right now, there are some changes that make imposible to compile the code. In particular this issue https://github.com/ros-controls/ros2_control/issues/279. There is some effort trying to revive this code https://github.com/ros-controls/ros2_control/pull/271

I will update the code when this issue is resolved.

farbod-new, 2020-12-31:
Thanks for letting us know. I thought I was doing something wrong.

> Hi @ar0usel ,
> 
> Right now, there are some changes that make imposible to compile the code. In particular this issue [ros-controls/ros2_control#279](https://github.com/ros-controls/ros2_control/issues/279). There is some effort trying to revive this code [ros-controls/ros2_control#271](https://github.com/ros-controls/ros2_control/pull/271)
> 
> I will update the code when this issue is resolved.

ahcorde, 2021-01-11:
I have created this new PR https://github.com/ros-simulation/gazebo_ros2_control/pull/44. This should fix the issues.

ahcorde, 2021-02-15:
Building this package with the Foxy deb packages should be fine.

closed 2021-02-15 by ahcorde