ros-controls/gazebo_ros2_control #43: hardware_interface/EffortJointInterface hasn't been implemented yet?

JaehyunShim, 2021-01-05:
Hi, 

As the question says, hardware_interface/EffortJointInterface seems to not work yet. Changing it to hardware_interface/PositionJointInterface makes everything work but that is not what I am trying to implement in my software. Any advice? 

Thank you very much,
Jaehyun

bmagyar, 2021-01-05:
Which version of ros2_control are you trying to use for this? The one from the master branch? There's currently no working version of `gazebo_ros2_control` with the Foxy released version. The code you can find in this repo is an early attempt that is now very much out of date.

Drojas251, 2021-01-05:
@bmagyar do you know which ros2 version is this repo working with?

JaehyunShim, 2021-01-06:
@bmagyar As below.

```sh
$ git clone https://github.com/ros-controls/ros2_control.git
$ git clone https://github.com/ros-controls/ros2_controllers.git
$ git clone https://github.com/ros-simulation/gazebo_ros2_control.git
$ cd ros2_control && git reset --hard 3dc62e28e3bc8cf636275825526c11d13b554bb6 && cd ..
$ cd ros2_controllers && git reset --hard 83c494f460f1c8675f4fdd6fb8707b87e81cb197 && cd ..
$ cd gazebo_ros2_control && git reset --hard 3dfe04d412d5be4540752e9c1165ccf25d7c51fb && cd ..
```
The reason seems to be that hardware_interface/EffortJointInterface is not compatible with joint_state_controller/JointStateController, whereas PositionJointInterface and VelocityJointInterface are. 

@Drojas251 
I am testing with ROS2 Foxy.

JaehyunShim, 2021-01-06:
Hi @ahcorde

Any advice? I see sample codes added by you for the position and velocity controls in the demo package but not for the effort control. 

Thank you in advance, 
Jaehyun

bmagyar, 2021-01-06:
> @bmagyar do you know which ros2 version is this repo working with?

Excellent question. The plugin here is in a proof of concept stage, it worked at some point, my best guess would be to try matching the time of this repo to some checkout of ros2_control and ros2_controllers but it's very manual and not a way forward as we did major changes since then so whatever you do on top, it will break as soon as you try using a released version.

JaehyunShim, 2021-01-15:
ping @ahcorde

ahcorde, 2021-01-15:
The effortJointInterface is already coded but I didn't add any test. I will try to add one today

JaehyunShim, 2021-01-18:
@ahcorde 
Thank you very much. I will test it and get back to you soon.

JaehyunShim, 2021-01-18:
Thank you. It works!

closed 2021-01-18 by JaehyunShim