ros-controls/gazebo_ros2_control #47: Plugin won't load with Moveit sourced

swilcock0, 2021-01-13:
I've been trying to get this library working when I also have Moveit sourced in Foxy. If I run

```bash
# Download repos
mkdir ~/control_ws/src -p
cd ~/control_ws/src
git clone https://github.com/ros-simulation/gazebo_ros2_control 
git clone https://github.com/ros-controls/ros2_control 
git clone https://github.com/ros-controls/ros2_controllers

# Switch GR2C to latest PR
cd gazebo_ros2_control
git fetch origin pull/44/head
git checkout FETCH_HEAD
cd ../..

# Build repos
. /opt/ros/foxy/setup.bash
colcon build --merge-install --symlink-install
. install/setup.bash

# Run demo
ros2 launch gazebo_ros2_control_demos cart_example_position.launch.py

# Seperate terminal
ros2 run gazebo_ros2_control_demos example_position 
```

the demo runs great. However, once I source my moveit workspace that I'm currently using for motion planning, the plugin can no longer be imported 
```bash
. ~/moveit_ws/install/setup.bash

ros2 launch gazebo_ros2_control_demos cart_example_position.launch.py &
ros2 run gazebo_ros2_control_demos example_position 
```

gives 
```console
node created
terminate called after throwing an instance of 'std::runtime_error'
  what():  could not get action server
```

On further inspection, trying it with Gazebo set to verbose output, it says that it is failing to load the plugin:
```
[spawn_entity.py-2] [INFO] [1610565015.851695000] [spawn_entity]: Spawn status: SpawnEntity: Successfully spawned entity [cartpole]
[Err] [Plugin.hh:178] Failed to load plugin libgazebo_ros2_control.so: /home/ros/control_ws/install/lib/libgazebo_ros2_control.so: undefined symbol: _ZN18controller_manager17ControllerManager17switch_controllerERKSt6vectorINSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEESaIS7_EESB_ibRKN6rclcpp8DurationE
```

Another error I had earlier (which I'm struggling to recreate now) returned an error in the plugin Load function.

My assumption is that the historic versions of ros2_control and ros2_controllers specified in Moveit's version control are the issue. As I see you've been testing with Moveit @ahcorde, I'd be interested to know how you've been getting them to play nicely!

ahcorde, 2021-01-15:
Hi @swilcock0,

The [moveit2.repos](https://raw.githubusercontent.com/ros-planning/moveit2/main/moveit2.repos) file install a specific version of `ros2_control` (old one). You should have another one in the `gazebo_ros2_control` workspace. You have two version of the same library and pluginlib is trying to load the wrong one.

I have updated the [Dockerfile](https://github.com/ros-simulation/gazebo_ros2_control/pull/44/commits/90a0b6f3fa39c8996c8173a9a3cacdaf916cd354). Let me know if you can reproduce it.

swilcock0, 2021-01-15:
Thanks for the reply, that's working great now, and building moveit with the latest ros2_control and controllers has fixed it for my application

closed 2021-01-15 by swilcock0