ros-controls/gazebo_ros2_control #49: Add ros2_controllers as exec_dependency in next Foxy synchronization

ahcorde, 2021-02-03:
Reference to this comment https://github.com/ros-simulation/gazebo_ros2_control/pull/44#discussion_r569765040

Add `effort_controller`, `velocity_controller` and `joint trajectory controller` as `exec_dependency`

bmagyar, 2021-02-04:
While this may sound intuitive, I think we may run into a bunch of packaging and dependency issues. The plugin itself shouldn't even have launch files, a much more maintainable setup would be to have a `gazebo_ros2_control_demos` package where you can shove all scripts and asset files + runtime dependencies.

Karsten1987, 2021-02-12:
I honestly think this should not be an enhancement rather than an actual bug. I just ran into this in a freshly installed ubuntu and the launch file crashes pretty badly without any hints:
```
~/workspace/ros2/ros2_control_ws $ ros2 launch gazebo_ros2_control_demos cart_example_position.launch.py
[INFO] [launch]: All log files can be found below /home/knk1pal/.ros/log/2021-02-11-19-24-40-095406-knk1pal-desktop-94280
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [gzserver-1]: process started with pid [94282]
[INFO] [gzclient   -2]: process started with pid [94284]
[INFO] [robot_state_publisher-3]: process started with pid [94287]
[INFO] [spawn_entity.py-4]: process started with pid [94289]
[robot_state_publisher-3] Parsing robot urdf xml string.
[robot_state_publisher-3] Link slideBar had 1 children
[robot_state_publisher-3] Link cart had 0 children
[robot_state_publisher-3] [INFO] [1613100280.531889377] [robot_state_publisher]: got segment cart
[robot_state_publisher-3] [INFO] [1613100280.531991254] [robot_state_publisher]: got segment slideBar
[robot_state_publisher-3] [INFO] [1613100280.532006048] [robot_state_publisher]: got segment world
[spawn_entity.py-4] [INFO] [1613100280.884423657] [spawn_entity]: Spawn Entity started
[spawn_entity.py-4] [INFO] [1613100280.884755934] [spawn_entity]: Loading entity published on topic robot_description
[spawn_entity.py-4] [INFO] [1613100280.885949316] [spawn_entity]: Waiting for entity xml on robot_description
[spawn_entity.py-4] [INFO] [1613100280.991258736] [spawn_entity]: Waiting for service /spawn_entity, timeout = 5
[spawn_entity.py-4] [INFO] [1613100280.992204321] [spawn_entity]: Waiting for service /spawn_entity
[spawn_entity.py-4] [INFO] [1613100281.749584266] [spawn_entity]: Calling service /spawn_entity
[spawn_entity.py-4] [INFO] [1613100281.885642581] [spawn_entity]: Spawn status: SpawnEntity: Successfully spawned entity [cartpole]
[gzserver-1] [INFO] [1613100281.907895669] [gazebo_ros2_control]: Loading gazebo_ros2_control plugin
[gzserver-1] [INFO] [1613100281.913425389] [gazebo_ros2_control]: Starting gazebo_ros2_control plugin in namespace: /
[gzserver-1] [INFO] [1613100281.913928641] [gazebo_ros2_control]: Starting gazebo_ros2_control plugin in ros 2 node: gazebo_ros2_control
[gzserver-1] [INFO] [1613100281.920382961] [gazebo_ros2_control]: connected to service!! robot_state_publisher
[gzserver-1] [INFO] [1613100281.921563856] [gazebo_ros2_control]: Recieved urdf from param server, parsing...
[INFO] [spawn_entity.py-4]: process has finished cleanly [pid 94289]
[gzserver-1] [INFO] [1613100281.965261873] [gazebo_ros2_control]: Loading joint: slider_to_cart
[gzserver-1] [INFO] [1613100281.965319936] [gazebo_ros2_control]: 	Command:
[gzserver-1] [INFO] [1613100281.965382091] [gazebo_ros2_control]: 		 position
[gzserver-1] [INFO] [1613100281.966137730] [gazebo_ros2_control]: 	State:
[gzserver-1] [INFO] [1613100281.966167436] [gazebo_ros2_control]: 		 position
[gzserver-1] [INFO] [1613100281.966763995] [gazebo_ros2_control]: 		 velocity
[gzserver-1] [INFO] [1613100281.966799265] [gazebo_ros2_control]: 		 effort
[gzserver-1] [INFO] [1613100281.968665842] [gazebo_ros2_control]: Loading controller_manager
[gzserver-1] [WARN] [1613100281.980678942] [gazebo_ros2_control]:  Desired controller update period (0.01 s) is slower than the gazebo simulation period (0.001 s).
[gzserver-1] [INFO] [1613100281.980875189] [controller_manager]: Loading controller 'joint_trajectory_controller'
[gzserver-1] [ERROR] [1613100281.980908832] [controller_manager]: Available classes:
[gzserver-1] [ERROR] [1613100281.980920531] [controller_manager]: controller_manager/test_controller
[gzserver-1] [ERROR] [1613100281.980927902] [controller_manager]: Loader for controller 'joint_trajectory_controller' not found.
[gzserver-1] terminate called without an active exception
[gzserver-1] Aborted (core dumped)
[ERROR] [gzserver-1]: process has died [pid 94282, exit code 134, cmd 'gzserver                                                                    -s libgazebo_ros_init.so   -s libgazebo_ros_factory.so       '].
[INFO] [gzclient   -2]: process has finished cleanly [pid 94284]
^C[WARNING] [launch]: user interrupted with ctrl-c (SIGINT)
[robot_state_publisher-3] [INFO] [1613100291.621538047] [rclcpp]: signal_handler(signal_value=2)
[INFO] [robot_state_publisher-3]: process has finished cleanly [pid 94287]
```

ahcorde, 2021-02-15:
AS I mentioned in the PR `Holding this PR to the next Foxy sync`

PR #56 (2021-02-19): add ros2_controllers as exec dependency
fixes #49

once merged, I believe this needs a re-release
commit b4236ef8 (2021-02-19): add ros2_controllers as exec dependency (#56)

fixes #49
closed 2021-02-19 by ahcorde