ros-controls/gazebo_ros2_control #57: Unnecessary parameter in demo

JaehyunShim, 2021-02-15:
As the title says, the below lines should be removed. (it does not cause any problem now since they are removed afterward by [this line]( https://github.com/ros-controls/ros2_controllers/blob/master/velocity_controllers/src/joint_group_velocity_controller.cpp#L44))

https://github.com/ros-simulation/gazebo_ros2_control/blob/master/gazebo_ros2_control_demos/config/cartpole_controller_effort.yaml#L15

https://github.com/ros-simulation/gazebo_ros2_control/blob/master/gazebo_ros2_control_demos/config/cartpole_controller_velocity.yaml#L15

Karsten1987, 2021-02-15:
I had a similar comment for the ros2_control demos. might be a copy-paste error?
cc @destogl

destogl, 2021-03-01:
Yes, this is definitely a copy-paste error. @JaehyunShim would you open a PR?

ahcorde, 2021-04-08:
PR https://github.com/ros-simulation/gazebo_ros2_control/pull/68

JaehyunShim, 2021-04-09:
@ahcorde  Thanks!

closed 2021-04-19 by ahcorde