ros-controls/gazebo_ros2_control #63: rocker instruction does not work

JaehyunShim, 2021-03-02:
Hi, 

[This rocker instruction](https://github.com/ros-simulation/gazebo_ros2_control#using-rocker) does not work. The below line is the error message from the terminal. 

```sh
Failed to build detector image
WARNING unable to detect os for base image 'gazebo_ros2_control:latest', maybe the base image does not exist
```

Regards,
Jaehyun

closed 2021-03-02 by JaehyunShim

ahcorde, 2021-03-02:
Probably you have discovered the solution, but you should compile your own docker file

JaehyunShim, 2021-03-02:
@ahcorde 

Yeah, that was the problem and thank you for your reply. I will reopen this thread if I have other relevant questions. 

Jaehyun

alimhn, 2024-02-16:
i have the same problem:

no more output and success not detected
Failed to build detector image
WARNING unable to detect os for base image 'zenoh-carla-bridge', maybe the base image does not exist.