ros-controls/gazebo_ros2_control #66: Can't lower the controller manager update rate

jordan-palacios, 2021-03-17:
As it is right now it's not possible to lower the controller manager update rate without affecting the simulation. This is because the frequency at which the positions are written to gazebo is coupled with the controller manager update rate.

You can easily reproduce the issue by lowering the rate to 100hz in the `rrbot_moveit_demo_nodes`. You'll see the arm trembling: position is set, arm falls for 10ms (instead of 1ms), position is set, etc.

Given that not all controllers need to or can't simply run at 1Khz this a problem.

One simple fix is to make the `write()` independent from the controller manager update rate. Like this:

```C++
// Called by the world update start event
void GazeboRosControlPrivate::Update()
{
  // Get the simulation time and period
  gazebo::common::Time gz_time_now = parent_model_->GetWorld()->SimTime();
  rclcpp::Time sim_time_ros(gz_time_now.sec, gz_time_now.nsec);
  rclcpp::Duration sim_period = sim_time_ros - last_update_sim_time_ros_;

  if (sim_period >= control_period_) {
    last_update_sim_time_ros_ = sim_time_ros;
    controller_manager_->read();
    controller_manager_->update();
  }

  // always do this
  controller_manager_->write();
}
```

Though I don't know what other possible implications this could have.

arjun-kg, 2021-03-30:
I'm trying to use gazebo control for universal robots, 1KHz (controller_manager update rate) does seem too much for the robot to handle (the joints slowly break apart), while at 100Hz or so, I observe the trembling you describe. The fix you describe updates positions at the rate of Gazebo's simulation period (0.001s or 1KHz), which is again, the original scenario where the joints break apart. I'm not sure why this is happening

v-lopez, 2021-05-28:
@ahcorde Can we get someone to look into this? It's preventing us from releasing robots to foxy/galactic due to the trembling.

We can provide a PR with the changes above if you determine it's the correct approach.

v-lopez, 2021-05-28:
> I'm trying to use gazebo control for universal robots, 1KHz (controller_manager update rate) does seem too much for the robot to handle (the joints slowly break apart), while at 100Hz or so, I [...]

@arjun-kg Did you find the cause? Are you sending different commands on each write?

closed 2021-05-28 by ahcorde