ros-controls/gazebo_ros2_control #71: CMake Error at CMakeLists.txt:6 (find_package):   "controller_manager"

newcanopies, 2021-04-27:
Hi,

I am able to build `/gazebo_ros2_control_demos`
but not /gazebo_ros2_control 
```
Starting >>> gazebo_ros2_control
--- stderr: gazebo_ros2_control
CMake Error at CMakeLists.txt:6 (find_package):
  By not providing "Findcontroller_manager.cmake" in CMAKE_MODULE_PATH this
  project has asked CMake to find a package configuration file provided by
  "controller_manager", but CMake did not find one.

  Could not find a package configuration file provided by
  "controller_manager" with any of the following names:

    controller_managerConfig.cmake
    controller_manager-config.cmake

  Add the installation prefix of "controller_manager" to CMAKE_PREFIX_PATH or
  set "controller_manager_DIR" to a directory containing one of the above
  files.  If "controller_manager" provides a separate development package or
  SDK, be sure it has been installed.

---
Failed   <<< gazebo_ros2_control [1.77s, exited with code 1]

Summary: 0 packages finished [1.87s]
  1 package failed: gazebo_ros2_control
  1 package had stderr output: gazebo_ros2_control
```

any ideas?

ahcorde, 2021-04-27:
You should install `ros2_control`, you have two options:

 - Install sources from this repository  or
 - `sudo apt-get install ros-foxy-ros2-control`

`ros2_control` is not released yet in Galatic, if you are using this version you must install ros2_control from sources.

Please ask this kind of question here: https://answers.gazebosim.org/questions/

closed 2021-04-27 by ahcorde

newcanopies, 2021-04-28:
Thank you @ahcorde 

Not sure if this a local or global issue, but for Foxy, I had to directly `git clone https://github.com/ros-controls/ros2_control.git`

because my apt-get is fully updated after successfully installing moveit2
but  `sudo apt-get install ros-foxy-ros2-control` gave this error:

```
[sudo] password for user:
Reading package lists... Done
Building dependency tree
Reading state information... Done
The following packages were automatically installed and are no longer required:
  python3-xkit ubuntu-drivers-common
Use 'sudo apt autoremove' to remove them.
The following additional packages will be installed:
  ros-foxy-controller-interface ros-foxy-controller-manager ros-foxy-controller-manager-msgs ros-foxy-hardware-interface
  ros-foxy-ros2-control-test-assets ros-foxy-ros2controlcli
The following NEW packages will be installed:
  ros-foxy-controller-interface ros-foxy-controller-manager ros-foxy-controller-manager-msgs ros-foxy-hardware-interface
  ros-foxy-ros2-control ros-foxy-ros2-control-test-assets ros-foxy-ros2controlcli
0 upgraded, 7 newly installed, 0 to remove and 10 not upgraded.
Need to get 1,264 kB of archives.
After this operation, 12.8 MB of additional disk space will be used.
Do you want to continue? [Y/n] Y
```
I
```
Err:1 http://packages.ros.org/ros2/ubuntu focal/main amd64 ros-foxy-hardware-interface amd64 0.4.0-1focal.20210415.032844
  404  Not Found [IP: 64.50.233.100 80]
Err:2 http://packages.ros.org/ros2/ubuntu focal/main amd64 ros-foxy-controller-interface amd64 0.4.0-1focal.20210415.045005
  404  Not Found [IP: 64.50.233.100 80]
Err:3 http://packages.ros.org/ros2/ubuntu focal/main amd64 ros-foxy-controller-manager-msgs amd64 0.4.0-1focal.20210415.023245
  404  Not Found [IP: 64.50.233.100 80]
Err:4 http://packages.ros.org/ros2/ubuntu focal/main amd64 ros-foxy-controller-manager amd64 0.4.0-1focal.20210415.045203
  404  Not Found [IP: 64.50.233.100 80]
Err:5 http://packages.ros.org/ros2/ubuntu focal/main amd64 ros-foxy-ros2-control-test-assets amd64 0.4.0-1focal.20210408.165615
  404  Not Found [IP: 64.50.233.100 80]
Err:6 http://packages.ros.org/ros2/ubuntu focal/main amd64 ros-foxy-ros2controlcli amd64 0.4.0-1focal.20210415.050212
  404  Not Found [IP: 64.50.233.100 80]
Err:7 http://packages.ros.org/ros2/ubuntu focal/main amd64 ros-foxy-ros2-control amd64 0.4.0-1focal.20210415.050937
  404  Not Found [IP: 64.50.233.100 80]
E: Failed to fetch http://packages.ros.org/ros2/ubuntu/pool/main/r/ros-foxy-hardware-interface/ros-foxy-hardware-interface_0.4.0-1focal.20210415.032844_amd64.deb  404  Not Found [IP: 64.50.233.100 80]
E: Failed to fetch http://packages.ros.org/ros2/ubuntu/pool/main/r/ros-foxy-controller-interface/ros-foxy-controller-interface_0.4.0-1focal.20210415.045005_amd64.deb  404  Not Found [IP: 64.50.233.100 80]
E: Failed to fetch http://packages.ros.org/ros2/ubuntu/pool/main/r/ros-foxy-controller-manager-msgs/ros-foxy-controller-manager-msgs_0.4.0-1focal.20210415.023245_amd64.deb  404  Not Found [IP: 64.50.233.100 80]
E: Failed to fetch http://packages.ros.org/ros2/ubuntu/pool/main/r/ros-foxy-controller-manager/ros-foxy-controller-manager_0.4.0-1focal.20210415.045203_amd64.deb  404  Not Found [IP: 64.50.233.100 80]
E: Failed to fetch http://packages.ros.org/ros2/ubuntu/pool/main/r/ros-foxy-ros2-control-test-assets/ros-foxy-ros2-control-test-assets_0.4.0-1focal.20210408.165615_amd64.deb  404  Not Found [IP: 64.50.233.100 80]
E: Failed to fetch http://packages.ros.org/ros2/ubuntu/pool/main/r/ros-foxy-ros2controlcli/ros-foxy-ros2controlcli_0.4.0-1focal.20210415.050212_amd64.deb  404  Not Found [IP: 64.50.233.100 80]
E: Failed to fetch http://packages.ros.org/ros2/ubuntu/pool/main/r/ros-foxy-ros2-control/ros-foxy-ros2-control_0.4.0-1focal.20210415.050937_amd64.deb  404  Not Found [IP: 64.50.233.100 80]
E: Unable to fetch some archives, maybe run apt-get update or try with --fix-missing?
```

not sure if relevant but, I am **not** using amd64 seen above

processor        : x86_64

   RMW MIDDLEWARE
middleware name    : rmw_fastrtps_cpp

   ROS 2 INFORMATION
distribution name      : foxy
distribution type      : ros2
distribution status    : active
release platforms      : {'ubuntu': ['focal']}

newcanopies, 2021-04-28:
same error after updating ROS2 key:
`curl http://repo.ros2.org/repos.key | sudo apt-key add -`

newcanopies, 2021-04-28:
Hi
Unfortunately, the build fail persists, even after successfully building `ros2_control` in the same `ROS2_ws`

Same error at:
```
Starting >>> gazebo_ros2_control
--- stderr: gazebo_ros2_control
CMake Error at CMakeLists.txt:6 (find_package):
  By not providing "Findcontroller_manager.cmake" in CMAKE_MODULE_PATH this
  project has asked CMake to find a package configuration file provided by
  "controller_manager", but CMake did not find one.
```

thanks for your help!

newcanopies, 2021-04-29:
hi @ahcorde 
`ros2_control` is built and sourced, 
but `gazebo_ros2_control` fails to build. How to solve?
Thank you!

ahcorde, 2021-04-29:
Can you clarify your setup ? 

 - ROS 2 version
 - packages inside the workspace

newcanopies, 2021-04-29:
finally solved -- with a master COLCON workspace for all ROS2 packages:

export   COLCON_WS=~/ws_ROS2
mkdir -p   $COLCON_WS/src
cd   $COLCON_WS/src
git clone some-ros-pkg-repo
cd   $COLCON_WS/
colcon build --symlink-install --packages-select <name-of-pkg><name-of-another-pkg>

the below original workspace tree was failing to build inside gazebo-ros2-control:

~/ws_ROS2/
                 |__gazebo_ros2_control/src
                 |__gazebo_ros_pkgs/src
                 |__ros2_control/src
                 |__etc/src

This COLCON master workspace unification prompted my other Issue question.
Thanks very much again!