ros-controls/gazebo_ros2_control #73: Unstable behaviour of Position Controller in UR10

panagelak, 2021-05-06:
Hello,

I recently tested your gazebo_ros2_control plugin (master branch) using position interface with a ur10 robot + MoveIt

by just modifying the rrbot moveit demo for the ur10 

I noticed that the joint_trajectory_controller  was unstable causing the robot to take a long time to go in the default position
and breaking up like so

[image]

Reducing the inertial values by a factor of 100 seemed to fix the problem in the beginning but it becomes unstable after a while
or after using Moveit to move it
(i haven't tested effort or velocity controllers yet since i don't want to tune pid's)

Any ideas?

Thanks a lot!

brennand, 2021-06-09:
Did you ever fix this as I have a very similar behavior.

newcanopies, 2021-06-10:
@panagelak please share a https://gist.github.com of the:

 adapted UR10 xacro with the `joint_trajectory_controller`
 moveit config

panagelak, 2021-06-10:
@newcanopies  

Hello i wasnt very familiar with gist and it seems i could only share individual files

I made a repo with my config and some simple instructions (for foxy) hope this helps

https://github.com/panagelak/ur10_gazebo_ros_control2

Hope this helps!
Thanks

newcanopies, 2021-06-10:
Thank you @panagelak I'm looking into tweaking the values to see if comes down to params (https://www.youtube.com/watch?v=pJheYXHP4jM)

Will update

newcanopies, 2021-06-11:
are you getting the ros2 errors:
```
[ros2-6] ros2: error: argument Call `ros2 <command> -h` for more detailed usage.: invalid choice: 'control' (choose from 'action', 'bag', 'component', 'daemon', 'doctor', 'extension_points', 'extensions', 'interface', 'launch', 'lifecycle', 'multicast', 'node', 'param', 'pkg', 'run', 'security', 'service', 'test', 'topic', 'wtf')
**[ERROR] [ros2-6]: process has died [pid 10363, exit code 2, cmd 'ros2 control load_start_controller joint_trajectory_controller'].**
[INFO] [gzclient   -2]: process has finished cleanly [pid 10167]
```

<details>
```
[gzserver-1] [INFO] [1623424230.447833300] [gazebo_ros2_control]: Loading controller_manager
[INFO] [spawn_entity.py-4]: process has finished cleanly [pid 10172]
[INFO] [ros2-5]: process started with pid [10298]
[gzserver-1] [WARN] [1623424230.468205600] [gazebo_ros2_control]:  Desired controller update period (0.01 s) is slower than the gazebo simulation period (0.001 s).
[gzserver-1] [INFO] [1623424230.468415400] [gazebo_ros2_control]: **Loaded gazebo_ros2_control.**
[ros2-5] usage: ros2 [-h] Call `ros2 <command> -h` for more detailed usage. ...
[ros2-5] ros2: error: argument Call `ros2 <command> -h` for more detailed usage.: invalid choice: 'control' (choose from 'action', 'bag', 'component', 'daemon', 'doctor', 'extension_points', 'extensions', 'interface', 'launch', 'lifecycle', 'multicast', 'node', 'param', 'pkg', 'run', 'security', 'service', 'test', 'topic', 'wtf')
[ERROR] [ros2-5]: process has died [pid 10298, exit code 2, cmd 'ros2 control load_start_controller joint_state_controller'].
[INFO] [ros2-6]: process started with pid [10363]
[ros2-6] usage: ros2 [-h] Call `ros2 <command> -h` for more detailed usage. ...
[ros2-6] ros2: error: argument Call `ros2 <command> -h` for more detailed usage.: invalid choice: 'control' (choose from 'action', 'bag', 'component', 'daemon', 'doctor', 'extension_points', 'extensions', 'interface', 'launch', 'lifecycle', 'multicast', 'node', 'param', 'pkg', 'run', 'security', 'service', 'test', 'topic', 'wtf')
**[ERROR] [ros2-6]: process has died [pid 10363, exit code 2, cmd 'ros2 control load_start_controller joint_trajectory_controller'].**
[INFO] [gzclient   -2]: process has finished cleanly [pid 10167]
```

newcanopies, 2021-06-11:
https://groups.google.com/g/moveit-users/c/zNRG50tHuuY/m/V10uDx8Slw4J

panagelak, 2021-06-11:
@newcanopies 

I'm not sure why it doesn't recognize the ros2 control load_start_controller joint_state_controller/joint_trajectory_controller  cli command, i think it requires to have build ros_control from source to do so (or moveit2 which include ros control).
Furthermore in this launch file on the rrbot it uses the same command on line 90 https://github.com/ros-simulation/gazebo_ros_demos/blob/ahcorde/port/ros2/rrbot_gazebo/launch/rrbot_world.launch.py

I followed this instructions to install moveit2 https://moveit.ros.org/install-moveit2/source/

P.S since i dont want to have all the moveit2 packages in my devel workspace (requiring to blacklist them etc)
i installed the moveit2 in a seperate workspace so my bashrc looks something like this (notice the --extend)

source /opt/ros/foxy/setup.bash
source ~/ws_moveit2/install/setup.bash --extend
source ~/colcon_ws/install/setup.bash

let me know if this fixes that

newcanopies, 2021-06-11:
the recommendation seems to be to install debian `ros2_control` and resolve conflicts with source built `moveit2`

have a look at my PR, tuning the gazebo yaml config seems promising. 

if your ur10 package is in `~/devel_ws` what's in `~/colcon_ws` ?

next I'll have a look at Rviz error from the ros2 launch ur10_moveit_demo_nodes ur10_gazebo_demo.launch.py
```
viz2-6] Warning: Invalid frame ID "wrist_1_link" passed to canTransform argument source_frame - frame does not exist
[rviz2-6]          at line 133 in /tmp/binarydeb/ros-foxy-tf2-0.13.10/src/buffer_core.cpp
[rviz2-6] Warning: Invalid frame ID "wrist_2_link" passed to canTransform argument source_frame - frame does not exist
[rviz2-6]          at line 133 in /tmp/binarydeb/ros-foxy-tf2-0.13.10/src/^C
```

newcanopies, 2021-06-11:
in that case, is the ur10 control package located in `~/colcon_ws` ?

since commenting out the `update_rate` in gazebo yaml removes the instability,
could be that the URDF macros require a matching <update_rate>100</update_rate> tag 
as well as further gazebo yaml param tweaks

newcanopies, 2021-06-11:
in your case, is the ur10 control package located in `~/colcon_ws` ?

since commenting out the controller_manager  `update_rate` in gazebo yaml removes the instability,
could be that the URDF macros require a matching <update_rate>100</update_rate> tag 
as well as correct yaml definitions of the joint state and trajectory controllers

have you tried generating the ur10_gazebo_ros_control2 yamls and gazebo xacros via (the unported) ROS1 moveit setup assistant?

here's a UR5 controller yaml tweaked to hold non-zero position https://github.com/ros-simulation/gazebo_ros2_control/pull/44#issuecomment-765253220

panagelak, 2021-06-12:
hi @newcanopies  

i experimented a little with the gazebo_controllers.yaml (on the ur10_description pkg) but it didn't seem to fix it by removing the <update_rate> 

can you possibly make a pr with your fix or share your fork? 

i didn't know they have release ros2_control on debian is it on testing? i have install moveit2 from source which installs ros2_control also. In colcon_ws i have only the repo i shared + gazebo_ros2_control plugin (main)

thanks a lot for your help so far, appreciate it!

newcanopies, 2021-06-12:
hi @panagelak I already pushed the PR to your repo. I am getting this result based on debian `ros2_control`
[image]

Try these [install/remove instructions](https://industrial-training-master.readthedocs.io/en/foxy/_source/session1/ros2/2-Installing-Existing-Packages.html#scan-n-plan-application-problem-statement) if you choose to test debian packages instead. It looks like the Rviz2 error about TF frame conversion is due to wrong TF package version overlay. 

Currently, I am debugging based on `ros2_control` devel packages from the [demos](https://github.com/ros-controls/ros2_control_demos) and documentation therein. Also check the 6Dbot branch where a 2 Hz update rate was coded in.
I tried various values but instability only stops if no update rate is set for the controller_manager. Meanwhile. each controller (state,trajectory, position) has an update rate that may or may not need tweaking.... I'm looking for documentation that clarifies this. 

It's also possible some of these should be included as <build> or <exec> depends under the ur10 description+control package.xml

Briancbn, 2021-06-12:
@newcanopies @panagelak This is my follow-up comments. https://github.com/ros-simulation/gazebo_ros2_control/pull/44#discussion_r565177746.

Please also refer to https://github.com/ros-simulation/gazebo_ros2_control/issues/54#issuecomment-775580231 for the workaround 1. I am still using this workaround with the latest gazebo_ros2_control for my own project.

I hasn't been follow up on this issue lately, maybe @destogl and @ahcorde could provide the best way and estimated timeline in this getting resolved.

newcanopies, 2021-06-12:
> Please also refer to [#54 (comment)](https://github.com/ros-simulation/gazebo_ros2_control/issues/54#issuecomment-775580231) for the workaround 1. I am still using this workaround with the latest gazebo_ros2_control for my own project.
> 
> I hasn't been follow up on this issue lately, maybe @destogl and @ahcorde could provide the best way and estimated timeline in this getting resolved.

Thanks very much @Briancbn - which `ros2_control` package repo are you using and how do you exclude conflict with the older ros2_control inside `moveit2` ?

Briancbn, 2021-06-12:
Latest version for almost everything, we tested it just two days ago.

Don't forget to add these line as the `follow_joint_trajectory` parameters under `controller.yaml` following the moveit2 panda example.
```yaml
pos_joint_traj_controller:
  ros__parameters:
    joints:
      -
      -
      ...
    command_interfaces:
      - position
    state_interfaces:
      - position
      - velocity
```
also don't forget to load the controllers in your Moveit2 launch file. Follow the example here.
https://github.com/ros-planning/moveit2/blob/c029942dbb95387ecccf10a41d48e310450d3528/moveit_demo_nodes/run_moveit_cpp/launch/run_moveit_cpp.launch.py#L135-L154

Latest Moveit2
```yaml
repositories:
  # moveit2
  moveit2:
    type: git
    url: https://github.com/ros-planning/moveit2
    version: 2.1.4
  moveit_msgs:
    type: git
    url: https://github.com/ros-planning/moveit_msgs
    version: 2.0.1
  moveit_resources:
    type: git
    url: https://github.com/ros-planning/moveit_resources
    version: 2.0.2
  geometric_shapes:
    type: git
    url: https://github.com/ros-planning/geometric_shapes
    version: 2.0.2
  srdfdom:
    type: git
    url: https://github.com/ros-planning/srdfdom
    version: 2.0.2

  # ros2_control
  ros2_control:
    type: git
    url: https://github.com/ros-controls/ros2_control
    version: 0.7.0
  ros2_controllers:
    type: git
    url: https://github.com/ros-controls/ros2_controllers
    version: d989e7caa2d3e862b3e4e8e1eb7934f1f216eeae

  # warehouse
  warehouse_ros:
    type: git
    url: https://github.com/ros-planning/warehouse_ros
    version: 2.0.1

  warehouse_ros_mongo:
    type: git
    url: https://github.com/ros-planning/warehouse_ros_mongo
    version: 2.0.1
```

gazebo_ros2_control
```yaml
repositories:
  gazebo_ros2_control:
    type: git
    url: https://github.com/ros-simulation/gazebo_ros2_control
    version: a33ea2eeb37094177103084e065626d63ce31a7d
```

Drojas251, 2021-06-28:
@Briancbn Are you installing those versions of ros_contro and ros_controllers from source? I have installed the versions of these packages that you have listed, yet I am still getting TF2 frame conversion errors. Do these versions you have listed still work together?

shonigmann, 2021-08-13:
fwiw, I can build everything using the default branches for each dependency (moveit2, ros2_control, gazebo_ros2_control, etc) rather than specific commits and things now behave as expected. In #54, I mention a few pitfalls / tweaks to watch out for. But I don't think it is still _necessary_ to build using specific versions (you of course still can if you like a bit of stability in your life😄)

dvalenciar, 2021-09-09:
> Did you ever fix this as I have a very similar behavior.

Me too, any idea how to solve it?

AlezzzRd, 2022-05-09:
I had a very similar problem with gazebo11 and I finally solved it by adding friction and damping parameters in each joint. like
`
  <joint name="$(arg prefix)joint_6" type="continuous">
    <parent link="$(arg prefix)link_5" />
    <child link="$(arg prefix)link_6" />
    <origin xyz="${joint_6_xyz}" rpy="${joint_6_rpy}" />
    <axis xyz="${joint_6_axis}" />
    <limit effort="${joint_6_max_effort}" velocity="${joint_6_max_speed}" />
    <dynamics damping="60.0" friction="4.0"/>
  </joint>
`

[image]