ros-controls/gazebo_ros2_control #90: Prismatic joint drops although the position command says "do not move"!

farbod-new, 2021-09-30:
Hi all,

I have a vertical prismatic joint that I want to control using the "position control" interface. My position command requires the joint to stay at a point for a while, but the joint creeps downwards despite a constant position command. See the video clip.

https://user-images.githubusercontent.com/76223948/135539766-5f45e070-9204-4343-be48-cbbd8cad72b0.mp4

Why is this happening?

Here is my urdf:
```
<?xml version="1.0" ?>
<!-- =================================================================================== -->
<!-- |    This document was autogenerated by xacro from cartpole.xacro.urdf            | -->
<!-- |    EDITING THIS FILE BY HAND IS NOT RECOMMENDED                                 | -->
<!-- =================================================================================== -->
<robot name="cartopole">
  <link name="world"/>
  <link name="slideBar">
    <visual>
      <geometry>
        <box size="30 0.05 0.05"/>
      </geometry>
      <origin xyz="0 0 0"/>
      <material name="green">
        <color rgba="0 0.8 .8 1"/>
      </material>
    </visual>
    <inertial>
      <mass value="100"/>
      <inertia ixx="1.0" ixy="0.0" ixz="0.0" iyy="1.0" iyz="0.0" izz="1.0"/>
    </inertial>
  </link>
  <link name="cart">
    <visual>
      <geometry>
        <box size="0.5 0.5 0.2"/>
      </geometry>
      <origin xyz="0 0 0"/>
      <material name="blue">
        <color rgba="0 0 .8 1"/>
      </material>
    </visual>
    <inertial>
      <mass value="1"/>
      <inertia ixx="1.0" ixy="0.0" ixz="0.0" iyy="1.0" iyz="0.0" izz="1.0"/>
    </inertial>
  </link>
  <joint name="world_to_base" type="fixed">
    <origin rpy="0 0 0" xyz="0 0 1"/>
    <parent link="world"/>
    <child link="slideBar"/>
  </joint>
  <joint name="slider_to_cart" type="prismatic">
    <axis xyz="0 0 1"/>
    <origin xyz="0.0 0.0 0.0"/>
    <parent link="slideBar"/>
    <child link="cart"/>
    <limit effort="1000.0" lower="0" upper="15" velocity="30"/>
    <dynamics damping="0.0" friction="0.0"/>
  </joint>

  <ros2_control name="GazeboSystem" type="system">
    <hardware>
      <plugin>gazebo_ros2_control/GazeboSystem</plugin>
    </hardware>
    <joint name="slider_to_cart">
      <command_interface name="position">
        <param name="min">0</param>
        <param name="max">15</param>
      </command_interface>
[... 196 lines cut ...]
  rclcpp_action::ClientGoalHandle<control_msgs::action::FollowJointTrajectory>::SharedPtr
    goal_handle = goal_handle_future.get();
  if (!goal_handle) {
    RCLCPP_ERROR(node->get_logger(), "Goal was rejected by server");
    return 1;
  }
  RCLCPP_ERROR(node->get_logger(), "Goal was accepted by server");

  // Wait for the server to be done with the goal
  auto result_future = action_client->async_get_result(goal_handle);
  RCLCPP_INFO(node->get_logger(), "Waiting for result");
  if (rclcpp::spin_until_future_complete(node, result_future) !=
    rclcpp::FutureReturnCode::SUCCESS)
  {
    RCLCPP_ERROR(node->get_logger(), "get result call failed :(");
    return 1;
  }

  std::cout << "async_send_goal" << std::endl;
  rclcpp::shutdown();

  return 0;
}

```

sh-cobot, 2023-04-12:
Did you ever solve this issue?

farbod-new, 2023-04-15:
Yes! I added a rather large friction to the prismatic joint. First, I calculated the weight of the object (link). Then, added a friction force that was greater than the weight of the object (link).

I hope this helps.

christophfroehlich, 2023-12-30:
I suppose this was fixed with #213, please reopen of this is not the case.

closed 2023-12-30 by christophfroehlich