ros-controls/gazebo_ros2_control #98: cannot compile

midu3030, 2021-11-09:
when running

`docker build -t gazebo_ros2_control .`

i get this error

```
Step 5/8 : RUN cd /home/ros2_ws/   && . /opt/ros/foxy/setup.sh   && colcon build --merge-install
 ---> Running in 08c031c1e67a
Starting >>> gazebo_ros2_control
--- stderr: gazebo_ros2_control
In file included from /home/ros2_ws/src/gazebo_ros2_control/gazebo_ros2_control/src/gazebo_system.cpp:21:
/home/ros2_ws/src/gazebo_ros2_control/gazebo_ros2_control/include/gazebo_ros2_control/gazebo_system.hpp:41:3: error: ‘CallbackReturn’ does not name a type
   41 |   CallbackReturn on_init(const hardware_interface::HardwareInfo & system_info)
      |   ^~~~~~~~~~~~~~
/home/ros2_ws/src/gazebo_ros2_control/gazebo_ros2_control/include/gazebo_ros2_control/gazebo_system.hpp:51:3: error: ‘CallbackReturn’ does not name a type
   51 |   CallbackReturn on_activate(const rclcpp_lifecycle::State & previous_state) override;
      |   ^~~~~~~~~~~~~~
/home/ros2_ws/src/gazebo_ros2_control/gazebo_ros2_control/include/gazebo_ros2_control/gazebo_system.hpp:54:3: error: ‘CallbackReturn’ does not name a type
   54 |   CallbackReturn on_deactivate(const rclcpp_lifecycle::State & previous_state) override;
      |   ^~~~~~~~~~~~~~
/home/ros2_ws/src/gazebo_ros2_control/gazebo_ros2_control/src/gazebo_system.cpp:352:1: error: ‘CallbackReturn’ does not name a type
  352 | CallbackReturn
      | ^~~~~~~~~~~~~~
/home/ros2_ws/src/gazebo_ros2_control/gazebo_ros2_control/src/gazebo_system.cpp:373:1: error: ‘CallbackReturn’ does not name a type
  373 | CallbackReturn GazeboSystem::on_activate(const rclcpp_lifecycle::State & previous_state)
      | ^~~~~~~~~~~~~~
/home/ros2_ws/src/gazebo_ros2_control/gazebo_ros2_control/src/gazebo_system.cpp:378:1: error: ‘CallbackReturn’ does not name a type
  378 | CallbackReturn GazeboSystem::on_deactivate(const rclcpp_lifecycle::State & previous_state)
      | ^~~~~~~~~~~~~~
In file included from /home/ros2_ws/src/gazebo_ros2_control/gazebo_ros2_control/src/gazebo_ros2_control_plugin.cpp:43:
/home/ros2_ws/src/gazebo_ros2_control/gazebo_ros2_control/include/gazebo_ros2_control/gazebo_system.hpp:41:3: error: ‘CallbackReturn’ does not name a type
   41 |   CallbackReturn on_init(const hardware_interface::HardwareInfo & system_info)
      |   ^~~~~~~~~~~~~~
/home/ros2_ws/src/gazebo_ros2_control/gazebo_ros2_control/include/gazebo_ros2_control/gazebo_system.hpp:51:3: error: ‘CallbackReturn’ does not name a type
   51 |   CallbackReturn on_activate(const rclcpp_lifecycle::State & previous_state) override;
      |   ^~~~~~~~~~~~~~
/home/ros2_ws/src/gazebo_ros2_control/gazebo_ros2_control/include/gazebo_ros2_control/gazebo_system.hpp:54:3: error: ‘CallbackReturn’ does not name a type
   54 |   CallbackReturn on_deactivate(const rclcpp_lifecycle::State & previous_state) override;
      |   ^~~~~~~~~~~~~~
/home/ros2_ws/src/gazebo_ros2_control/gazebo_ros2_control/src/gazebo_ros2_control_plugin.cpp: In member function ‘virtual void gazebo_ros2_control::GazeboRosControlPlugin::Load(gazebo::physics::ModelPtr, sdf::v9::ElementPtr)’:
/home/ros2_ws/src/gazebo_ros2_control/gazebo_ros2_control/src/gazebo_ros2_control_plugin.cpp:305:85: error: no matching function for call to ‘hardware_interface::ResourceManager::import_component(std::remove_reference<std::unique_ptr<gazebo_ros2_control::GazeboSystemInterface>&>::type, __gnu_cxx::__alloc_traits<std::allocator<hardware_interface::HardwareInfo>, hardware_interface::HardwareInfo>::value_type&)’
  305 |     resource_manager_->import_component(std::move(gazeboSystem), control_hardware[i]);
      |                                                                                     ^
In file included from /opt/ros/foxy/include/controller_manager/controller_manager.hpp:39,
                 from /home/ros2_ws/src/gazebo_ros2_control/gazebo_ros2_control/include/gazebo_ros2_control/gazebo_ros2_control_plugin.hpp:42,
                 from /home/ros2_ws/src/gazebo_ros2_control/gazebo_ros2_control/src/gazebo_ros2_control_plugin.cpp:42:
/opt/ros/foxy/include/hardware_interface/resource_manager.hpp:151:8: note: candidate: ‘void hardware_interface::ResourceManager::import_component(std::unique_ptr<hardware_interface::ActuatorInterface>)’
  151 |   void import_component(std::unique_ptr<ActuatorInterface> actuator);
      |        ^~~~~~~~~~~~~~~~
/opt/ros/foxy/include/hardware_interface/resource_manager.hpp:151:8: note:   candidate expects 1 argument, 2 provided
/opt/ros/foxy/include/hardware_interface/resource_manager.hpp:169:8: note: candidate: ‘void hardware_interface::ResourceManager::import_component(std::unique_ptr<hardware_interface::SensorInterface>)’
  169 |   void import_component(std::unique_ptr<SensorInterface> sensor);
      |        ^~~~~~~~~~~~~~~~
/opt/ros/foxy/include/hardware_interface/resource_manager.hpp:169:8: note:   candidate expects 1 argument, 2 provided
/opt/ros/foxy/include/hardware_interface/resource_manager.hpp:187:8: note: candidate: ‘void hardware_interface::ResourceManager::import_component(std::unique_ptr<hardware_interface::SystemInterface>)’
  187 |   void import_component(std::unique_ptr<SystemInterface> system);
      |        ^~~~~~~~~~~~~~~~
/opt/ros/foxy/include/hardware_interface/resource_manager.hpp:187:8: note:   candidate expects 1 argument, 2 provided
/home/ros2_ws/src/gazebo_ros2_control/gazebo_ros2_control/src/gazebo_ros2_control_plugin.cpp: In member function ‘void gazebo_ros2_control::GazeboRosControlPrivate::Update()’:
/home/ros2_ws/src/gazebo_ros2_control/gazebo_ros2_control/src/gazebo_ros2_control_plugin.cpp:373:57: error: no matching function for call to ‘controller_manager::ControllerManager::update(rclcpp::Time&, rclcpp::Duration&)’
  373 |     controller_manager_->update(sim_time_ros, sim_period);
      |                                                         ^
In file included from /home/ros2_ws/src/gazebo_ros2_control/gazebo_ros2_control/include/gazebo_ros2_control/gazebo_ros2_control_plugin.hpp:42,
                 from /home/ros2_ws/src/gazebo_ros2_control/gazebo_ros2_control/src/gazebo_ros2_control_plugin.cpp:42:
/opt/ros/foxy/include/controller_manager/controller_manager.hpp:133:37: note: candidate: ‘controller_interface::return_type controller_manager::ControllerManager::update()’
  133 |   controller_interface::return_type update();
      |                                     ^~~~~~
/opt/ros/foxy/include/controller_manager/controller_manager.hpp:133:37: note:   candidate expects 0 arguments, 2 provided
In file included from /opt/ros/foxy/include/class_loader/class_loader_core.hpp:57,
[... 15 lines cut ...]
/opt/ros/foxy/include/hardware_interface/system_interface.hpp:48:23: note: 	‘virtual hardware_interface::return_type hardware_interface::SystemInterface::configure(const hardware_interface::HardwareInfo&)’
   48 |   virtual return_type configure(const HardwareInfo & system_info) = 0;
      |                       ^~~~~~~~~
/opt/ros/foxy/include/hardware_interface/system_interface.hpp:116:23: note: 	‘virtual hardware_interface::return_type hardware_interface::SystemInterface::start()’
  116 |   virtual return_type start() = 0;
      |                       ^~~~~
/opt/ros/foxy/include/hardware_interface/system_interface.hpp:122:23: note: 	‘virtual hardware_interface::return_type hardware_interface::SystemInterface::stop()’
  122 |   virtual return_type stop() = 0;
      |                       ^~~~
/opt/ros/foxy/include/hardware_interface/system_interface.hpp:128:23: note: 	‘virtual std::string hardware_interface::SystemInterface::get_name() const’
  128 |   virtual std::string get_name() const = 0;
      |                       ^~~~~~~~
/opt/ros/foxy/include/hardware_interface/system_interface.hpp:134:18: note: 	‘virtual hardware_interface::status hardware_interface::SystemInterface::get_status() const’
  134 |   virtual status get_status() const = 0;
      |                  ^~~~~~~~~~
make[2]: *** [CMakeFiles/gazebo_hardware_plugins.dir/build.make:63: CMakeFiles/gazebo_hardware_plugins.dir/src/gazebo_system.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:80: CMakeFiles/gazebo_hardware_plugins.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
make[2]: *** [CMakeFiles/gazebo_ros2_control.dir/build.make:63: CMakeFiles/gazebo_ros2_control.dir/src/gazebo_ros2_control_plugin.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:107: CMakeFiles/gazebo_ros2_control.dir/all] Error 2
make: *** [Makefile:141: all] Error 2
---
Failed   <<< gazebo_ros2_control [12.2s, exited with code 2]

```

MarcoMagriDev, 2021-11-23:
I faced the same issue. If you are using the Foxy distribution try to switch to branch "foxy" before compiling. 
In my case, this approach solved the problem (I can even compile it without using docker with a simple colcon build).

JeanMarchal, 2021-11-24:
I got the same error and the closed issue "Colcon build error, is this foxy supported? #86" by Kakcalu13 helped me to solve it : https://github.com/ros-simulation/gazebo_ros2_control/issues/86

Hope it works for you too 👍

bmagyar, 2021-11-25:
I cordially invite you all to fix the underlying issue. Getting stuff from
closed issues may work once but it only leads to more people complaining
and using the same set of artesanal solutions to something they expect to
work.

Check in the repo, the dockerfile is likely simply outdated. Please test a
fix with your setup and submit a pull request with that.

Yes, you are right. I could have fixed the problem while I wrote this
message. I still prefer however to teach people to fish instead of feeding
them. I eagerly await your pull request.

bmagyar, 2021-11-26:
@ahcorde was kind enough to do this for ya:
#101  #102