ros-controls/gazebo_ros2_control #99: example_position not working

phat-truongxuan, 2021-11-15:
Hi 
My setup: Ubuntu 20.04 Focal Fossa , Ros Galactic 
I tried cloning the repo and ran 
    ros2 launch gazebo_ros2_control_demos cart_example_position.launch.py
then 
    ros2 run gazebo_ros2_control_demos example_position
but the cart just stayed still, I tried echo:
    ros2 topic echo /joint_trajectory_controller/state
and the desire position changed, the actual did not ( as shown on gazebo ) and the error position did followed along the position, which means the system did aware the cart did not move. the terminal that I used fro bringing up gazebo showed "Goal reached, success!
" which is kind of weird  . I installed all the needed packages and dependencies. 
P/s the issue just happened today, I ran this repo last week from today and it worked perfectly , any help will be much appreciated

ahcorde, 2021-11-15:
@phat-truongxuan, 

How did you install `ros2_control` ? you need to install packages from the testing repo or compile then from sources. The right packages are not yet sync.

phat-truongxuan, 2021-11-15:
@ahcorde hi, thank you for replying 
I install with 
sudo apt install ros-galactic-ros2-control
so you mean I need to install from the repo https://github.com/ros-controls/ros2_control ? the current apt install method packages not yet sync and won"t be able to work ?

ahcorde, 2021-11-15:
packages are not sync, or you try `testing` or you compile from sources. Next sync we will be able to use it.

phat-truongxuan, 2021-11-15:
do you know when is the next sync being released? I will try compile from source and feedback the result

ahcorde, 2021-11-15:
It's happening every month https://discourse.ros.org/c/release/galactic/72

closed 2022-01-18 by ahcorde