ros-controls/gazebo_ros2_control #108: Default joint position for simulated robot

christophfroehlich, 2022-02-08:
Hi there! I just have seen a long-awaited feature with commit https://github.com/ros-simulation/gazebo_ros2_control/commit/26216b49eb19417b2f7dc75fe61407a07c89bd60 to set the initial joint position by @destogl 
The example works fine, but it seems that it works only with a position command interface? 

Just to be sure that I got that right: Setting an initial position in `test_cart_effort.xacro.urdf` does not have the desired effect when launching with
`ros2 launch gazebo_ros2_control_demos cart_example_effort.launch.py`
despite the console output
> Loading joint: slider_to_cart
> [gzserver-1] [INFO] [1644310803.234515970] [gazebo_ros2_control]: 	State:
> [gzserver-1] [INFO] [1644310803.234586772] [gazebo_ros2_control]: 		 position
> [gzserver-1] [INFO] [1644310803.235033020] [gazebo_ros2_control]: 			 found initial value: 1.000000

If this is the (intended) case, it is no direct replacement for the `-J ` option for `gazebo_ros spawn_entity.py`? E.g., as requested here 
https://github.com/ros-simulation/gazebo_ros_pkgs/issues/1233 or https://answers.gazebosim.org/question/27584/ros2-setting-initial-joint-position-at-spawn/

Is there any other known workaround available you could point me to? Thanks.

destogl, 2022-02-12:
@christophfroehlich can you please test #110 if it fixes your issue?

christophfroehlich, 2022-02-14:
@destogl yes, thanks!
I already implemented something similar like you did last week, but realized that my version sets the initial values at the first simulation step, not before it. (I added a command in the `::write` method and not within the `::registerJoints` method like you did.)
For my robot this makes a difference, because with the chosen coordinate system it has a self collision with all joint positions being zero. By setting them to an "initial value" at the first simulation iteration, the robot crashes obviously.

closed 2022-02-14 by christophfroehlich

destogl, 2022-02-18:
@christophfroehlich can you please comment for required changes on #110 or approve PR there?

christophfroehlich, 2022-02-18:
It is working already as it is! I cannot approve the PR in the github workflow unless I'm assigned as a reviewer by a maintainer I guess.

destogl, 2022-02-20:
> It is working already as it is! I cannot approve the PR in the github workflow unless I'm assigned as a reviewer by a maintainer I guess.

you can always review/approve review. It will just not be green as if a maintainer does it. Nevertheless, it is valuable input fro maintainers knowing that somebody else tested the code and is happy with it :smile:

PR #110 (2022-07-22): Fix setting initial values if command interfaces are not defined.
This should fix #108.

The reason for the issue is that if the command interface does not exist, the initial value is never propagated to the simulator.
﻿