ros-controls/gazebo_ros2_control #116: Error parsing URDF in gazebo_ros2_control plugin, plugin not active : no ros2_control tag

AmetistDrake, 2022-03-10:
I am trying to launch the cart_example_position demo with ros2 galactic, but I get the error `no ros2_control tag`. The model loads into gazebo, but I am unable to control with the `example_position` node.

The problem is wierd to me, since I checked the urdf file in `gazebo_ros_control_demos/urdf/test_cart_position.xacro.urdf`, and there is a ros2_control tag.

I first run:
```
$ ros2 launch gazebo_ros2_control_demos cart_example_position.launch.py
```
The output:
```
[INFO] [launch]: All log files can be found below ...
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [gzserver-1]: process started with pid [119803]
[INFO] [gzclient   -2]: process started with pid [119805]
[INFO] [robot_state_publisher-3]: process started with pid [119807]
[INFO] [spawn_entity.py-4]: process started with pid [119810]
[robot_state_publisher-3] Link slideBar had 1 children
[robot_state_publisher-3] Link cart had 0 children
[robot_state_publisher-3] [INFO] [1646917390.859059113] [robot_state_publisher]: got segment cart
[robot_state_publisher-3] [INFO] [1646917390.859153373] [robot_state_publisher]: got segment slideBar
[robot_state_publisher-3] [INFO] [1646917390.859172378] [robot_state_publisher]: got segment world
[spawn_entity.py-4] [INFO] [1646917391.480074745] [spawn_entity]: Spawn Entity started
[spawn_entity.py-4] [INFO] [1646917391.480819853] [spawn_entity]: Loading entity published on topic robot_description
[spawn_entity.py-4] [INFO] [1646917391.483586253] [spawn_entity]: Waiting for entity xml on robot_description
[spawn_entity.py-4] [INFO] [1646917391.485859246] [spawn_entity]: Waiting for service /spawn_entity, timeout = 5
[spawn_entity.py-4] [INFO] [1646917391.486570312] [spawn_entity]: Waiting for service /spawn_entity
[spawn_entity.py-4] [INFO] [1646917392.749840927] [spawn_entity]: Calling service /spawn_entity
[spawn_entity.py-4] [INFO] [1646917393.013780900] [spawn_entity]: Spawn status: SpawnEntity: Successfully spawned entity [cartpole]
[spawn_entity.py-4] /opt/ros/galactic/lib/python3.8/site-packages/rclpy/qos.py:307: UserWarning: DurabilityPolicy.RMW_QOS_POLICY_DURABILITY_TRANSIENT_LOCAL is deprecated. Use DurabilityPolicy.TRANSIENT_LOCAL instead.
[spawn_entity.py-4]   warnings.warn(
[gzserver-1] [INFO] [1646917393.050729599] [gazebo_ros2_control]: Loading gazebo_ros2_control plugin
[gzserver-1] [INFO] [1646917393.056361207] [gazebo_ros2_control]: Starting gazebo_ros2_control plugin in namespace: /
[gzserver-1] [INFO] [1646917393.056595761] [gazebo_ros2_control]: Starting gazebo_ros2_control plugin in ros 2 node: gazebo_ros2_control
[gzserver-1] [INFO] [1646917393.056755299] [gazebo_ros2_control]: Loading parameter file /home/dani/Desktop/gazebo_ros2_control/install/gazebo_ros2_control_demos/share/gazebo_ros2_control_demos/config/cartpole_controller.yaml
[gzserver-1] 
[gzserver-1] [INFO] [1646917393.062013314] [gazebo_ros2_control]: connected to service!! robot_state_publisher
[gzserver-1] [ERROR] [1646917393.064470766] [rclcpp]: Received invalid sequence number. Ignoring...
[gzserver-1] [INFO] [1646917393.064867009] [gazebo_ros2_control]: Recieved urdf from param server, parsing...
[gzserver-1] [ERROR] [1646917393.065579076] [gazebo_ros2_control]: Error parsing URDF in gazebo_ros2_control plugin, plugin not active : no ros2_control tag
[INFO] [spawn_entity.py-4]: process has finished cleanly [pid 119810]
[INFO] [ros2-5]: process started with pid [119924]
[ros2-5] Could not contact service /controller_manager/load_controller
[ERROR] [ros2-5]: process has died [pid 119924, exit code 1, cmd 'ros2 control load_controller --set-state start joint_state_broadcaster'].
[INFO] [ros2-6]: process started with pid [120051]
[ros2-6] Could not contact service /controller_manager/load_controller
[ERROR] [ros2-6]: process has died [pid 120051, exit code 1, cmd 'ros2 control load_controller --set-state start joint_trajectory_controller'].
```

Then I tried:
```
ros2 run gazebo_ros2_control_demos example_position
```
The output:
```
node created
terminate called after throwing an instance of 'std::runtime_error'
  what():  could not get action server
```

Blebot0, 2023-06-05:
Any updates on this? Did you find a solution??

suchetanrs, 2023-06-20:
Hey, did you find any fix?

uby56, 2023-08-07:
Hi i am facing the same issue can anyone tell me how to solve this problem
[image]

christophfroehlich, 2023-08-25:
> Hi i am facing the same issue can anyone tell me how to solve this problem [image]

I don't think this is related to the original question of this issue, because you try to load a custom robot. The error comes from the ros2_control hardware_interface looking for a min value, which is actually optional only. maybe you have added a min-tag, but have malformed its value?