ros-controls/gazebo_ros2_control #123: How to inherit gazebo_ros2_control::GazeboSystem(gazebo_hardware_plugins) elegantly?

CaptainKAZ, 2022-04-07:
Hi I want to write a plugin to expand default hardware plugin (`gazebo_ros2_control::GazeboSystem`)
So I `#include <gazebo_ros2_control/gazebo_system.hpp>` and inherit the origin class like this `class GazeboXXXExpandedSystem : public gazebo_ros2_control::GazeboSystem` and then override `read` and `write` function to make my own behaiver
Question is, when I compile my plugin, the compiler throws error that class `gazebo_ros2_control::GazeboSystemPrivate` is undefined.
At last I have to copy the whole definition of `gazebo_ros2_control::GazeboSystemPrivate` into my src make the compiler happy. 
Is there any way I can do this more elegantly?

eslamsalahelsheikh, 2022-06-13:
Hi, I've the same problem. did you find a solution yet?

Blebot0, 2023-06-05:
Any updates on this? Did you find a solution?