ros-controls/gazebo_ros2_control #124: "Exception occured in the Load function of plugin...", assumes all plugins have base class type gazebo_ros2_control::GazeboSystemInterface

JacksonVanderkooy, 2022-05-09:
For reference I'm using the [galactic](https://github.com/ros-simulation/gazebo_ros2_control/tree/galactic) branch.

I've encountered an issue a few times where the plugin causes gazebo to crash if any hardware systems other than the one in the package are used. For example...
Functioning URDF:
```
<ros2_control name="example" type="system">
    <hardware>
        <plugin>gazebo_ros2_control/GazeboSystem</plugin>
    </hardware>
</ros2_control>
```
Non-Functioning URDF:
```
<ros2_control name="example" type="system">
    <hardware>
        <plugin>fake_components/GenericSystem</plugin>
    </hardware>
</ros2_control>
```
The error generated is `[Err] [Model.cc:1160] Exception occured in the Load function of plugin with name[gazebo_ros2_control] and filename[libgazebo_ros2_control.so]. This plugin will not run.`

I did some poking around and by adding a try/catch to lines 289-290 of [gazebo_ros2_control/src/gazebo_ros2_control_plugin.cpp](https://github.com/ros-simulation/gazebo_ros2_control/blob/c89584dcd4dcfe70a19b76ee57544a3e31e88926/gazebo_ros2_control/src/gazebo_ros2_control_plugin.cpp) (what I found to be the line that causes the error, not the reason for the error) I got the exception message of `According to the loaded plugin descriptions the class fake_components/GenericSystem with base class type gazebo_ros2_control::GazeboSystemInterface does not exist. Declared types are  gazebo_ros2_control/GazeboSystem`.
I found this to be odd since the base class of fake_components is definitely not `gazebo_ros2_control::GazeboSystemInterface`.

I followed it back to what I believe to be the root cause of the problem at lines 86-87 of [gazebo_ros2_control/src/gazebo_ros2_control_plugin.cpp](https://github.com/ros-simulation/gazebo_ros2_control/blob/c89584dcd4dcfe70a19b76ee57544a3e31e88926/gazebo_ros2_control/src/gazebo_ros2_control_plugin.cpp).
```
boost::shared_ptr<pluginlib::ClassLoader<
    gazebo_ros2_control::GazeboSystemInterface>> robot_hw_sim_loader_;
```
It appears to only load it's own plugin and then assume that all other plugins have the same base class, especially since the _only_ declared type it lists is `gazebo_ros2_control/GazeboSystem`.  This is about where my knowledge is completely exhausted so I'm not sure what a fix would be. This just results in not being able to run a simulation at the same time as having any actual hardware active which I'll admit is a niche requirement. 

The only thing I've found to be able to get around this is to just make sure that the only type of ros2_control plugin anywhere in the urdf is `gazebo_ros2_control/GazeboSystem`  otherwise it will crash gazebo. Maybe it's worth a fix? but if not hopefully this post will help someone else who was having the same problem as me. Apologies if this is missing something or doesn't justify a post, I haven't reported many issues in the past.

jclinton830, 2023-10-20:
Why was this closed? I am still having this issue in humble. Was a fix ever released?

christophfroehlich, 2023-10-21:
> Why was this closed? I am still having this issue in humble. Was a fix ever released?

This issue is still open, only the linked one was closed 5 years ago.

Can you elaborate your use-case for loading gazebo_ros2_control and a different hardware component at the same time?

jclinton830, 2023-11-13:
Apologies. You are right there is no use case for such a scenario. I was just parsing the `true` instead of `True` to my xacro file which triggered the default case `false` to be loaded and was causing this error. I then came to the conclusion that this may be a bug, but I was wrong.

christophfroehlich, 2024-01-11:
> I then came to the conclusion that this may be a bug, but I was wrong.

When I understand correctly, this issue can be closed? Feel free to reopen otherwise.

MarioCavero, 2024-04-10:
I am getting the same error. I have tried to minimize it in a simple example. I think it was working for me before, I installed via apt, maybe sourcing gazebo or gazebo-11 setup.bash? But i am encountering the same...
_my_gazebo_controller.launch.py_
```

import os
from launch_ros.actions import Node
from launch import LaunchDescription
from launch.substitutions import Command
from launch.actions import ExecuteProcess
from ament_index_python.packages import get_package_share_directory

def generate_launch_description():

	
	#robot_model = 'm1013' 

	xacro_file = get_package_share_directory('learm_ros2') + '/urdf/learm-gazebo.urdf.xacro'

	
	# Robot State Publisher 
	robot_state_publisher = Node(package    ='robot_state_publisher',
								 executable ='robot_state_publisher',
								 name       ='robot_state_publisher',
								 output     ='both',
								 parameters =[{'robot_description': Command(['xacro', ' ', xacro_file])           
								}])

	# Spawn the robot in Gazebo  
	spawn_entity_robot = Node(package     ='gazebo_ros', 
							  executable  ='spawn_entity.py', 
							  arguments   = ['-entity', 'learm_ros2', '-topic', 'robot_description'],
							  output      ='screen')

	# Gazebo   
	world_file_name = 'table_world.world'
	world = os.path.join(get_package_share_directory('learm_ros2'), 'worlds', world_file_name)
	gazebo_node = ExecuteProcess(cmd=['gazebo', '--verbose', world,'-s', 'libgazebo_ros_factory.so'], output='screen')

	# load and START the controllers in launch file
	
	# load_joint_state_broadcaster = ExecuteProcess(
	# 									cmd=['ros2', 'control', 'load_controller', '--set-state', 'start','joint_state_broadcaster'],
	# 									output='screen')

	
	# load_joint_trajectory_controller = ExecuteProcess( 
	# 									cmd=['ros2', 'control', 'load_controller', '--set-state', 'start', 'joint_trajectory_controller'], 
	# 									output='screen')

	return LaunchDescription([robot_state_publisher, spawn_entity_robot, gazebo_node,   ])
```

which logs:

```
ros2 launch learm_ros2 my_gazebo_controller.launch.py 
[INFO] [launch]: All log files can be found below /home/mk/.ros/log/2024-04-10-22-25-27-095701-mk-95463
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [robot_state_publisher-1]: process started with pid [95465]
[INFO] [spawn_entity.py-2]: process started with pid [95467]
[INFO] [gazebo-3]: process started with pid [95469]
[robot_state_publisher-1] [INFO] [1712780727.334350804] [robot_state_publisher]: got segment base_link
[robot_state_publisher-1] [INFO] [1712780727.334462250] [robot_state_publisher]: got segment finger_left_link
[robot_state_publisher-1] [INFO] [1712780727.334484575] [robot_state_publisher]: got segment finger_right_link
[robot_state_publisher-1] [INFO] [1712780727.334489681] [robot_state_publisher]: got segment forearm_link
[robot_state_publisher-1] [INFO] [1712780727.334494152] [robot_state_publisher]: got segment grip_left_link
[robot_state_publisher-1] [INFO] [1712780727.334498512] [robot_state_publisher]: got segment grip_right_link
[robot_state_publisher-1] [INFO] [1712780727.334502913] [robot_state_publisher]: got segment hand_link
[robot_state_publisher-1] [INFO] [1712780727.334507305] [robot_state_publisher]: got segment humerus_link
[robot_state_publisher-1] [INFO] [1712780727.334511699] [robot_state_publisher]: got segment shoulder_link
[... 255 lines cut ...]
    <plugin>gazebo_ros2_control/GazeboSystem</plugin>
  </hardware>
<joint name="shoulder_pan">
    <command_interface name="position">
```
    
    and long...
```
     </joint>
</ros2_control>

<gazebo>

    <plugin filename="libgazebo_ros2_control.so" name="gazebo_ros2_control">
      <!-- <robot_param>robot_description</robot_param> -->
      <!-- <robot_param_node>robot_state_publisher</robot_param_node> -->
      <!-- <parameters>$(find learm_ros2_description)/config/combined_controllers.yaml</parameters> -->
      <parameters>/home/mk/ros2_ws/src/learm_ros2_description/config/combined_controllers.yaml</parameters>
      <!-- <controller_manager_name>simulation_controller_manager</controller_manager_name> -->
    </plugin>
</gazebo>

</robot>

```

ipa-ych, 2024-04-15:
I have been encountering the similar problem recently (since apt update & upgrade), I am not able to launch projects using gazebo classic & ros2_controllers anymore, although they have been working fine previously. I am using 22.04 and Humble.

Here are the error logs when trying to launch a minimum cart_demo:
```
~$ ros2 launch gazebo_ros2_control_demos cart_example_position.launch.py 
[INFO] [launch]: All log files can be found below /home/nhg-yc/.ros/log/2024-04-15-17-22-38-134693-IPA-UBN1452-117287
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [gzserver-1]: process started with pid [117288]
[INFO] [gzclient-2]: process started with pid [117290]
[INFO] [robot_state_publisher-3]: process started with pid [117292]
[INFO] [spawn_entity.py-4]: process started with pid [117294]
[robot_state_publisher-3] [INFO] [1713194560.072430343] [robot_state_publisher]: got segment cart
[robot_state_publisher-3] [INFO] [1713194560.072521576] [robot_state_publisher]: got segment slideBar
[robot_state_publisher-3] [INFO] [1713194560.072530904] [robot_state_publisher]: got segment world
[spawn_entity.py-4] [INFO] [1713194560.832709465] [spawn_entity]: Spawn Entity started
[spawn_entity.py-4] [INFO] [1713194560.832983255] [spawn_entity]: Loading entity published on topic robot_description
[spawn_entity.py-4] /opt/ros/humble/local/lib/python3.10/dist-packages/rclpy/qos.py:307: UserWarning: DurabilityPolicy.RMW_QOS_POLICY_DURABILITY_TRANSIENT_LOCAL is deprecated. Use DurabilityPolicy.TRANSIENT_LOCAL instead.
[spawn_entity.py-4]   warnings.warn(
[spawn_entity.py-4] [INFO] [1713194560.835415526] [spawn_entity]: Waiting for entity xml on robot_description
[spawn_entity.py-4] [INFO] [1713194560.846123153] [spawn_entity]: Waiting for service /spawn_entity, timeout = 30
[spawn_entity.py-4] [INFO] [1713194560.846552335] [spawn_entity]: Waiting for service /spawn_entity
[spawn_entity.py-4] [INFO] [1713194561.351887864] [spawn_entity]: Calling service /spawn_entity
[spawn_entity.py-4] [INFO] [1713194561.517410720] [spawn_entity]: Spawn status: SpawnEntity: Successfully spawned entity [cart]
[gzserver-1] [INFO] [1713194561.557830376] [gazebo_ros2_control]: Loading gazebo_ros2_control plugin
[gzserver-1] [INFO] [1713194561.560710685] [gazebo_ros2_control]: Starting gazebo_ros2_control plugin in namespace: /
[gzserver-1] [INFO] [1713194561.560939448] [gazebo_ros2_control]: Starting gazebo_ros2_control plugin in ros 2 node: gazebo_ros2_control
[gzserver-1] [INFO] [1713194561.563337461] [gazebo_ros2_control]: connected to service!! robot_state_publisher
[gzserver-1] [INFO] [1713194561.565738936] [gazebo_ros2_control]: Received urdf from param server, parsing...
[gzserver-1] [INFO] [1713194561.566268822] [gazebo_ros2_control]: Loading parameter files /home/nhg-yc/ros2_ws/ur_ws/install/gazebo_ros2_control_demos/share/gazebo_ros2_control_demos/config/cart_controller.yaml
[gzserver-1] [ERROR] [1713194561.631520303] [gazebo_ros2_control]: The plugin failed to load for some reason. Error: According to the loaded plugin descriptions the class mock_components/GenericSystem with base class type gazebo_ros2_control::GazeboSystemInterface does not exist. Declared types are  gazebo_ros2_control/GazeboSystem
[gzserver-1] 
[gzserver-1] [INFO] [1713194561.631984354] [gazebo_ros2_control]: Loading controller_manager
[INFO] [spawn_entity.py-4]: process has finished cleanly [pid 117294]
[INFO] [ros2-5]: process started with pid [117420]
[gzserver-1] [WARN] [1713194561.706988302] [gazebo_ros2_control]:  Desired controller update period (0.01 s) is slower than the gazebo simulation period (0.001 s).
[gzserver-1] [INFO] [1713194561.708691831] [gazebo_ros2_control]: Loaded gazebo_ros2_control.
[gzclient-2] 
[gzclient-2] libcurl: (60) SSL: no alternative certificate subject name matches target host name 'fuel.ignitionrobotics.org'
[gzserver-1] [INFO] [1713194562.562039468] [controller_manager]: Loading controller 'joint_state_broadcaster'
[gzserver-1] [INFO] [1713194562.593730927] [controller_manager]: Configuring controller 'joint_state_broadcaster'
[gzserver-1] [INFO] [1713194562.594637738] [joint_state_broadcaster]: 'joints' or 'interfaces' parameter is empty. All available state interfaces will be published
[gzserver-1] [ERROR] [1713194562.632747249] [joint_state_broadcaster]: None of requested interfaces exist. Controller will not run.
[gzserver-1] [WARN] [1713194562.632807349] [joint_state_broadcaster]: Error occurred while doing error handling.
[gzserver-1] [ERROR] [1713194562.632855960] [controller_manager]: After activation, controller 'joint_state_broadcaster' is in state 'unconfigured' (1), expected 'active' (3).
[ros2-5] Successfully loaded controller joint_state_broadcaster into state active
[INFO] [ros2-5]: process has finished cleanly [pid 117420]
[INFO] [ros2-6]: process started with pid [117477]
[gzserver-1] [INFO] [1713194563.859181951] [controller_manager]: Loading controller 'joint_trajectory_controller'
[gzserver-1] [WARN] [1713194563.947196138] [joint_trajectory_controller]: [Deprecated]: "allow_nonzero_velocity_at_trajectory_end" is set to true. The default behavior will change to false.
[gzserver-1] [INFO] [1713194563.952167670] [controller_manager]: Configuring controller 'joint_trajectory_controller'
[gzserver-1] [INFO] [1713194563.952878892] [joint_trajectory_controller]: No specific joint names are used for command interfaces. Using 'joints' parameter.
[... cut ...]
[gzserver-1]  slider_to_cart/position
[gzserver-1] ]
[gzserver-1] 
[gzserver-1] [ERROR] [1713194564.008115692] [controller_manager]: Could not switch controllers since prepare command mode switch was rejected.
[ros2-6] Error activating controller, check controller_manager logs
```
**Error**: According to the loaded plugin descriptions the class mock_components/GenericSystem with base class type gazebo_ros2_control::GazeboSystemInterface does not exist. Declared types are  gazebo_ros2_control/GazeboSystem

christophfroehlich, 2024-04-15:
@ipa-ych Which versions have you installed now? (e.g., via `ros2 pkg xml ros2_control -t version` and `ros2 pkg xml gazebo_ros2_control -t version`

ipa-ych, 2024-04-15:
ros2 pkg xml ros2_control -t version
2.40.0
ros2 pkg xml gazebo_ros2_control -t version
0.4.6
Problem is now resolved by rebooting & re-install. But it occured early in the day after a simple update/ upgrade related to moveit

christophfroehlich, 2024-04-15:
This might be some issue with the pluginlib sourced in an old workspace?

Mystique03, 2024-12-03:
@ipa-ych Hi!! I am having the same issue (ros2 humble; 22.04; gazebo classic); how did you solve it? reinstall what?