ros-controls/gazebo_ros2_control #126: Getting librayLoadException when using ros2_control

muttistefano, 2022-05-18:
Hi all, im trying to use the ros2_control package to control multiple URs in gazebo.

So far I'm loading the ros2_control node in launch like this :
```
package="controller_manager",  
namespace="sweepee_1",  
executable="ros2_control_node",  
parameters=[robot_description_1,initial_joint_controllers_1 ],  
output="screen")
```
In the URDF the library is imported I such way:
```
<xacro:if value="$(arg sim_gazebo)">
    <gazebo reference="world"> 
    </gazebo>
    <gazebo>
      <plugin filename="libgazebo_ros2_control.so" name="gazebo_ros2_control">
        <ros>
          <namespace>/$(arg prefix)</namespace>
        </ros>
        <parameters>$(arg simulation_controllers)</parameters>
        <!-- <robot_param>robot_description</robot_param> -->
        <robot_param_node>robot_state_publisher</robot_param_node>
      </plugin>
    </gazebo>
  </xacro:if>
```
Where different nodes are under different namespaces, but when I launch it I get this:

```
[ros2_control_node-5] terminate called after throwing an instance of 'pluginlib::LibraryLoadException'
  [ros2_control_node-5]   what():  According to the loaded plugin descriptions the class gazebo_ros2_control/GazeboSystem with base class type hardware_interface::SystemInterface does not exist. Declared types are  fake_components/GenericSystem test_hardware_components/TestSystemCommandModes test_hardware_components/TestTwoJointSystem test_system ur_robot_driver/URPositionHardwareInterface
  [ERROR] [ros2_control_node-5]: process has died [pid 88542, exit code -6, cmd '/opt/ros/galactic/lib/controller_manager/ros2_control_node --ros-args -r __ns:=/sweepee_1 --params-file /tmp/launch_params_fe1c4ubr --params-file /home/kolmogorov/Documents/2ROS/lampo_ws_ros2/install/lampo_description/share/lampo_description/config/ur_controllers_1.yaml]
```

I cannot tell why this error comes from.
Thanks

saikrn112, 2022-09-25:
Hi, even I am facing the same issue. did you find any resolution for this?

bmagyar, 2022-09-26:
This is a pluginlib error guys which always happens when either
* you don't have the right plugins installed or built
* your environment setup is not correct
which results in the plugin you are looking for *not being available*. 

The error is pretty clear on this, please read it
```
  [ros2_control_node-5]   what():  According to the loaded plugin descriptions the class gazebo_ros2_control/GazeboSystem with base class type hardware_interface::SystemInterface does not exist. Declared types are  fake_components/GenericSystem test_hardware_components/TestSystemCommandModes test_hardware_components/TestTwoJointSystem test_system ur_robot_driver/URPositionHardwareInterface
```

closed 2022-09-26 by bmagyar