ros-controls/gazebo_ros2_control #131: joint_state_broadcaster configuration fails

christophfroehlich, 2022-05-30:
**Describe the bug**
I tried to use the humble or rolling releases and struggle with launching some robots. 
While `ros2 launch ros2_control_demo_bringup rrbot.launch.py` from [ros2_control_demos](https://github.com/ros-controls/ros2_control_demos.git) works, some others using gazebo don't, e.g. `ros2 launch ros2_control_demo_bringup rrbot_system_position_only_gazebo.launch.py` throws
> [gzserver-1] [INFO] [1653938781.009092923] [controller_manager]: Configuring controller 'joint_state_broadcaster'
> [gzserver-1] [INFO] [1653938781.010581373] [joint_state_broadcaster]: 'joints' or 'interfaces' parameter is empty. All available state interfaces will be published
> [gzserver-1] [ERROR] [1653938781.036334316] [joint_state_broadcaster]: None of requested interfaces exist. Controller will not run.
> [gzserver-1] [WARN] [1653938781.036404912] []: Error occurred while doing error handling.
> [gzserver-1] [ERROR] [1653938781.036445872] [controller_manager]: After activating, controller 'joint_state_broadcaster' is in state 'unconfigured', expected Active

Same error with the example `ros2 launch gazebo_ros2_control_demos cart_example_velocity.launch.py` from [gazebo_ros2_control_demos](https://github.com/ros-simulation/gazebo_ros2_control).

I can't find any differences in the `joint_state_broadcaster` configuration (yaml, launch-files, urdf). 
Is this linked to ros2_controllers, ros2_control, gazebo_ros2_control, or the example files?

**To Reproduce**
Steps to reproduce the behavior:
1. Install ros2-rolling on Ubuntu 22.04
2. Download [ros2_control_demos](https://github.com/ros-controls/ros2_control_demos.git) 
3. Install its dependencies with rosdep and compile it
4. Run `ros2 launch ros2_control_demo_bringup rrbot_system_position_only_gazebo.launch.py`
5. See error

**Environment:**
I manually compiled humble on Ubuntu 20.04 with the depencies found in the [release repositories for humble](https://github.com/ros2-gbp/ros2_control-release).
Additionally, I tried a freshly installed Ubuntu 22.04 with ros2-rolling (and humble, but there are some releases missing or rather old releases with the debians) with the same errors.

andrewlycas, 2022-06-02:
I ran into a very similar issue. The joint_state_broadcaster wasn't finding any of the state interfaces, so it wasn't running. Turns out now in humble the hardware interfaces now have state logic around them, so they need to be set in the active state (3). 

So your ros service call will be something like: 

`ros2 service call controller_manager/set_hardware_component_state controller_manager_msgs/srv/SetHardwareComponentState '{name: ''HardwareSystem", target_state: {id: 3, label: ""}}'`

I can look at the specific demo use case you have later to provide a more tailored solution, but thought this might be enough to get you started

christophfroehlich, 2022-06-09:
I tried to put that in the given launch files but was not able to get that running. Have you further worked on that?

destogl, 2022-06-28:
I will check this in the next few days and let you know what the real issue is.

samiamlabs, 2022-07-13:
I ran into this same issue while working on getting clearpath husky to work in simulation on humble.

#139 kind of resolves it @destogl, but it requires you set:

```xml
<ros2_control name="GazeboSimSystem" type="system">
```

All the demos have:

```xml
<ros2_control name="GazeboSystem" type="system">
```

So they don't work atm.

christophfroehlich, 2022-07-25:
Thanks for all the latest fixes and PRs, it works for me now!

closed 2022-07-25 by christophfroehlich