ros-controls/gazebo_ros2_control #141: Parser error Couldn't parse params file

EliasB23, 2022-07-15:
Hello, this is my first post so feel free to correct if I miss anything or ask for extra info. 
System Info: 
-SO: Ubuntu 20.04 (on VirtualBox)
-ROS 2 Foxy 

Error: 
While trying to load a simulation using Gazebo Classic (11), the log shows, between many other things, that it couldn't parse the ros_controller.yaml from de urdf (I checked it), I'm not doing anything different from the examples and yet it doesn't load. If I try loading a only the gazebo and spawn entity, it doesn't load the model, but if I load everything (gazebo+rviz+moveit+ros_control) it loads failing only to parse de .yaml. 

the last part of the .urdf:
`<ros2_control name="GazeboSystem" type="system">
      <hardware>
        <plugin>fake_components/GenericSystem</plugin>
      </hardware>
      <joint name="joint1_base">
        <command_interface name="position">
          <param name="min">0</param>
          <param name="max">6.28</param>
        </command_interface>
        <state_interface name="position">
          <param name="initial_value">1.0</param>
        </state_interface>
        <state_interface name="velocity"/>
      </joint>
      <joint name="joint2_arm1">
        <command_interface name="position">
          <param name="min">-1.57</param>
          <param name="max">1.57</param>
        </command_interface>
        <state_interface name="position">
          <param name="initial_value">1.0</param>
        </state_interface>
        <state_interface name="velocity"/>
      </joint>
      <joint name="joint3_arm2">
        <command_interface name="position">
          <param name="min">0</param>
          <param name="max">3.1415</param>
        </command_interface>
        <state_interface name="position">
          <param name="initial_value">1.0</param>
        </state_interface>
        <state_interface name="velocity"/>
      </joint>
      <joint name="joint4_arm3">
        <command_interface name="position">
          <param name="min">0</param>
          <param name="max">3.1415</param>
        </command_interface>
        <state_interface name="position">
          <param name="initial_value">1.0</param>
        </state_interface>
        <state_interface name="velocity"/>
      </joint>
      <joint name="joint5_end">
        <command_interface name="position">
          <param name="min">0</param>
          <param name="max">6.28</param>
        </command_interface>
        <state_interface name="position">
          <param name="initial_value">1.0</param>
        </state_interface>
        <state_interface name="velocity"/>
      </joint>
    </ros2_control>
    <gazebo>
     <plugin filename="libgazebo_ros2_control.so" name="gazebo">
      <robot_param>robot_description</robot_param>
      <robot_param_node>robot_state_publisher</robot_param_node>
      <parameters>$(find-pkg-prefix gpr_robot_moveit2_config)/config/ros_controllers.yaml</parameters>
     </plugin>
[... 343 lines cut ...]
[gazebo-5] 
[**gazebo-5] [ERROR] [1657909702.690854445] [gazebo]: parser error Couldn't parse params file: '--params-file $(find-pkg-prefix gpr_robot_moveit2_config)/config/ros_controllers.yaml'. Error: Error opening YAML file, at /tmp/binarydeb/ros-foxy-rcl-yaml-param-parser-1.1.13/src/parser.c:198, at /tmp/binarydeb/ros-foxy-rcl-1.1.13/src/rcl/arguments.c:388
[gazebo-5]** 
[INFO] [spawn_entity.py-6]: process has finished cleanly [pid 2009]
[rviz2-4] [INFO] [1657909703.142402978] [interactive_marker_display_94104070580928]: Connected on namespace: rviz_moveit_motion_planning_display/robot_interaction_interactive_marker_topic
[rviz2-4] Link link0_base had 1 children
[rviz2-4] Link link1_base had 1 children
[rviz2-4] Link link2_arm1 had 1 children
[rviz2-4] Link link3_arm2 had 1 children
[rviz2-4] Link link4_arm3 had 1 children
[rviz2-4] Link link5_end had 0 children
[rviz2-4] [INFO] [1657909703.158824421] [moveit_ros_visualization.motion_planning_frame]: group arm_manipulator
[rviz2-4] [INFO] [1657909703.158865134] [moveit_ros_visualization.motion_planning_frame]: Constructing new MoveGroup connection for group 'arm_manipulator' in namespace ''
[rviz2-4] [WARN] [1657909703.160005911] [rcl.logging_rosout]: Publisher already registered for provided node name. If this is due to multiple nodes with the same name then all logs for that logger name will go out over the existing publisher. As soon as any node with that name is destructed it will unregister the publisher, preventing any further logs for that name from being published on the rosout topic.
[rviz2-4] [INFO] [1657909703.214881471] [move_group_interface]: Ready to take commands for planning group arm_manipulator.
[rviz2-4] [INFO] [1657909703.215268484] [move_group_interface]: Looking around: no
[rviz2-4] [INFO] [1657909703.215293156] [move_group_interface]: Replanning: no
[rviz2-4] [INFO] [1657909703.311747570] [interactive_marker_display_94104070580928]: Sending request for interactive markers
[rviz2-4] [INFO] [1657909703.317053376] [moveit_ros_visualization.motion_planning_frame_planning]: POPULATING PLANNERS 1 grp: arm_manipulator
[rviz2-4] [INFO] [1657909703.318089862] [moveit_ros_visualization.motion_planning_frame_planning]: planner id: CHOMP
[rviz2-4] [INFO] [1657909703.483860484] [interactive_marker_display_94104070580928]: Service response received for initialization
`

If anyone has an idea please let me know. I'm aware that Foxy is not being maintained and I'm trying porting this to humble but it will take me a little more time, and would like to have this working too. 
Thanks in advance!

bmagyar, 2022-07-18:
I'm sorry but this is not really readable.

First off, I think you have better chances for getting help at https://answers.ros.org/questions, try to always post there first.
Second, try attaching files instead of copying them into a page that tries to format them according to the hieroglyphs it finds in them, that helps readability :D You should also post the commands you used to reproduce the problem to help others see what's up. (You may have done it but it got lost in the text for me)

closed 2022-07-18 by bmagyar

EliasB23, 2022-07-18:
I didn't know about attaching files, sorry about that! I will post it in there, thanks!