ros-controls/gazebo_ros2_control #148: Writing my own controller

mohseninima, 2022-07-29:
I can't seem to find any sort of minimal example on writing my own controller and having it interface with gazebo_ros2_control.

Documentation is very sparse and I have been spending 2 weeks just trying to write my own proportional controller to put a torque on a simple arm in Gazebo.

I just want to take my current position error and output a torque to a specific joint based on a gain that is given in the yaml.

How do I do this? How do I write my control action to the joint? How do I send a position command to the controller?

bmagyar, 2022-07-30:
Have you checked control.ros.org ?

destogl, 2022-07-30:
In one of the repositoriesin ros2_control and ros2_controllers is a very, very detailed doc how to do that with exact individual steps.

mohseninima, 2022-07-30:
I saw the documentation [https://control.ros.org/master/doc/ros2_controllers/doc/writing_new_controller.html](here).

There is nothing on:
Connecting my controller to Gazebo using gazebo_ros2_control
Setting up actuators in gazeob_ros2_control
How to tell ros2_control that my control output is an effort
How to read the joint position data
How to output my control effort to each joint
How to read data from the yaml
How to take in a command using a ros node
What the purpose is of each function: "init, command_interface_configuration, state_interface_configuration, on_configure, on_activate, on_deactivate, update"

destogl, 2022-08-03:
> I saw the documentation [here](https://control.ros.org/master/doc/ros2_controllers/doc/writing_new_controller.html).
>
> There is nothing on: Connecting my controller to Gazebo using [...]

Have you checked hardware examples in ros2_control_demos repository? Those all questions are not related to controller but to hardware and interaction with controllers that is shown in the first examples... Did you checked examples in this repository?

How to take in a command using a ros node What the purpose is of each function: "init, command_interface_configuration, state_interface_configuration, on_configure, on_activate, on_deactivate, update"

Have you check doxygen documentaton of `controller_interface`?

Have you checked all the presentation/videos and arch diagrams we have in the [resources](https://control.ros.org/master/doc/resources/resources.html) page?

I will close this because it is not an issue with this repository at all. Feel free to continue commenting/answering to it, nevertheless.

closed 2022-08-03 by destogl

bmagyar, 2022-08-03:
Better yet, go to https://answers.ros.org/ instead