ros-controls/gazebo_ros2_control #152: Porting PID controllers to ROS2 controllers

Alex-Zongo, 2022-09-24:
Hi guys... I am currently working on a humanoid control project with ROS2. The problem I faces, is that the robot is unable to keep it's starting position and it slips and move by itself on its back. Compared to the implementation in ROS Noetic with PID controllers which by the way is working fine, the only difference was the pid parameters missing in ROS2 controllers.

Is there anyone working on porting the PID controllers from ROS1? or anyone working on bipedal walking with ROS2 who found a way around that kind of problem?

bmagyar, 2022-09-25:
I can only say that whatever is out there on a PR exists. What's not on a
PR, does not exist as far as we are concerned.

You are very welcome to provide a PR with PID controllers in ROS2

Alex-Zongo, 2022-10-03:
@bmagyar Thanks. I will check it out.

closed 2023-05-09 by christophfroehlich