ros-controls/gazebo_ros2_control #155: parser error Couldn't parse params file

padhupradheep, 2022-10-28:
Working on Humble

I was trying to utilize the wonderful gazebo plugin from this package, but I found when you have to link the config yaml with all the controller params, you need to specify the absolute path. For some reason the plugin cannot parse the relative path. Like for example `$(find xyz_package)/configs/your_controller.yaml`. 

I also tried setting the path to a xacro property and assigning it. But for some reason none of the xacro commands are working especially in the `parameters` line. Somehow the gazebo_ros2_control plugin does not account for the xacro commands, rather just consider it as a string. 

``` <xacro:property name="yaml_file" value="$(find neo_simulation2)/configs/arms/joint_names_ec66_description.yaml" />
    <gazebo>
        <plugin filename="libgazebo_ros2_control.so" name="gazebo_ros2_control">
           <parameters> ${yaml_file}</parameters>
        </plugin>
    </gazebo> ```

This is an issue somewhere originating with this line https://github.com/ros-controls/gazebo_ros2_control/blob/f64e56af4800120150460ffba15cd73db511521c/gazebo_ros2_control/src/gazebo_ros2_control_plugin.cpp#L239 

Maybe it's just [rcl_parse_arguments()](http://docs.ros.org/en/humble/p/rcl/generated/function_arguments_8h_1a193be7478d13952ed6f551143307d6a9.html?highlight=rcl_parse_arguments#_CPPv419rcl_parse_argumentsiPPCKc15rcl_allocator_tP15rcl_arguments_t) function.  

I apologize if I'm doing something wrong whille assigning the parameters.

This issue is close to https://github.com/ros-controls/gazebo_ros2_control/issues/141

christophfroehlich, 2022-11-20:
Do I understand that correctly, you are trying something like this
https://github.com/ros-controls/gazebo_ros2_control/blob/370bc756674f5f76fbcbe41b1995b2233705e49a/gazebo_ros2_control_demos/urdf/test_cart_position.xacro.urdf#L68-L72
and it does not work? What is the exact error message?

Have you tried running `xacro` on your xacro file?
```bash
xacro your_file.xacro > result.urdf
```
and the path should be expanded correctly.

padhupradheep, 2022-11-20:
> and it does not work? 

Exactly

> What is the exact error message?

This is the exact line where the error occurs 

https://github.com/ros-controls/gazebo_ros2_control/blob/370bc756674f5f76fbcbe41b1995b2233705e49a/gazebo_ros2_control/src/gazebo_ros2_control_plugin.cpp#L250

> Have you tried running xacro on your xacro file?

Yes, as I said above:

>  I also tried setting the path to a xacro property and assigning it. But for some reason none of the xacro commands are working especially in the parameters line. Somehow the gazebo_ros2_control plugin does not account for the xacro commands, rather just consider it as a string.

christophfroehlich, 2022-11-20:
- This plugin itself does not parse xacro commands, you have to run xacro to create a proper urdf before you pass it to this plugin
https://github.com/ros-controls/gazebo_ros2_control/blob/370bc756674f5f76fbcbe41b1995b2233705e49a/gazebo_ros2_control_demos/launch/cart_example_position.launch.py#L36-L45
- Is the demo from this repository working?
- Back to your example with `$(find package)`: Is the path expanded correctly in the `result.urdf`, if you run xacro as I suggested above?

padhupradheep, 2022-11-20:
Ah well, I get your point now.. I also had a URDF file and I thought I processed the Xacro by doing something like this, whilst launching the robot_state_publisher:

```
start_robot_state_publisher_cmd = Node(
        package='robot_state_publisher',
        executable='robot_state_publisher',
        name='robot_state_publisher',
        output='screen',
        parameters=[{'robot_description':Command([
            "xacro", " ", urdf]), 'use_sim_time': use_sim_time}]
        )
```

Seems that I need to use the `doc.toxml()`..

Thank you very much for the nice solution.. Really appreciate your time..

closed 2022-11-20 by padhupradheep