ros-controls/gazebo_ros2_control #170: gazebo_ros2_control build failures on Rolling

nuclearsandwich, 2023-01-04:
ROS distro: Rolling
OS version: Ubuntu 22.04 (Jammy)
gazebo_ros2_control version: 0.4.0-1

Link to failing build: https://build.ros2.org/view/Rbin_uJ64/job/Rbin_uJ64__gazebo_ros2_control__ubuntu_jammy_amd64__binary/100/

Build log excerpt:
```
00:07:19.450 /usr/lib/ccache/c++ -DBOOST_ALL_NO_LIB -DBOOST_ATOMIC_DYN_LINK -DBOOST_DATE_TIME_DYN_LINK -DBOOST_FILESYSTEM_DYN_LINK -DBOOST_IOSTREAMS_DYN_LINK -DBOOST_PROGRAM_OPTIONS_DYN_LINK -DBOOST_REGEX_DYN_LINK -DBOOST_SYSTEM_DYN_LINK -DBOOST_TEST_DYN_LINK -DBOOST_THREAD_DYN_LINK -DDEFAULT_RMW_IMPLEMENTATION=rmw_fastrtps_cpp -DLIBBULLET_VERSION=3.05 -DLIBBULLET_VERSION_GT_282 -DRCUTILS_ENABLE_FAULT_INJECTION -Dgazebo_ros2_control_EXPORTS -I/tmp/binarydeb/ros-rolling-gazebo-ros2-control-0.4.0/include -I/opt/ros/rolling/include -I/opt/ros/rolling/include/rclcpp_lifecycle -I/opt/ros/rolling/include/rclcpp_action -I/usr/include/gazebo-11 -I/usr/include/bullet -I/usr/include/sdformat-9.7 -I/usr/include/OGRE -I/usr/include/OGRE/Terrain -I/usr/include/OGRE/Paging -isystem /opt/ros/rolling/include/angles -isystem /opt/ros/rolling/include/pluginlib -isystem /opt/ros/rolling/include/rclcpp -isystem /opt/ros/rolling/include/ament_index_cpp -isystem /opt/ros/rolling/include/control_msgs -isystem /opt/ros/rolling/include/lifecycle_msgs -isystem /opt/ros/rolling/include/rcpputils -isystem /opt/ros/rolling/include/rcutils -isystem /opt/ros/rolling/include/sensor_msgs -isystem /opt/ros/rolling/include/controller_manager_msgs -isystem /opt/ros/rolling/include/diagnostic_msgs -isystem /opt/ros/rolling/include/std_msgs -isystem /usr/include/simbody -isystem /usr/include/ignition/math6 -isystem /usr/include/ignition/transport8 -isystem /usr/include/ignition/msgs5 -isystem /usr/include/ignition/common3 -isystem /usr/include/ignition/fuel_tools4 -isystem /opt/ros/rolling/include/class_loader -isystem /opt/ros/rolling/include/libstatistics_collector -isystem /opt/ros/rolling/include/builtin_interfaces -isystem /opt/ros/rolling/include/rosidl_runtime_c -isystem /opt/ros/rolling/include/rosidl_typesupport_interface -isystem /opt/ros/rolling/include/fastcdr -isystem /opt/ros/rolling/include/rosidl_runtime_cpp -isystem /opt/ros/rolling/include/rosidl_typesupport_fastrtps_cpp -isystem /opt/ros/rolling/include/rmw -isystem /opt/ros/rolling/include/rosidl_typesupport_fastrtps_c -isystem /opt/ros/rolling/include/rosidl_typesupport_introspection_c -isystem /opt/ros/rolling/include/rosidl_typesupport_introspection_cpp -isystem /opt/ros/rolling/include/rcl -isystem /opt/ros/rolling/include/rcl_interfaces -isystem /opt/ros/rolling/include/rcl_logging_interface -isystem /opt/ros/rolling/include/rcl_yaml_param_parser -isystem /opt/ros/rolling/include/libyaml_vendor -isystem /opt/ros/rolling/include/tracetools -isystem /opt/ros/rolling/include/statistics_msgs -isystem /opt/ros/rolling/include/rosgraph_msgs -isystem /opt/ros/rolling/include/rosidl_typesupport_cpp -isystem /opt/ros/rolling/include/rosidl_typesupport_c -isystem /opt/ros/rolling/include/rcl_action -isystem /opt/ros/rolling/include/action_msgs -isystem /opt/ros/rolling/include/unique_identifier_msgs -isystem /usr/include/eigen3 -isystem /usr/include/sdformat-9.7/sdf/.. -isystem /usr/include/ignition/cmake2 -isystem /usr/include/uuid -isystem /opt/ros/rolling/include/geometry_msgs -isystem /opt/ros/rolling/include/trajectory_msgs -isystem /opt/ros/rolling/include/rcl_lifecycle -g -O2 -ffile-prefix-map=/tmp/binarydeb/ros-rolling-gazebo-ros2-control-0.4.0=. -flto=auto -ffat-lto-objects -flto=auto -ffat-lto-objects -fstack-protector-strong -Wformat -Werror=format-security -DNDEBUG -Wdate-time -D_FORTIFY_SOURCE=2  -fPIC -Wall -Wextra -Wconversion -Wno-sign-conversion -Wpedantic -Wnon-virtual-dtor -Woverloaded-virtual -I/usr/include/uuid -MD -MT CMakeFiles/gazebo_ros2_control.dir/src/gazebo_ros2_control_plugin.cpp.o -MF CMakeFiles/gazebo_ros2_control.dir/src/gazebo_ros2_control_plugin.cpp.o.d -o CMakeFiles/gazebo_ros2_control.dir/src/gazebo_ros2_control_plugin.cpp.o -c /tmp/binarydeb/ros-rolling-gazebo-ros2-control-0.4.0/src/gazebo_ros2_control_plugin.cpp
00:07:44.577 In file included from /opt/ros/rolling/include/controller_manager/controller_manager.hpp:43,
00:07:44.577                  from /tmp/binarydeb/ros-rolling-gazebo-ros2-control-0.4.0/include/gazebo_ros2_control/gazebo_ros2_control_plugin.hpp:42,
00:07:44.577                  from /tmp/binarydeb/ros-rolling-gazebo-ros2-control-0.4.0/src/gazebo_ros2_control_plugin.cpp:42:
00:07:44.577 /opt/ros/rolling/include/diagnostic_updater/diagnostic_updater.hpp: In constructor ‘diagnostic_updater::Updater::Updater(std::shared_ptr<rclcpp::node_interfaces::NodeBaseInterface>, std::shared_ptr<rclcpp::node_interfaces::NodeClockInterface>, std::shared_ptr<rclcpp::node_interfaces::NodeLoggingInterface>, std::shared_ptr<rclcpp::node_interfaces::NodeParametersInterface>, std::shared_ptr<rclcpp::node_interfaces::NodeTimersInterface>, std::shared_ptr<rclcpp::node_interfaces::NodeTopicsInterface>, double)’:
00:07:44.577 /opt/ros/rolling/include/diagnostic_updater/diagnostic_updater.hpp:389:55: warning: conversion from ‘double’ to ‘rcl_duration_value_t’ {aka ‘long int’} may change value [-Wfloat-conversion]
00:07:44.577   389 |     period_(rclcpp::Duration::from_nanoseconds(period * 1e9)),
00:07:44.577       |                                                ~~~~~~~^~~~~
00:07:44.577 /opt/ros/rolling/include/diagnostic_updater/diagnostic_updater.hpp:399:57: warning: conversion from ‘double’ to ‘rcl_duration_value_t’ {aka ‘long int’} may change value [-Wfloat-conversion]
00:07:44.577   399 |     period_ = rclcpp::Duration::from_nanoseconds(period * 1e9);
00:07:44.577       |                                                  ~~~~~~~^~~~~
00:07:44.577 /opt/ros/rolling/include/diagnostic_updater/diagnostic_updater.hpp: In member function ‘void diagnostic_updater::Updater::setPeriod(double)’:
00:07:44.577 /opt/ros/rolling/include/diagnostic_updater/diagnostic_updater.hpp:423:57: warning: conversion from ‘double’ to ‘rcl_duration_value_t’ {aka ‘long int’} may change value [-Wfloat-conversion]
00:07:44.577   423 |     setPeriod(rclcpp::Duration::from_nanoseconds(period * 1e9));
00:07:44.577       |                                                  ~~~~~~~^~~~~
00:07:44.577 /opt/ros/rolling/include/diagnostic_updater/diagnostic_updater.hpp: In member function ‘void diagnostic_updater::Updater::broadcast(int, std::string)’:
00:07:44.577 /opt/ros/rolling/include/diagnostic_updater/diagnostic_updater.hpp:456:22: warning: conversion from ‘int’ to ‘unsigned char’ may change value [-Wconversion]
00:07:44.577   456 |       status.summary(lvl, msg);
00:07:44.577       |                      ^~~
00:07:44.577 /tmp/binarydeb/ros-rolling-gazebo-ros2-control-0.4.0/src/gazebo_ros2_control_plugin.cpp: In member function ‘virtual void gazebo_ros2_control::GazeboRosControlPlugin::Load(gazebo::physics::ModelPtr, sdf::v9::ElementPtr)’:
00:07:44.577 /tmp/binarydeb/ros-rolling-gazebo-ros2-control-0.4.0/src/gazebo_ros2_control_plugin.cpp:288:67: error: ‘__gnu_cxx::__alloc_traits<std::allocator<hardware_interface::HardwareInfo>, hardware_interface::HardwareInfo>::value_type’ {aka ‘struct hardware_interface::HardwareInfo’} has no member named ‘hardware_class_type’
00:07:44.577   288 |     std::string robot_hw_sim_type_str_ = control_hardware_info[i].hardware_class_type;
00:07:44.577       |                                                                   ^~~~~~~~~~~~~~~~~~~
```

destogl, 2023-01-05:
This is fixed in #169. We have to trigger a new release. @bmagyar can you do that?

destogl, 2023-01-05:
closing for now.

closed 2023-01-05 by destogl