ros-controls/gazebo_ros2_control #173: "_mimic" joint interface name suffix causes Moveit2 "Joint not found" Error

shonigmann, 2023-01-11:
Is the mimic joint suffix that gets added to interface names ([here](https://github.com/ros-controls/gazebo_ros2_control/blob/6718ae9e60a43befc816968a2a1aa05395fc691b/gazebo_ros2_control/src/gazebo_system.cpp#L199)) necessary? If so, what is its intent? I can't find anything in the `ros2_control` source code that is expecting a mimic suffix, but maybe I'm not looking in the right place. 

I ask because I was trying to control a robot that had mimic joints with Moveit, simulated in gazebo, the Moveit RobotModel implementation complains loudly that `Joint 'joint_name_mimic' not found in model 'model_name'`, seemingly because the `GazeboSystem` `hardware_interface` assigns a new name to the joint that doesn't exist in the URDF/SRDF/yaml config files. It took me longer than I'd like to admit that the source of the error was `gazebo_ros2_control`'s change and not a mistake that I had made in one of the description files. 

Commenting out the ["_mimic" suffix]((https://github.com/ros-controls/gazebo_ros2_control/blob/6718ae9e60a43befc816968a2a1aa05395fc691b/gazebo_ros2_control/src/gazebo_system.cpp#L199)) doesn't appear to break anything, and allows Moveit to function as expected. 

Thanks in advance for any feedback/insights!

miso037, 2023-05-05:
Hi i have the same issue, although it is a dumb fix it works for suppressing Move_group error. You can add the following lines to your robot.urdf generated by MoveIt.

<joint name="panda_finger_joint2_mimic" type="fixed">
    <parent link="world" />
    <child link="random" />
  </joint>
  <link name="random"/>

In my case, the "panda_finger_joint2" is the mimicked joint that MoveIt moves.

Wiktor-99, 2023-11-30:
Today I faced same issue. Adding this suffix implicitly it's not intuitive. I removed adding this suffix from plugin and then launched simulation, everything seems to work just fine (gazebo, rviz and moveit). Is it necessary? 

If it is, some logs should be printed to make it more explicite.

PR #297 (2024-04-23): Rewrite mimic joints
The component_parser of the resource_manager now already parses the mimic joint information from the URDF. The old way for defining it in the `ros2_control` tag is deprecated, see https://github.com/ros-controls/ros2_control/pull/1256.

If this one is approved, I'll make the same changes for gz_ros2_control as well.

- This PR uses the info from the `hardware_info` structure instead of parsing it again.
- From my understanding, mimic means "mimic position", but the [URDF](http://wiki.ros.org/urdf/XML/joint) spec is not very clear in this point. So I implemented it that ind [... cut]
closed 2024-04-23 by ahcorde