ros-controls/gazebo_ros2_control #185: Proposal: GPIO tags support, mapped to gazebo::Transport topics

roncapat, 2023-03-13:
This is a proposal (not sure if it's the right place for that) about adding to `gazebo_ros2_control` the possibility to read/write Gazebo topics. For example, in an use-case of mine, I needed an elegant way for my controller to set some kind of value to the (simulated) hardware, so to a Gazebo plugin. I checked the Gazebo API and found out that apart from introspection (read-only way to communicate) the other common solution (found in the PX4 SITL packages for example) is to bridge plugins via gazebo messages (Google protobufs).

Any interest in a function like that? Since I'm working on it already, it might be possible for me to provide a clean implementation (with some extra work).

ahcorde, 2023-03-14:
Happy to review the PR, I would suggest to include a tutorial or example too.

roncapat, 2023-03-18:
A possible work rationale here is to:
- 1 - interpret every command or state interface named differently from "position", "velocity", "effort" as write/read (pub/sub) proxies to gazebo topics.
- 2 - extend behavior to provide compatibility not only with `double` datatype but also at least `int` and `bool` (gazebo "Any" messages support different basic data types).
- 3 - introduce support to GPIOs (leveraging facilities developed in point 1 and 2)

Other desiderata?