ros-controls/gazebo_ros2_control #186: Subscriber in controller configuration

muttistefano, 2023-03-17:
Hi all, 
I have a controller that needs to read the robot description topic during the configuration, and waits in configuration until the topic is received.
When i use it with the real robot and the ros controller manager, everything works well.
When i use it with gazebo through the gazebo ros control the topic subsciber wont receive the topic and so it hangs, waiting forever.
It seems like the callback is not take care of.
I'm not sure the source of this comes from this package or gazebo itself.
Thanks

christophfroehlich, 2023-03-27:
Hi. I'm using a similar structure as you are describing, so this should work in principle.
To dig into this you have to give us some more information, maybe you can upload or give us access to your launch file + URDF?

muttistefano, 2024-01-19:
I close the issue since i passed to GZ.
Thanks

closed 2024-01-19 by muttistefano