ros-controls/gazebo_ros2_control #188: [Err] [Plugin.hh:212] Failed to load plugin libgazebo_ros_control.so: libgazebo_ros_control.so: cannot open shared object file: No suchfile or directory

anand-010, 2023-03-25:
I am following this guide from articulated robotics https://articulatedrobotics.xyz/mobile-robot-12-ros2-control/.
i've already installed.
sudo apt install ros-humble-ros2-control ros-humble-ros2-controllers ros-humble-gazebo-ros2-control.
but during launch i've got this error.

`[Err] [Plugin.hh:212] Failed to load plugin libgazebo_ros_control.so: libgazebo_ros_control.so: cannot open shared object file: No suchfile or directory.`
and
`[Err] [Model.cc:1160] Exception occured in the Load function of plugin with name[gazebo_ros2_control] and filename[libgazebo_ros2_control.so]. This plugin will not run.`
i've checked the file using the command 
`ls /usr/lib/x86_64-linux-gnu/gazebo-11/plugins | grep libgazebo_ros2_control.so`
but that file not found in plugin directory.
Is this not available for ros2 humble?.
how do i solve it.
where can i found this file to install it manually?.

anand-010, 2023-03-25:
log.
```
Warning [parser_urdf.cc:3027] urdf2sdf: revolute joint [Tip_Gripper_Idol_Joint] with limits: lowStop[3.14] > highStop[-3.14], switching the two.
[INFO] [1679715236.387070533] [gazebo_ros2_control]: Loading gazebo_ros2_control plugin
[INFO] [1679715236.404079925] [gazebo_ros2_control]: Starting gazebo_ros2_control plugin in namespace: /
[INFO] [1679715236.405457275] [gazebo_ros2_control]: Starting gazebo_ros2_control plugin in ros 2 node: gazebo_ros2_control
[INFO] [1679715236.405979906] [gazebo_ros2_control]: Loading parameter file /home/user/ros2_moveo/install/moveo_moveit_config/share/moveo_moveit_config/config/ros2_controllers.yaml

[INFO] [1679715236.416286275] [gazebo_ros2_control]: connected to service!! robot_state_publisher
[INFO] [1679715296.511643774] [gazebo_ros2_control]: Recieved urdf from param server, parsing...
[INFO] [1679715296.593671949] [gazebo_ros2_control]: Loading joint: Joint_1
[INFO] [1679715296.593759480] [gazebo_ros2_control]:    State:
[INFO] [1679715296.594361962] [gazebo_ros2_control]:             position
[INFO] [1679715296.594404703] [gazebo_ros2_control]:                     found initial value: 0.000000
[INFO] [1679715296.594470460] [gazebo_ros2_control]:             velocity
[INFO] [1679715296.594500998] [gazebo_ros2_control]:    Command:
[INFO] [1679715296.594520993] [gazebo_ros2_control]:             position
[INFO] [1679715296.594601784] [gazebo_ros2_control]: Loading joint: Joint_2
[INFO] [1679715296.594626381] [gazebo_ros2_control]:    State:
[INFO] [1679715296.594638847] [gazebo_ros2_control]:             position
[INFO] [1679715296.594650680] [gazebo_ros2_control]:                     found initial value: 0.000000
[INFO] [1679715296.594664901] [gazebo_ros2_control]:             velocity
[INFO] [1679715296.594676526] [gazebo_ros2_control]:    Command:
[INFO] [1679715296.594692974] [gazebo_ros2_control]:             position
[INFO] [1679715296.594747487] [gazebo_ros2_control]: Loading joint: Joint_3
[INFO] [1679715296.594783566] [gazebo_ros2_control]:    State:
[INFO] [1679715296.594816336] [gazebo_ros2_control]:             position
[INFO] [1679715296.594849622] [gazebo_ros2_control]:                     found initial value: 0.000000
[INFO] [1679715296.594886816] [gazebo_ros2_control]:             velocity
[INFO] [1679715296.594921471] [gazebo_ros2_control]:    Command:
[INFO] [1679715296.594955570] [gazebo_ros2_control]:             position
[INFO] [1679715296.594996854] [gazebo_ros2_control]: Loading joint: Joint_4
[INFO] [1679715296.595030232] [gazebo_ros2_control]:    State:
[INFO] [1679715296.595057229] [gazebo_ros2_control]:             position
[INFO] [1679715296.595357483] [gazebo_ros2_control]:                     found initial value: 0.000000
[INFO] [1679715296.595393695] [gazebo_ros2_control]:             velocity
[INFO] [1679715296.595415972] [gazebo_ros2_control]:    Command:
[INFO] [1679715296.595433839] [gazebo_ros2_control]:             position
[INFO] [1679715296.595468058] [gazebo_ros2_control]: Loading joint: Joint_5
[INFO] [1679715296.595491367] [gazebo_ros2_control]:    State:
[INFO] [1679715296.595505004] [gazebo_ros2_control]:             position
[INFO] [1679715296.595551399] [gazebo_ros2_control]:                     found initial value: 0.000000
[INFO] [1679715296.595609503] [gazebo_ros2_control]:             velocity
[INFO] [1679715296.595657636] [gazebo_ros2_control]:    Command:
[INFO] [1679715296.595756073] [gazebo_ros2_control]:             position
[INFO] [1679715296.595945970] [resource_manager]: Initialize hardware 'GazeboSystem'
[INFO] [1679715296.596234585] [resource_manager]: Successful initialization of hardware 'GazeboSystem'
[INFO] [1679715296.596544645] [resource_manager]: 'configure' hardware 'GazeboSystem'
[INFO] [1679715296.596634922] [resource_manager]: Successful 'configure' of hardware 'GazeboSystem'
[INFO] [1679715296.596668183] [resource_manager]: 'activate' hardware 'GazeboSystem'
[INFO] [1679715296.596691468] [resource_manager]: Successful 'activate' of hardware 'GazeboSystem'
[Err] [Model.cc:1160] Exception occured in the Load function of plugin with name[gazebo_ros2_control] and filename[libgazebo_ros2_control.so]. This plugin will not run.
[Wrn] [Publisher.cc:135] Queue limit reached for topic /gazebo/default/user_camera/pose, deleting message. This warning is printed only once.
```

anand-010, 2023-03-25:
Tried to install from source but got error.
```
/home/user/ros3_ws/src/ /gazebo_ros2_control/src/gazebo_ros2_control_plugin.cpp: In member function 'virtual void gazebo_ros2_control::GazeboRosControlPlugin::Load(gazebo::physics::ModelPtr, sdf::v9::ElementPtr)':
/home/user/ros3_ws/src/ /gazebo_ros2_control/src/gazebo_ros2_control_plugin.cpp:290:67: error: '__gnu_cxx::__alloc_traits<std::allocator<hardware_interface::HardwareInfo>, hardware_interface::HardwareInfo>::value_type' {aka 'struct hardware_interface::HardwareInfo'} has no member named 'hardware_plugin_name'
  290 |     std::string robot_hw_sim_type_str_ = control_hardware_info[i].hardware_plugin_name;
      |                                                                   ^~~~~~~~~~~~~~~~~~~~
gmake[2]: *** [CMakeFiles/gazebo_ros2_control.dir/build.make:76: CMakeFiles/gazebo_ros2_control.dir/src/gazebo_ros2_control_plugin.cpp.o] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:139: CMakeFiles/gazebo_ros2_control.dir/all] Error 2
gmake: *** [Makefile:146: all] Error 2
---
Failed   <<< gazebo_ros2_control [2min 0s, exited with code 2]
```

bmagyar, 2023-03-25:
Well something is definitely off as you should not be seeing "[Err] [Plugin.hh:212] Failed to load plugin libgazebo_ros_control.so:" in this context at all. 

I suggest talking to whoever wrote that tutorial.

closed 2023-04-05 by anand-010