ros-controls/gazebo_ros2_control #191: protobuf version

psychedelic-coder, 2023-04-10:
Hi, in foxy version,  I have tried 3.17.3 and 3.15.8 but both versions can't work and generated errors as below. I wonder I have used wrong version, what's the version suit for foxy version?
And the errors are as below:
from /home/liuchen/ROS/ros2_ws/src/gazebo_ros2_control/gazebo_ros2_control/src/gazebo_ros2_control_plugin.cpp:42:
/usr/include/gazebo-11/gazebo/msgs/wrench_stamped.pb.h:203:33: error: ‘InternalMetadataWithArena’ in namespace ‘google::protobuf::internal’ does not name a type; did you mean ‘InternalMetadata’?
  203 |   ::google::protobuf::internal::InternalMetadataWithArena _internal_metadata_;
      |                                 ^~~~~~~~~~~~~~~~~~~~~~~~~
      |                                 InternalMetadata
/usr/include/gazebo-11/gazebo/msgs/wrench_stamped.pb.h:141:8: error: ‘bool gazebo::msgs::WrenchStamped::MergePartialFromCodedStream(google::protobuf::io::CodedInputStream*)’ marked ‘final’, but is not virtual
  141 |   bool MergePartialFromCodedStream(
      |        ^~~~~~~~~~~~~~~~~~~~~~~~~~~
/usr/include/gazebo-11/gazebo/msgs/wrench_stamped.pb.h:143:8: error: ‘void gazebo::msgs::WrenchStamped::SerializeWithCachedSizes(google::protobuf::io::CodedOutputStream*) const’ marked ‘final’, but is not virtual

bmagyar, 2023-04-10:
Can you please give us a simple example to reproduce the issue? You can try with ros2_control_demos so simplify things

psychedelic-coder, 2023-04-11:
> Can you please give us a simple example to reproduce the issue? You can try with ros2_control_demos so simplify things

Sorry I didn't describe issue clearly. I just cd my_workspace and then colcon build and the output are as follows:
[image]
I found that my gazebo version is gazebo11, maybe it can't work with foxy project?
Thanks a lot!

psychedelic-coder, 2023-04-11:
I have solved this issue by removing protobuf I installed mannually and then reinstall probuf:
sudo apt reinstall libprotobuf-dev

closed 2023-04-11 by psychedelic-coder

bmagyar, 2023-04-12:
Happy you managed to figure it out!

Many thanks for submitting the solution to it, not many people do it and it's great for posterity.