ros-controls/gazebo_ros2_control #201: gazebo process has dies loading ros2 controller

VineetTambe, 2023-05-25:
Hey,

I was able to run the cart_position_example demo a while back but now, all of a sudden when I launch the demo I get the following error after the joint trajectory controller is loaded - 
```
[gzserver-1] [INFO] [1685047419.007274481] [joint_trajectory_controller]: Command interfaces are [position] and state interfaces are [position velocity].
[gzserver-1] [INFO] [1685047419.007380154] [joint_trajectory_controller]: Using 'splines' interpolation method.
[gzserver-1] [INFO] [1685047419.008779607] [joint_trajectory_controller]: Controller state will be published at 50.00 Hz.
[gzserver-1] [INFO] [1685047419.011421384] [joint_trajectory_controller]: Action status changes will be monitored at 20.00 Hz.
[ros2-6] Sucessfully loaded controller joint_trajectory_controller into state active
[INFO] [ros2-6]: process has finished cleanly [pid 34551]
[ERROR] [gzserver-1]: process has died [pid 34363, exit code -11, cmd 'gzserver -slibgazebo_ros_init.so -slibgazebo_ros_factory.so -slibgazebo_ros_force_system.so'].
```

I tried manually loading the controllers and it happens repeatedly after I load the `joint_trajectory_controller`

ahcorde, 2023-05-25:
This info is not enough to help you, can you provide all the log ?

VineetTambe, 2023-05-25:
Yes sure!
```
 ros2 launch gazebo_ros2_control_demos cart_example_position.launch.py 
[INFO] [launch]: All log files can be found below /home/vrex/.ros/log/2023-05-25-19-36-43-358833-robo-133196
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [gzserver-1]: process started with pid [133198]
[INFO] [gzclient-2]: process started with pid [133200]
[INFO] [robot_state_publisher-3]: process started with pid [133202]
[INFO] [spawn_entity.py-4]: process started with pid [133204]
[robot_state_publisher-3] [INFO] [1685057804.730060747] [robot_state_publisher]: got segment cart
[robot_state_publisher-3] [INFO] [1685057804.730194634] [robot_state_publisher]: got segment slideBar
[robot_state_publisher-3] [INFO] [1685057804.730213547] [robot_state_publisher]: got segment world
[spawn_entity.py-4] [INFO] [1685057805.219355615] [spawn_entity]: Spawn Entity started
[spawn_entity.py-4] [INFO] [1685057805.219770010] [spawn_entity]: Loading entity published on topic robot_description
[spawn_entity.py-4] /opt/ros/humble/local/lib/python3.10/dist-packages/rclpy/qos.py:307: UserWarning: DurabilityPolicy.RMW_QOS_POLICY_DURABILITY_TRANSIENT_LOCAL is deprecated. Use DurabilityPolicy.TRANSIENT_LOCAL instead.
[spawn_entity.py-4]   warnings.warn(
[spawn_entity.py-4] [INFO] [1685057805.222041022] [spawn_entity]: Waiting for entity xml on robot_description
[spawn_entity.py-4] [INFO] [1685057805.223898228] [spawn_entity]: Waiting for service /spawn_entity, timeout = 30
[spawn_entity.py-4] [INFO] [1685057805.224297948] [spawn_entity]: Waiting for service /spawn_entity
[spawn_entity.py-4] [INFO] [1685057806.229527477] [spawn_entity]: Calling service /spawn_entity
[spawn_entity.py-4] [INFO] [1685057806.348208767] [spawn_entity]: Spawn status: SpawnEntity: Successfully spawned entity [cartpole]
[gzserver-1] [INFO] [1685057806.476919037] [gazebo_ros2_control]: Loading gazebo_ros2_control plugin
[gzserver-1] [INFO] [1685057806.480847009] [gazebo_ros2_control]: Starting gazebo_ros2_control plugin in namespace: /
[gzserver-1] [INFO] [1685057806.481034493] [gazebo_ros2_control]: Starting gazebo_ros2_control plugin in ros 2 node: gazebo_ros2_control
[gzserver-1] [INFO] [1685057806.482027798] [gazebo_ros2_control]: Loading parameter files /home/vrex/workspaces/ros2_ws/install/gazebo_ros2_control_demos/share/gazebo_ros2_control_demos/config/cartpole_controller.yaml
[gzserver-1] [INFO] [1685057806.517989576] [gazebo_ros2_control]: connected to service!! robot_state_publisher
[gzserver-1] [INFO] [1685057806.522457626] [gazebo_ros2_control]: Recieved urdf from param server, parsing...
[INFO] [spawn_entity.py-4]: process has finished cleanly [pid 133204]
[INFO] [ros2-5]: process started with pid [133310]
[gzserver-1] [WARN] [1685057806.691381274] [gazebo_ros2_control]: The position_proportional_gain parameter was not defined, defaulting to: 0.1
[gzserver-1] [INFO] [1685057806.733179591] [gazebo_ros2_control]: Loading joint: slider_to_cart
[gzserver-1] [INFO] [1685057806.733421418] [gazebo_ros2_control]: 	State:
[gzserver-1] [INFO] [1685057806.733474243] [gazebo_ros2_control]: 		 position
[gzserver-1] [INFO] [1685057806.733898139] [gazebo_ros2_control]: 			 found initial value: 1.000000
[gzserver-1] [INFO] [1685057806.733930568] [gazebo_ros2_control]: 		 velocity
[gzserver-1] [INFO] [1685057806.734117337] [gazebo_ros2_control]: 		 effort
[gzserver-1] [INFO] [1685057806.734193986] [gazebo_ros2_control]: 	Command:
[gzserver-1] [INFO] [1685057806.734217804] [gazebo_ros2_control]: 		 position
[gzserver-1] [INFO] [1685057806.735019751] [resource_manager]: Initialize hardware 'GazeboSystem' 
[gzserver-1] [INFO] [1685057806.736212343] [resource_manager]: Successful initialization of hardware 'GazeboSystem'
[gzserver-1] [INFO] [1685057806.736489599] [resource_manager]: 'configure' hardware 'GazeboSystem' 
[gzserver-1] [INFO] [1685057806.736497187] [resource_manager]: Successful 'configure' of hardware 'GazeboSystem'
[gzserver-1] [INFO] [1685057806.736502492] [resource_manager]: 'activate' hardware 'GazeboSystem' 
[gzserver-1] [INFO] [1685057806.736507030] [resource_manager]: Successful 'activate' of hardware 'GazeboSystem'
[gzserver-1] [INFO] [1685057806.737123941] [gazebo_ros2_control]: Loading controller_manager
[gzserver-1] [WARN] [1685057806.919715853] [gazebo_ros2_control]:  Desired controller update period (0.01 s) is slower than the gazebo simulation period (0.001 s).
[gzserver-1] [INFO] [1685057806.921863472] [gazebo_ros2_control]: Loaded gazebo_ros2_control.
[gzserver-1] [INFO] [1685057807.632086956] [controller_manager]: Loading controller 'joint_state_broadcaster'
[gzserver-1] [INFO] [1685057807.688403811] [controller_manager]: Setting use_sim_time=True for joint_state_broadcaster to match controller manager (see ros2_control#325 for details)
[gzserver-1] [INFO] [1685057807.708741177] [controller_manager]: Configuring controller 'joint_state_broadcaster'
[gzserver-1] [INFO] [1685057807.709196433] [joint_state_broadcaster]: 'joints' or 'interfaces' parameter is empty. All available state interfaces will be published
[ros2-5] Sucessfully loaded controller joint_state_broadcaster into state active
[INFO] [ros2-5]: process has finished cleanly [pid 133310]
[INFO] [ros2-6]: process started with pid [133367]
[gzserver-1] [INFO] [1685057809.047349793] [controller_manager]: Loading controller 'joint_trajectory_controller'
[... cut ...]
[gzserver-1] [INFO] [1685057809.163305651] [joint_trajectory_controller]: Controller state will be published at 50.00 Hz.
[gzserver-1] [INFO] [1685057809.164476862] [joint_trajectory_controller]: Action status changes will be monitored at 20.00 Hz.
[ros2-6] Sucessfully loaded controller joint_trajectory_controller into state active
[INFO] [ros2-6]: process has finished cleanly [pid 133367]
[ERROR] [gzserver-1]: process has died [pid 133198, exit code -11, cmd 'gzserver -slibgazebo_ros_init.so -slibgazebo_ros_factory.so -slibgazebo_ros_force_system.so'].
```

Thank you for helping me out!

VineetTambe, 2023-05-25:
Another peculiar thing that is happening is:
when I run `ros2 action list`
I can see the joint trajectory controller `/joint_trajectory_controller/follow_joint_trajectory`

But when I run the node I get the following:
```
$ros2 run gazebo_ros2_control_demos example_position
node created
terminate called after throwing an instance of 'std::runtime_error'
  what():  could not get action server
[ros2run]: Aborted
```

VineetTambe, 2023-05-30:
Update on this:
Strangely, the velocity example runs without any issues for me.
While the position example keeps on dying!
Can it be an issue with the action server?