ros-controls/gazebo_ros2_control #204: coclon build make errors with protobuf?

luvwinnie, 2023-06-04:
I;m facing these errors when building with in ROS2 humble environment. Anyone know how to solve this problem or knowing that version of protobuf is using?

```/usr/include/gazebo-11/gazebo/msgs/world_modify.pb.h:143:30: error: ‘google::protobuf::uint8* gazebo::msgs::WorldModify::InternalSerializeWithCachedSizesToArray(bool, google::protobuf::uint8*) const’ marked ‘final’, but is not virtual
  143 |   ::google::protobuf::uint8* InternalSerializeWithCachedSizesToArray(
      |                              ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /usr/include/gazebo-11/gazebo/msgs/MessageTypes.hh:136,
                 from /usr/include/gazebo-11/gazebo/msgs/msgs.hh:35,
                 from /usr/include/gazebo-11/gazebo/physics/Entity.hh:28,
                 from /usr/include/gazebo-11/gazebo/physics/Model.hh:30,
                 from /workspaces/isaac_ros-dev/src/gazebo_ros2_control/gazebo_ros2_control/include/gazebo_ros2_control/gazebo_ros2_control_plugin.hpp:45,
                 from /workspaces/isaac_ros-dev/src/gazebo_ros2_control/gazebo_ros2_control/src/gazebo_ros2_control_plugin.cpp:42:
/usr/include/gazebo-11/gazebo/msgs/wrench_stamped.pb.h:145:30: error: ‘google::protobuf::uint8* gazebo::msgs::WrenchStamped::InternalSerializeWithCachedSizesToArray(bool, google::protobuf::uint8*) const’ marked ‘final’, but is not virtual
  145 |   ::google::protobuf::uint8* InternalSerializeWithCachedSizesToArray(
      |                              ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/workspaces/isaac_ros-dev/src/gazebo_ros2_control/gazebo_ros2_control/src/gazebo_ros2_control_plugin.cpp: In member function ‘virtual void gazebo_ros2_control::GazeboRosControlPlugin::Load(gazebo::physics::ModelPtr, sdf::v9::ElementPtr)’:
/workspaces/isaac_ros-dev/src/gazebo_ros2_control/gazebo_ros2_control/src/gazebo_ros2_control_plugin.cpp:290:67: error: ‘__gnu_cxx::__alloc_traits<std::allocator<hardware_interface::HardwareInfo>, hardware_interface::HardwareInfo>::value_type’ {aka ‘struct hardware_interface::HardwareInfo’} has no member named ‘hardware_plugin_name’
  290 |     std::string robot_hw_sim_type_str_ = control_hardware_info[i].hardware_plugin_name;
      |                                                                   ^~~~~~~~~~~~~~~~~~~~
make[2]: *** [CMakeFiles/gazebo_hardware_plugins.dir/build.make:76: CMakeFiles/gazebo_hardware_plugins.dir/src/gazebo_system.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:165: CMakeFiles/gazebo_hardware_plugins.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
make[2]: *** [CMakeFiles/gazebo_ros2_control.dir/build.make:76: CMakeFiles/gazebo_ros2_control.dir/src/gazebo_ros2_control_plugin.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:139: CMakeFiles/gazebo_ros2_control.dir/all] Error 2
make: *** [Makefile:146: all] Error 2
---```

christophfroehlich, 2023-06-04:
Concerning the second error, `hardware_plugin_name` was [introduced in rolling ](https://github.com/ros-controls/ros2_control/blob/bc8052a6d48bc786eb7ffe17cfca9af92f583332/hardware_interface/include/hardware_interface/hardware_info.hpp#L114C7-L115).

Is it possible that you mixed up your ROS distros, or maybe try to compile gazebo_ros2_control from master branch within a humble environment?

gachiemchiep, 2023-08-15:
Switching from master branch into humble branch fixed the above problem.

hmpcjayaweera, 2023-08-23:
Hi, I got the second error (d::string robot_hw_sim_type_str_ = control_hardware_info[i].hardware_plugin_name;) even when I use the humble branch, please can you explain how to fix this

christophfroehlich, 2023-08-23:
> Hi, I got the second error (d::string robot_hw_sim_type_str_ = control_hardware_info[i].hardware_plugin_name;) even when I use the humble branch, please can you explain how to fix this

There is no variable called `hardware_plugin_name` used in humble. Are you sure that you have switched the branches successfully? Try a clean build (`rm -rf build/gazebo_ros2_control`)

hmpcjayaweera, 2023-08-23:
Thanks for your quick reply, yes it is now working. Thanks

closed 2023-08-23 by ahcorde