ros-controls/gazebo_ros2_control #217: Broken link in docs

christophfroehlich, 2023-07-21:
The following link is broken:
https://github.com/ros-controls/gazebo_ros2_control/blob/6458276ca1eaf3ee646afa8e0833386f9785e585/doc/index.rst?plain=1#L289C77-L289C148

Is this section up-to-date? Does one really need Foxy? Can we link to the gazebo_ros2_control_demos at master branch?

ahcorde, 2023-07-21:
you can remove this, the repository doesn't exists https://github.com/ros-controls/gazebo_ros_demos

christophfroehlich, 2023-07-21:
But where is this launch file located? I couldn't find any tutorial in the [moveit2 docs](https://moveit.picknik.ai/humble/search.html?q=gazebo&check_keywords=yes&area=default)
```bash
  ros2 launch rrbot_moveit_demo_nodes rrbot_demo.launch.py
```

christophfroehlich, 2023-08-14:
@ahcorde what do you think? or should I remove this section completely?

ahcorde, 2023-08-14:
I would say remove

closed 2023-08-17 by christophfroehlich