ros-controls/gazebo_ros2_control #273: Loading controller failed，Exception thrown during init stage with message: expected [string] got [string_array]

PANJIACHENG-98, 2024-01-26:
[INFO] [ros2-4]: process has finished cleanly [pid 20770]
[INFO] [ros2-5]: process started with pid [20814]
[INFO] [ros2-6]: process started with pid [20816]
[INFO] [ros2-7]: process started with pid [20818]
[INFO] [ros2-8]: process started with pid [20820]
[gazebo-1] [INFO] [1706239490.497724358] [controller_manager]: Loading controller 'Head_joint_trajectory_controller'
[gazebo-1] Exception thrown during init stage with message: expected [string] got [string_array] 
[gazebo-1] [ERROR] [1706239490.526122183] [controller_manager]: Could not initialize the controller named 'Head_joint_trajectory_controller'
[gazebo-1] [INFO] [1706239490.541068223] [controller_manager]: Loading controller 'Body_joint_trajectory_controller'
[ros2-8] Error loading controller, check controller_manager logs
[gazebo-1] Exception thrown during init stage with message: expected [string] got [string_array] 
[gazebo-1] [ERROR] [1706239490.558322016] [controller_manager]: Could not initialize the controller named 'Body_joint_trajectory_controller'
[ros2-7] Error loading controller, check controller_manager logs
[gazebo-1] [INFO] [1706239490.575401082] [controller_manager]: Loading controller 'Right_arm_joint_trajectory_controller'
[gazebo-1] Exception thrown during init stage with message: expected [string] got [string_array] 
[gazebo-1] [ERROR] [1706239490.586553576] [controller_manager]: Could not initialize the controller named 'Right_arm_joint_trajectory_controller'
[gazebo-1] [INFO] [1706239490.602357917] [controller_manager]: Loading controller 'Left_arm_joint_trajectory_controller'
[ros2-5] Error loading controller, check controller_manager logs
[gazebo-1] Exception thrown during init stage with message: expected [string] got [string_array] 
[gazebo-1] [ERROR] [1706239490.630282542] [controller_manager]: Could not initialize the controller named 'Left_arm_joint_trajectory_controller'
[ros2-6] Error loading controller, check controller_manager logs
[ERROR] [ros2-8]: process has died [pid 20820, exit code 1, cmd 'ros2 control load_controller --set-state active Head_joint_trajectory_controller'].
[ERROR] [ros2-5]: process has died [pid 20814, exit code 1, cmd 'ros2 control load_controller --set-state active Right_arm_joint_trajectory_controller'].
[ERROR] [ros2-6]: process has died [pid 20816, exit code 1, cmd 'ros2 control load_controller --set-state active Left_arm_joint_trajectory_controller'].
[ERROR] [ros2-7]: process has died [pid 20818, exit code 1, cmd 'ros2 control load_controller --set-state active Body_joint_trajectory_controller'].

PANJIACHENG-98, 2024-01-26:
robotck@robotckvrm:~/ros2_ws$ ros2 launch dual_arm_gazebo gazebo.launch.py 
[INFO] [launch]: All log files can be found below /home/robotck/.ros/log/2024-01-26-11-24-23-194658-robotckvrm-20684
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [gazebo-1]: process started with pid [20685]
[INFO] [robot_state_publisher-2]: process started with pid [20687]
[INFO] [spawn_entity.py-3]: process started with pid [20689]
[robot_state_publisher-2] [INFO] [1706239463.446393660] [robot_state_publisher]: got segment base_footprint
[robot_state_publisher-2] [INFO] [1706239463.447034647] [robot_state_publisher]: got segment base_link
[robot_state_publisher-2] [INFO] [1706239463.447362213] [robot_state_publisher]: got segment link-Body
[robot_state_publisher-2] [INFO] [1706239463.447632796] [robot_state_publisher]: got segment link-H0
[robot_state_publisher-2] [INFO] [1706239463.447914577] [robot_state_publisher]: got segment link-H1
[robot_state_publisher-2] [INFO] [1706239463.448163628] [robot_state_publisher]: got segment link-J1
[robot_state_publisher-2] [INFO] [1706239463.448521471] [robot_state_publisher]: got segment link-L0
[robot_state_publisher-2] [INFO] [1706239463.448802501] [robot_state_publisher]: got segment link-L1
[robot_state_publisher-2] [INFO] [1706239463.449108173] [robot_state_publisher]: got segment link-L2
[robot_state_publisher-2] [INFO] [1706239463.449449372] [robot_state_publisher]: got segment link-L3
[robot_state_publisher-2] [INFO] [1706239463.449729957] [robot_state_publisher]: got segment link-L4
[robot_state_publisher-2] [INFO] [1706239463.449994481] [robot_state_publisher]: got segment link-L5
[robot_state_publisher-2] [INFO] [1706239463.450271709] [robot_state_publisher]: got segment link-L6
[robot_state_publisher-2] [INFO] [1706239463.450848680] [robot_state_publisher]: got segment link-L7
[robot_state_publisher-2] [INFO] [1706239463.451254760] [robot_state_publisher]: got segment link-R0
[robot_state_publisher-2] [INFO] [1706239463.451519215] [robot_state_publisher]: got segment link-R1
[robot_state_publisher-2] [INFO] [1706239463.451830331] [robot_state_publisher]: got segment link-R2
[robot_state_publisher-2] [INFO] [1706239463.452092579] [robot_state_publisher]: got segment link-R3
[robot_state_publisher-2] [INFO] [1706239463.452427572] [robot_state_publisher]: got segment link-R4
[robot_state_publisher-2] [INFO] [1706239463.452669246] [robot_state_publisher]: got segment link-R5
[robot_state_publisher-2] [INFO] [1706239463.452955135] [robot_state_publisher]: got segment link-R6
[robot_state_publisher-2] [INFO] [1706239463.453194305] [robot_state_publisher]: got segment link-R7
[robot_state_publisher-2] [INFO] [1706239463.453567495] [robot_state_publisher]: got segment world
[gazebo-1] Gazebo multi-robot simulator, version 11.10.2
[gazebo-1] Copyright (C) 2012 Open Source Robotics Foundation.
[gazebo-1] Released under the Apache 2 License.
[gazebo-1] http://gazebosim.org
[gazebo-1] 
[spawn_entity.py-3] [INFO] [1706239463.922335770] [spawn_entity]: Spawn Entity started
[spawn_entity.py-3] [INFO] [1706239463.922793702] [spawn_entity]: Loading entity published on topic robot_description
[spawn_entity.py-3] /opt/ros/humble/local/lib/python3.10/dist-packages/rclpy/qos.py:307: UserWarning: DurabilityPolicy.RMW_QOS_POLICY_DURABILITY_TRANSIENT_LOCAL is deprecated. Use DurabilityPolicy.TRANSIENT_LOCAL instead.
[spawn_entity.py-3]   warnings.warn(
[gazebo-1] Gazebo multi-robot simulator, version 11.10.2
[gazebo-1] Copyright (C) 2012 Open Source Robotics Foundation.
[gazebo-1] Released under the Apache 2 License.
[gazebo-1] http://gazebosim.org
[gazebo-1] 
[spawn_entity.py-3] [INFO] [1706239463.928699469] [spawn_entity]: Waiting for entity xml on robot_description
[spawn_entity.py-3] [INFO] [1706239464.085984963] [spawn_entity]: Waiting for service /spawn_entity, timeout = 30
[spawn_entity.py-3] [INFO] [1706239464.086367444] [spawn_entity]: Waiting for service /spawn_entity
[spawn_entity.py-3] [INFO] [1706239485.144069257] [spawn_entity]: Calling service /spawn_entity
[spawn_entity.py-3] [INFO] [1706239485.408602148] [spawn_entity]: Spawn status: SpawnEntity: Successfully spawned entity [dual_arm]
[INFO] [spawn_entity.py-3]: process has finished cleanly [pid 20689]
[INFO] [ros2-4]: process started with pid [20770]
[gazebo-1] [Msg] Waiting for master.
[gazebo-1] [Msg] Connected to gazebo master @ http://127.0.0.1:11345
[gazebo-1] [Msg] Publicized address: 192.168.122.130
[gazebo-1] [Wrn] [Event.cc:61] Warning: Deleting a connection right after creation. Make sure to save the ConnectionPtr from a Connect call
[gazebo-1] [INFO] [1706239486.984600643] [gazebo_ros2_control]: Loading gazebo_ros2_control plugin
[gazebo-1] [INFO] [1706239486.989064240] [gazebo_ros2_control]: Starting gazebo_ros2_control plugin in namespace: /
[gazebo-1] [INFO] [1706239486.989379609] [gazebo_ros2_control]: Starting gazebo_ros2_control plugin in ros 2 node: gazebo_ros2_control
[gazebo-1] [INFO] [1706239486.989465076] [gazebo_ros2_control]: Loading parameter files /home/robotck/ros2_ws/install/dual_arm_gazebo/share/dual_arm_gazebo/config/dual_arm_control.yaml
[gazebo-1] [INFO] [1706239486.997542802] [gazebo_ros2_control]: connected to service!! robot_state_publisher
[gazebo-1] [INFO] [1706239486.999247405] [gazebo_ros2_control]: Received urdf from param server, parsing...
[gazebo-1] [INFO] [1706239487.016262191] [gazebo_ros2_control]: Loading joint: J1
[... cut ...]
[ERROR] [ros2-8]: process has died [pid 20820, exit code 1, cmd 'ros2 control load_controller --set-state active Head_joint_trajectory_controller'].
[ERROR] [ros2-5]: process has died [pid 20814, exit code 1, cmd 'ros2 control load_controller --set-state active Right_arm_joint_trajectory_controller'].
[ERROR] [ros2-6]: process has died [pid 20816, exit code 1, cmd 'ros2 control load_controller --set-state active Left_arm_joint_trajectory_controller'].
[ERROR] [ros2-7]: process has died [pid 20818, exit code 1, cmd 'ros2 control load_controller --set-state active Body_joint_trajectory_controller'].

PANJIACHENG-98, 2024-01-26:
run： ros2 launch dual_arm_gazebo gazebo.launch.py 
at last， i can't load my controller,here is my controller yaml:
[image]
the another three controller are in the same config

PANJIACHENG-98, 2024-01-26:
beg for help please!!!!!

PANJIACHENG-98, 2024-01-26:
here is the complete info:
[image]

PANJIACHENG-98, 2024-01-26:
[image]
[image]
[image]
[image]
[image]

Wiktor-99, 2024-01-26:
As stated in the documentation, you should only configure joints in your configuration file.(https://control.ros.org/master/doc/ros2_controllers/position_controllers/doc/userdoc.html)

christophfroehlich, 2024-01-30:
> As stated in the documentation, you should only configure joints in your configuration file.(https://control.ros.org/master/doc/ros2_controllers/position_controllers/doc/userdoc.html)

correct. @PANJIACHENG-98 you name the controller *joint_trajectory_controller, but configured the type `position_controllers/JointGroupPositionController` which has only one parameter `joints` as written above.

PANJIACHENG-98, 2024-02-23:
well solved thanks bro

PANJIACHENG-98, 2024-02-23:
[image]

closed 2024-02-23 by christophfroehlich