ros-controls/gazebo_ros2_control #274: Errors after launching demos: controller manager issue?

tera-bytes, 2024-01-28:
Whenever I try to launch any of the existing demos such as diff_drive, the gazebo simulation launches fine but I get these errors:
[image]
So then any ros2 run command fails. I'm on ubuntu 22.04 using Humble. Any help would be much appreciated!

SXYuuuuuu, 2024-01-29:
I meet the same error, it make me crazy

ahcorde, 2024-01-29:
HI @tera-bytes 

can you copy the full trace instead of adding a picture?

christophfroehlich, 2024-01-29:
And please add the exact version of ros2_control and gazebo_ros2_control (i.e., by running `ros2 pkg xml ros2-control -t version`)

tera-bytes, 2024-01-29:
Hey, sorry about this but I actually managed to fix it today. I just deleted my old workspace and cloned and rebuilt. Everything works fine now. I didn't really do anything differently so I'm still not sure why I got those errors.

bmagyar, 2024-01-30:
What was in your workspace? Any chance it had gazebo_ros2_control, ros2_control or ros2_controllers?

tera-bytes, 2024-01-30:
> What was in your workspace? Any chance it had gazebo_ros2_control, ros2_control or ros2_controllers?
In the same workspace, I had cloned the github for the Ur5 Gazebo Simulation and it had gazebo_ros2_control and ros2_control.

christophfroehlich, 2024-02-22:
closing this as it seems that it have fixed itself

closed 2024-02-22 by christophfroehlich