ros-controls/gazebo_ros2_control #275: How can solve this problem?

uiinlee, 2024-01-29:
--- stderr: gazebo_ros2_control                                 
/home/uiin/ros2_ws/src/gazebo_ros2_control/src/gazebo_system.cpp:28:10: fatal error: hardware_interface/lexical_casts.hpp: No such file or directory
   28 | #include "hardware_interface/lexical_casts.hpp"
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
gmake[2]: *** [CMakeFiles/gazebo_hardware_plugins.dir/build.make:76: CMakeFiles/gazebo_hardware_plugins.dir/src/gazebo_system.cpp.o] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:165: CMakeFiles/gazebo_hardware_plugins.dir/all] Error 2
gmake[1]: *** Waiting for unfinished jobs....
/home/uiin/ros2_ws/src/gazebo_ros2_control/src/gazebo_ros2_control_plugin.cpp: In member function ‘virtual void gazebo_ros2_control::GazeboRosControlPlugin::Load(gazebo::physics::ModelPtr, sdf::v9::ElementPtr)’:
/home/uiin/ros2_ws/src/gazebo_ros2_control/src/gazebo_ros2_control_plugin.cpp:292:33: error: no matching function for call to ‘hardware_interface::ResourceManager::load_urdf(std::string&, bool, bool)’
  292 |     resource_manager_->load_urdf(urdf_string, false, false);
      |     ~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /opt/ros/humble/include/controller_manager/controller_manager.hpp:44,
                 from /home/uiin/ros2_ws/src/gazebo_ros2_control/include/gazebo_ros2_control/gazebo_ros2_control_plugin.hpp:42,
                 from /home/uiin/ros2_ws/src/gazebo_ros2_control/src/gazebo_ros2_control_plugin.cpp:42:
/opt/ros/humble/include/hardware_interface/resource_manager.hpp:84:8: note: candidate: ‘void hardware_interface::ResourceManager::load_urdf(const string&, bool)’
   84 |   void load_urdf(const std::string & urdf, bool validate_interfaces = true);
      |        ^~~~~~~~~
/opt/ros/humble/include/hardware_interface/resource_manager.hpp:84:8: note:   candidate expects 2 arguments, 3 provided
/home/uiin/ros2_ws/src/gazebo_ros2_control/src/gazebo_ros2_control_plugin.cpp:310:67: error: ‘__gnu_cxx::__alloc_traits<std::allocator<hardware_interface::HardwareInfo>, hardware_interface::HardwareInfo>::value_type’ {aka ‘struct hardware_interface::HardwareInfo’} has no member named ‘hardware_plugin_name’
  310 |     std::string robot_hw_sim_type_str_ = control_hardware_info[i].hardware_plugin_name;
      |                                                                   ^~~~~~~~~~~~~~~~~~~~
gmake[2]: *** [CMakeFiles/gazebo_ros2_control.dir/build.make:76: CMakeFiles/gazebo_ros2_control.dir/src/gazebo_ros2_control_plugin.cpp.o] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:139: CMakeFiles/gazebo_ros2_control.dir/all] Error 2
gmake: *** [Makefile:146: all] Error 2
---
Failed   <<< gazebo_ros2_control [14.7s, exited with code 2]

ahcorde, 2024-01-29:
As you can see in this PR https://github.com/ros-controls/gazebo_ros2_control/pull/266 where this change was included, the CI is compiling the code (it's red because one test is failing)

There are some new breaking changes in `ros2_control` which means you need to wait until the next Humble sync (which I think it was out last week) or you can compile `ros2_control` from source

ahcorde, 2024-01-29:
if you run:

```bash
sudo apt-get update
sudo apt-get upgrade
```

the you should be able to compile it

closed 2024-04-07 by christophfroehlich