ros-controls/gazebo_ros2_control #276: How to set the Effort while using JointGroupVelocityController in Gazebo

alextdbc, 2024-02-01:
Hello, im using the JointGroupVelocityController to control a 4-wheeled-robot. Each of these wheels is connected to the body by an robot arm, which an rotate to lift the body up and down. 

I want to create a simulation in Gazebo and control the arms with the velocity controller. For now i can control the velocity of the arms but when i want to lift the body by rotating the arm into the ground, the arms stop rotating, instead of lifting up the body. It seems like there is not enough torque. My effort limit is set to "10000", but when i am subscribing to the /robot_states-topic the effort is always zero.

So my question is, in which way i can set the effort for velocity control to lift the robot body with the velocity controller? 

Hopefully i can find some help here.

christophfroehlich, 2024-02-01:
I moved this issue to gazebo_ros2_control, because this is more a question of simulation than ros2_controllers. (I assumed you mean gazebo classic instead of gazebo? If not, I'll move it to gz_ros2_control).

If you use a velocity command interface with gazebo_ros2_control, then the velocity of the physics engine is "overwritten" and no effort is calculated, hence the state interface given back reads zero. But velocity interface is de-facto standard for wheeled robots, this should not be the problem here.

Have you tried to move the robot around by manually applying a force from the GUI? If this works (and you haven't accidentally fixed the robot in the world frame), then try to use a [effort_controllers/JointGroupEffortController](https://control.ros.org/master/doc/ros2_controllers/effort_controllers/doc/userdoc.html#effort-controllers-jointgroupeffortcontroller) for one arm, apply the effort and see what happens.

alextdbc, 2024-02-01:
Hi thank you for your answer. Youre right im using Gazebo classic.

My robot is not fixed to the world frame and i can move it around. I already tried to use the JointTrajectoryController to set the goal position and that worked completely fine. Now i've tried it with the EffortController and that worked also. Just when im using the VelocityController the Wheels and Arm arent spinning as soon as they get contact with the Ground.

christophfroehlich, 2024-02-01:
This is a drawback of the implementation with writing position or velocity values to the physics engine, see #240 for a similar issue.
To overcome this, I'd suggest to write [an own gazebo plugin](https://control.ros.org/master/doc/gazebo_ros2_control/doc/index.html#advanced-custom-gazebo-ros2-control-simulation-plugins), exposing velocity interfaces to your controller but writing effort to the joints in the simulation loop. In between you can use a high-gain PID controller or similar.