ros-controls/gazebo_ros2_control #278: No rule to make target needed by 'libgazebo_ros2_control.so'

xardarii, 2024-02-04:
Hello everyone!
I am facing an issue when trying to colcon build my workspace with gazebo_ros2_control package inside it. It used to build perfectly and I even used it and was working fine until a few days ago when I had some Ubuntu automatic software updates running and now it wont build gazebo_ros2_control. I dont know what the problem is but here is a screenshot from the terminal:

[image]

Ubuntu Version : 22.04.3 LTS
ROS Humble
Gazebo Version 11.10.2

christophfroehlich, 2024-02-05:
We don't have a direct dependency on ignition-math. gazebo_hardware_plugins has gazebo_dev as dependency, but this one hasn't been updated for a couple of months. 

Try a clean build and run rosdep before. :man_shrugging: 

Do you need a source build of this package, or could you just install the released binary instead?

christophfroehlich, 2024-04-25:
Closing due to inactivity.