ros-controls/gazebo_ros2_control #305: Error in destruction of rcl subscription handle: Failed to delete datareader

FeetPro, 2024-04-23:
Hello, I got some problems in sending command to gripper.
My gripper is shown successfully in gazebo with command:
`ros2 launch gazebo_ros2_control_demos gripper_mimic_joint_example.launch.py`
But when i send command to move the gripper in other terminal:
`ros2 run gazebo_ros2_control_demos example_gripper`
It appears error:

cannot publish data, at ./src/rmw_publish.cpp:62 during '__function__'
[ERROR] [1713850255.212319171] [gripper_test_node.rclcpp]: Error in destruction of rcl subscription handle: Failed to delete datareader, at ./src/subscription.cpp:52, at ./src/rcl/subscription.c:184
cannot publish data, at ./src/rmw_publish.cpp:62 during '__function__'
Fail in delete datareader, at ./src/rmw_service.cpp:104 during '__function__'
[ros2run]: Segmentation fault

how can i fix this?
I use Ros2 Humble and Ubuntu 22.04,
Thanks.

christophfroehlich, 2024-04-23:
This is a quick-and-dirty design of the example_gripper executable. It publishes the topics but does not exit cleanly. Check with `ros2 topic echo /gripper_controller/commands`, you'll receive the commands and the gripper should move.

gavanderhoorn, 2024-11-11:
Errors remind me of ros2/rclcpp#1766. This got 'fixed' in ros2/rmw_fastrtps#770, which unfortunately won't get backported.