ros-controls/gazebo_ros2_control #325: missing header ignition/common/Profile.hh when installing from source

david-wb, 2024-05-16:
**Describe the bug**
I'm trying to install `ros2_control` from source on ROS2 Humble / Ubuntu 22.04 as described [here](https://control.ros.org/humble/doc/getting_started/getting_started.html#building-from-source). Most packages build without errors, except it fails when building `gazebo_ros2_control`.

```
--- stderr: gazebo_ros2_control                                                                                                                                                                  
In file included from /usr/include/gazebo-11/gazebo/physics/Joint.hh:27,
                 from /home/chroma/pmb2_public_ws/src/ros-controls/gazebo_ros2_control/gazebo_ros2_control/include/gazebo_ros2_control/gazebo_system_interface.hpp:23,
                 from /home/chroma/pmb2_public_ws/src/ros-controls/gazebo_ros2_control/gazebo_ros2_control/include/gazebo_ros2_control/gazebo_system.hpp:29,
                 from /home/chroma/pmb2_public_ws/src/ros-controls/gazebo_ros2_control/gazebo_ros2_control/src/gazebo_system.cpp:23:
/usr/include/gazebo-11/gazebo/common/Event.hh:32:10: fatal error: ignition/common/Profiler.hh: No such file or directory
   32 | #include "ignition/common/Profiler.hh"
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
gmake[2]: *** [CMakeFiles/gazebo_hardware_plugins.dir/build.make:76: CMakeFiles/gazebo_hardware_plugins.dir/src/gazebo_system.cpp.o] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:165: CMakeFiles/gazebo_hardware_plugins.dir/all] Error 2
gmake[1]: *** Waiting for unfinished jobs....
In file included from /usr/include/gazebo-11/gazebo/common/Events.hh:25,
                 from /opt/ros/humble/include/gazebo_ros/executor.hpp:19,
                 from /opt/ros/humble/include/gazebo_ros/node.hpp:20,
                 from /home/chroma/pmb2_public_ws/src/ros-controls/gazebo_ros2_control/gazebo_ros2_control/src/gazebo_ros2_control_plugin.cpp:42:
/usr/include/gazebo-11/gazebo/common/Event.hh:32:10: fatal error: ignition/common/Profiler.hh: No such file or directory
   32 | #include "ignition/common/Profiler.hh"
```

I can see that `libignition-common3-dev` is installed.
```
$ apt show libignition-common3-dev
Package: libignition-common3-dev
Version: 3.14.0+dfsg-5
Priority: optional
Section: universe/libdevel
Source: ignition-common3
Origin: Ubuntu
Maintainer: Ubuntu Developers <ubuntu-devel-discuss@lists.ubuntu.com>
Original-Maintainer: Debian Science Maintainers <debian-science-maintainers@lists.alioth.debian.org>
Bugs: https://bugs.launchpad.net/ubuntu/+filebug
Installed-Size: 294 kB
Depends: libignition-common3-core-dev (= 3.14.0+dfsg-5), libignition-common3-av-dev (= 3.14.0+dfsg-5), libignition-common3-graphics-dev (= 3.14.0+dfsg-5), libignition-common3-profiler3 (= 3.14.0+dfsg-5), libignition-common3-events3 (= 3.14.0+dfsg-5)
Breaks: libignition-common-dev (<< 3.14.0+dfsg-1~)
Replaces: libignition-common-dev (<< 3.14.0+dfsg-1~)
Homepage: https://ignitionrobotics.org/libs/common
Download-Size: 51,1 kB
APT-Manual-Installed: no
APT-Sources: http://fr.archive.ubuntu.com/ubuntu jammy/universe amd64 Packages
Description: Collection of useful code used by robotics apps - Metapackage
 Ignition common is a component in the Ignition framework, a set of
 libraries designed to rapidly develop robot applications. A collection of
 useful classes and functions for handling many command tasks. This includes
 parsing 3D mesh files, managing console output, and using PID controllers.
 .
 Metapackage for development. Includes events and profiler.
```
But the expected header file is missing
```
$ ls /usr/include/ignition/common3/ignition/common/
events  events.hh  profiler
```

What to do? Thank you.

**To Reproduce**
Steps to reproduce the behavior:
1. Create a fresh installation of Ubuntu22.04 with ROS2 Humble
2. Install `ros2_control` from source.

**Environment (please complete the following information):**
 - OS: Ubuntu 22.04
 - Version: Humble

christophfroehlich, 2024-05-17:
I have a similar setup, the file is in `/usr/include/ignition/common4/ignition/common/Profiler.hh`. It seems that you have some dependencies missing?

Just to clarify, you have ran rosdep update, apt update, and rosdep install successfully? 
Do you have a fresh install of ubuntu? Try to apt upgrade, maybe something changed in the meantime.

christophfroehlich, 2024-08-08:
Closing due to inactivity.

closed 2024-08-08 by christophfroehlich