ros-controls/gazebo_ros2_control #342: Position_controller not working properly as wheel controller of mobile robot

HPCLOL, 2024-06-19:
I am trying to use ROS2 to make a mobile robot move in Gazebo, and the mobile robot just looks like a vehicle with four wheels. For some reason, I use the position_controller to control each wheel. When I post a series of trajectory points to the position controller, the wheels rotated  but the robot did not move, as if it is slipping. When I replace the controller with the velocity_controller or effort_controller , the wheels can drive the robot to move normally. I have checked the URDF file and there are no issues with the contact and friction between the wheels and the ground. 

I want to know why this happens and want to control the wheels through the position_controller to make the robot move.

My ros2 version is iron, and I use gazebo classic.

Thank you so much!

christophfroehlich, 2024-06-19:
I guess this is the same result as described here https://github.com/ros-controls/gazebo_ros2_control/issues/240. 
You could use https://control.ros.org/rolling/doc/gazebo_ros2_control/doc/index.html#using-pid-control-joints to overcome this issue.

HPCLOL, 2024-06-19:
> I guess this is the same result as described here #240. You could use https://control.ros.org/rolling/doc/gazebo_ros2_control/doc/index.html#using-pid-control-joints to overcome this issue.

Thank you so much, it works when I replace the command interface of original joints with pid-control-joints!

But I have a suggestion, the command_interface name for pid-control-joints should be "position_pid" but "position_PID" in  https://control.ros.org/rolling/doc/gazebo_ros2_control/doc/index.html#using-pid-control-joints. 

Actually, I noticed this webpage earlier and tried to use the pid-control-joints, but the controller did not activate successfully because the command interface name was incorrect. I didn't pay attention to this at the time and gave up trying to use it.

Thank you again！

christophfroehlich, 2024-06-19:
> But I have a suggestion, the command_interface name for pid-control-joints should be "position_pid" but "position_PID" in https://control.ros.org/rolling/doc/gazebo_ros2_control/doc/index.html#using-pid-control-joints.

I see, thanks for reporting. Do you mind submitting a PR to fix the [docs](https://github.com/ros-controls/gazebo_ros2_control/blob/master/doc/index.rst)?

closed 2024-06-19 by christophfroehlich

HPCLOL, 2024-06-19:
>  Do you mind submitting a PR to fix the [docs](https://github.com/ros-controls/gazebo_ros2_control/blob/master/doc/index.rst)?

Ok, I will submit a PR later.