ros-controls/gazebo_ros2_control #365: type 'position_controllers/JointGroupPositionController') not found

linhezi, 2024-08-08:
[gazebo-2] [ERROR] [1723104249.814950159] [controller_manager]: Loader for controller 'joint_trajectory_controller' (type 'position_controllers/JointGroupPositionController') not found.

where is the position_controllers? when I use 'ros2 control list_controller_types' ,there is no such controller .

how to use position to control ur10 robot , because it has 6 axis, when I use 'joint_trajectory_controller/JointTrajectoryController ', it comes to error
"gazebo-2  Exception thrown during init stage with message: parameter joints' has invalid type: expected [string array] got [string]gazebo-2][ERROR] [1723098977.460478317] [controller manager]: could not initialize the controller named joint trajectory controller

christophfroehlich, 2024-08-08:
Please have a look at the demos in this repository, install the dependencies (using rosdep for example). 

for joint trajectory controller, have a look at this [demo](https://control.ros.org/rolling/doc/ros2_control_demos/example_1/doc/userdoc.html) or the official [UR drivers](https://github.com/UniversalRobots/Universal_Robots_ROS2_Driver)

christophfroehlich, 2024-08-08:
Closing this issue, as it is not related to gazebo_ros2_control. Please open self-contained issues (one per problem) in the respective repositories.

closed 2024-08-08 by christophfroehlich