ros-controls/gazebo_ros2_control #366: When using an namespace for my omniroboter the controller_manager doesn't get an update_rate

AshBastian, 2024-08-13:
**Description of the bug**
I'm trying to use mutiple omni-directional roboter with the ROS navigationstack 2 and therefore I need to setup my own controller. When I only spawn 1 roboter without a namespace, the controller_manager gets loaded without any problems, but as soon as I give the roboter an namespace, the controller_manager doesn't receive the update_parameter anymore and therefore the whole gazebo_plugin fails.

**Steps to reproduce the behavior:**
```
import os

from ament_index_python.packages import get_package_share_directory

from launch import LaunchDescription
from launch.actions import IncludeLaunchDescription
from launch.launch_description_sources import PythonLaunchDescriptionSource
from launch_ros.actions import Node

def generate_launch_description():
    # Create the launch description and populate
    ld = LaunchDescription()

    bringup_dir = get_package_share_directory('gazebo_simulation')
    pkg_gazebo_ros = get_package_share_directory('gazebo_ros')

    urdf = os.path.join(bringup_dir, 'urdf', 'rs_robot.urdf')

    remappings = [('/tf', 'tf'),
                  ('/tf_static', 'tf_static')]
    
    start_gazebo_server_cmd = IncludeLaunchDescription(
            PythonLaunchDescriptionSource(
                os.path.join(pkg_gazebo_ros, 'launch', 'gzserver.launch.py')
            ),
            launch_arguments={'world': 'omni_nav_world2.model'}.items(),
        )

    start_gazebo_client_cmd = IncludeLaunchDescription(
            PythonLaunchDescriptionSource(
                os.path.join(pkg_gazebo_ros, 'launch', 'gzclient.launch.py')
            ),
        )

    omni_state_publisher = Node(
        package='robot_state_publisher',
        executable='robot_state_publisher',
        name='robot_state_publisher',
        namespace='',
        output='screen',
        parameters=[{'use_sim_time': False,
                        'publish_frequency': 10.0}],
        remappings=remappings,
        arguments=[urdf])
    
    spawn_omni = Node(
            package='gazebo_ros',
            executable='spawn_entity.py',
            arguments=[
                '-file', os.path.join(bringup_dir,'models', 'rs_robot/rs_robot.sdf'),
                '-entity', 'test',
                '-robot_namespace', '',
                '-x', '0.0', '-y', '0.0',
                '-z', '1.01', '-Y', '0.0',
                '-unpause',
            ],
            output='screen',
        )
    
    ld.add_action(start_gazebo_server_cmd)
    ld.add_action(start_gazebo_client_cmd)
    
    ld.add_action(omni_state_publisher)
    ld.add_action(spawn_omni)
    
    
    return ld
```
Change the namespace and -robot_namespace variable to something and the system fails

**Expected behavior**
The expected behavior would be, that the system loads the controller_manager into the namespace with all the parameter.

**Code important for the Execution**
My .yaml file that the system loads.
```
controller_manager:
  ros__parameters:
    update_rate: 1000  # Hz

    joint_state_broadcaster:
      type: joint_state_broadcaster/JointStateBroadcaster

    forward_velocity_controller:
      type: forward_command_controller/ForwardCommandController

forward_velocity_controller:
  ros__parameters:
    joints:
      - first_wheel_joint
      - second_wheel_joint
      - third_wheel_joint
      - fourth_wheel_joint
    interface_name: velocity
    command_interfaces:
      - velocity
    state_interfaces:
      - velocity
```

The part in the .sdf file wäre the gazebo plugin is called:
```
<ros2_control name="GazeboSystem" type="system">
    <hardware>
      <plugin>gazebo_ros2_control/GazeboSystem</plugin>
    </hardware>
    
    <joint name="first_wheel_joint">
      <command_interface name="velocity">
        <param name="min">-3.14</param>
        <param name="max">3.14</param>
      </command_interface>
      <state_interface name="velocity"/>
    </joint>

    <joint name="second_wheel_joint">
      <command_interface name="velocity">
        <param name="min">-3.14</param>
        <param name="max">3.14</param>
      </command_interface>
      <state_interface name="velocity"/>
    </joint>

    <joint name="third_wheel_joint">
      <command_interface name="velocity">
        <param name="min">-3.14</param>
        <param name="max">3.14</param>
      </command_interface>
      <state_interface name="velocity"/>
    </joint>

    <joint name="fourth_wheel_joint">
      <command_interface name="velocity">
        <param name="min">-3.14</param>
        <param name="max">3.14</param>
      </command_interface>
      <state_interface name="velocity"/>
    </joint>
  </ros2_control>
	
    <plugin name="gazebo_ros2_control" filename="libgazebo_ros2_control.so">
<parameters>/home/bastian/Documents/omni_navigation/install/gazebo_simulation/share/gazebo_simulation/models/rs_robot/config/controllers.yaml</parameters>
    </plugin>
```

**Environment (please complete the following information):**
 - OS: Ubuntu 22.04.4 LTS
 - Version Humble Version 16.0.10 (That is the version number of my rclcpp package)

**Additional Points for what I already found**
The system works as intended until the reset for the controller_manager in the gazebo_ros2_controll_plugin.cpp line 384
Here the systems doesn't load the update_rate and I believe the problem is in the executor that parses the information from the .yaml file incorrectly. Therefore the parameter for the update_rate is not set and the controller_manager therefore cannot set the value, which breaks the system. The arguments themself are given correctly to the executor in the lines 256 until 266 in the gazebo_ros2_controll_plugin.cpp.
[image]
[image]

christophfroehlich, 2024-08-13:
Is this maybe related to https://github.com/ros-controls/gazebo_ros2_control/pull/181?

christophfroehlich, 2024-08-13:
Or maybe this is simpler, and you should add the namespace as argument to the plugin as described [here](https://control.ros.org/rolling/doc/gazebo_ros2_control/doc/index.html#add-the-gazebo-ros2-control-plugin).

AshBastian, 2024-08-13:
> Or maybe this is simpler, and you should add the namespace as argument to the plugin as described [here](https://control.ros.org/rolling/doc/gazebo_ros2_control/doc/index.html#add-the-gazebo-ros2-control-plugin).

That doesn't do anything, I tried

AshBastian, 2024-08-13:
> Is this maybe related to #181?

And I tried to comment out the lines mentioned in the other issue, but that also doesn't have any effect on my problem

AshBastian, 2024-08-13:
[image]
[image]
It removes the __ns line from the arguments the executor will parse in the end.
I made it visible like this in the lines 256 of the gazebo_ros2_control_plugin.cpp
```
std::vector<const char *> argv;
  for (const auto & arg : arguments) {
    argv.push_back(reinterpret_cast<const char *>(arg.data()));
    RCLCPP_WARN_STREAM(impl_->model_nh_->get_logger(),arg.data());
  }
```

christophfroehlich, 2024-08-13:
I created a MWE in this https://github.com/ros-controls/gazebo_ros2_control/pull/367

`ros2 launch gazebo_ros2_control_demos cart_example_position_namespaced.launch.py`

runs successfully and 
```
ros2 control list_controllers -c /my_ns/controller_manager
joint_trajectory_controller[joint_trajectory_controller/JointTrajectoryController] active    
joint_state_broadcaster[joint_state_broadcaster/JointStateBroadcaster] active 
```

I think you forgot to namespace the parameters in the yaml file.

AshBastian, 2024-08-13:
I namespaced everything like you did, but it still doesn't work, the different right now is that you use the urdf as the startingpoint for the plugin, while I start it from an sdf file

AshBastian, 2024-08-13:
And I also tried to change my stuff into your example and it just doesn't load the urdf anymore, after that I tried to get your example unto my computer but that also just breaks when I try to colcon build it, so I cannot verify if your solution works on my system

christophfroehlich, 2024-08-13:
you can just take the new files and add it to the humble branch of this repo, or cherry-pick the PR on top of it. The demo should run on every distro.

AshBastian, 2024-08-13:
I'm really sorry, but I really don't know what I should do now to get the demo running on my system, because I'm not an expert on how to manages the git repos to get them on my system without destroying the rest of my ros

AshBastian, 2024-08-13:
Do I just pull the current repo over an pull request and then add the new files. Afterwards I colcon build it on my system, source it and start the launch file like you did to hopefully get the same result?

christophfroehlich, 2024-08-13:
As you are using Humble distro, you need to build the `humble` branch of this repo. Before that, add the three files of the [PR](https://github.com/ros-controls/gazebo_ros2_control/pull/367/files) in the respective folders.
As an alternative, you can build the rolling version of the ros2_control stack, see ["development version" here](https://control.ros.org/humble/doc/getting_started/getting_started.html).

AshBastian, 2024-08-14:
So the demo is running without any problems on my system, so could the problem be that my .sdf is wrong in some way, so the system is failing and if so, how would I need to change it?

```
<sdf version='1.7'>
  <model name='omni_robot'>

    <link name="base_footprint"/>

    <link name='base_link'>
      <inertial>
        <pose>0 0 0 0 0 0</pose>
        <mass>1.167</mass>
        <inertia>
          <ixx>0.0016</ixx>
          <ixy>0</ixy>
          <ixz>0</ixz>
          <iyy>0.0016</iyy>
          <iyz>0</iyz>
          <izz>0.0032</izz>
        </inertia>
      </inertial>
      <collision name='base_footprint_fixed_joint_lump__base_link_collision'>
        <pose>0 0 0 0 0 0</pose>
        <geometry>
          <mesh>
            <scale>1 1 1</scale>
            <uri>model://rs_robot/meshes/STL/base/base_link.stl</uri>
          </mesh>
        </geometry>
        <surface>
          <contact>
            <ode/>
          </contact>
          <friction>
            <ode>
              <mu>0.05</mu>
              <mu2>0.05</mu2>
            </ode>
          </friction>
        </surface>
      </collision>
      <visual name='base_footprint_fixed_joint_lump__base_link_visual'>
        <pose>0 0 0 0 0 0</pose>
        <geometry>
          <mesh>
            <scale>1 1 1</scale>
            <uri>model://rs_robot/meshes/DAE/base/base_link.dae</uri>
          </mesh>
        </geometry>
      </visual>
      <self_collide>0</self_collide>
      <sensor name='front_lrf_sensor' type='ray'>
        <always_on>true</always_on>
        <visualize>true</visualize>
        <update_rate>40</update_rate>
        <pose>0 0 0.1395 0 -0 0</pose>
          <ray>
            <scan>
              <horizontal>
                <samples>720</samples>
                <resolution>1.000000</resolution>
                <min_angle>-3.14158</min_angle>
                <max_angle>3.14158</max_angle>
              </horizontal>
            </scan>
            <range>
              <min>0.120000</min>
              <max>20</max>
              <resolution>0.015000</resolution>
            </range>
[... 3100 lines cut ...]
        <param name="min">-3.14</param>
        <param name="max">3.14</param>
      </command_interface>
      <state_interface name="velocity"/>
    </joint>

    <joint name="fourth_wheel_joint">
      <command_interface name="velocity">
        <param name="min">-3.14</param>
        <param name="max">3.14</param>
      </command_interface>
      <state_interface name="velocity"/>
    </joint>
  </ros2_control>
  
    <plugin name="gazebo_ros2_control" filename="libgazebo_ros2_control.so">
      <ros>
        <namespace>omni1</namespace>
      </ros>
      <parameters>/home/bastian/Documents/omni_navigation/install/gazebo_simulation/share/gazebo_simulation/models/rs_robot/config/controllers.yaml</parameters>
    </plugin>
  </model>
</sdf>

```

AshBastian, 2024-08-14:
It is used in my example that way, that is my I have both data formats, an urdf and an sdf file

christophfroehlich, 2024-08-14:
parts inside `<gazebo>` tags in the URDF are directly used in the internally generated sdf. You can test this with

```
xacro /path/to/model.urdf.xacro > MODEL.urdf
gz sdf -p MODEL.urdf > MODEL.sdf
```

AshBastian, 2024-08-14:
Thanks for the quick answer, I will try that and update here as soon as I'm finished

AshBastian, 2024-08-14:
[image]
[image]
It seems to work to load the urdf corrrectly, but now that I try to start the system gazebo just crashes

AshBastian, 2024-08-14:
And if I load your udrf file into my simulation, gazebo starts, but then I get the following error:
[image]

AshBastian, 2024-08-16:
I finally found the problems in my system, there are two things you need to look out for, to not get into the same problem as me, with the other additional information in this Issue
First of all, if you try to load the ```<plugin name="gazebo_ros2_control" filename="libgazebo_ros2_control.so">``` you need to use ```<parameters>$(find gazebo_simulation)/rest_of_path/controllers_namespaced.yaml</parameters>``` with the find ros_package, because the abolute path crashes the system.
Second of all you need to use ```<mesh filename="file:///$(find robot_models)/rest_of_path/model.stl"/>``` with the file:/// search, because if you don't use that, your gazebo will just crash

AshBastian, 2024-08-16:
Only one last thing I'm unable to get started, is to use an arg parameter in the urdf and the python file to set the namespace dynamically and not using specific files for each individual robot