ros-controls/gazebo_ros2_control #377: Build Failure: Unable to Build gazebo_ros2_control on Windows

SENAI-GilmarCorreia, 2024-09-24:
**Description**:

I am encountering an issue while trying to build the `gazebo_ros2_control` package on a Windows environment. The build fails with the following error log. I've anonymized any user-specific data in the output. Please advise on how to resolve this issue.

**Error Log (Anonymized)**:

```bash
[1/4] Building CXX object CMakeFiles\gazebo_hardware_plugins.dir\src\gazebo_system.cpp.obj
FAILED: CMakeFiles/gazebo_hardware_plugins.dir/src/gazebo_system.cpp.obj 
C:\Program Files\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.29.30133\bin\Hostx64\x64\cl.exe  /nologo /TP -DABSL_CONSUME_DLL -DDART_ACTIVE_LOG_LEVEL=2 -DDART_HAVE_spdlog=1 -DDEFAULT_RMW_IMPLEMENTATION=rmw_fastrtps_cpp -DFASTCDR_DYN_LINK -DFMT_HEADER_ONLY=1 -DFMT_SHARED -DPROTOBUF_USE_DLLS -DTBB_USE_DEBUG -DTINYXML2_DEBUG -DTINYXML2_IMPORT -D_CRT_SECURE_NO_WARNINGS -D_FILE_OFFSET_BITS=64 -D_SILENCE_ALL_CXX17_DEPRECATION_WARNINGS -D_SILENCE_CXX17_CODECVT_HEADER_DEPRECATION_WARNING -Dgazebo_hardware_plugins_EXPORTS -IC:\path\to\project\include -IC:\path\to\env\Library\include\gazebo-11 -IC:\vcpkg\installed\x64-windows\include\sdformat-9.8 -IC:\path\to\env\Library\include\OGRE -IC:\path\to\env\Library\include\SDL2 -external:IC:\opt\ros\humble\include -external:IC:\vcpkg\installed\x64-windows\include\ignition\math6 -external:IC:\vcpkg\installed\x64-windows\include\ignition\transport8 -external:IC:\Python38\lib\site-packages\numpy\core\include -external:W0 /DWIN32 /D_WINDOWS /W3 /GR /EHsc  /MDd /Zi /Ob0 /Od /RTC1 -std:c++17 /W4 /utf-8 /wd4251 /showIncludes /FoCMakeFiles\gazebo_hardware_plugins.dir\src\gazebo_system.cpp.obj /FdCMakeFiles\gazebo_hardware_plugins.dir\ /FS -c C:\path\to\project\src\gazebo_ros2_control\gazebo_ros2_control\src\gazebo_system.cpp
cl : Command line warning D9025 : overriding '/W3' with '/W4'

C:\path\to\env\Library\include\absl/base/prefetch.h(183): warning C4100: 'addr': unreferenced formal parameter
C:\path\to\env\Library\include\google/protobuf/inlined_string_field.h(378): warning C4100: 'arena': unreferenced formal parameter
C:\path\to\env\Library\include\absl/container/internal/hashtablez_sampler.h(210): warning C4100: 'inline_element_size': unreferenced formal parameter
C:\path\to\env\Library\include\google/protobuf/descriptor.h(2239): warning C4100: 'message': unreferenced formal parameter
C:\path\to\env\Library\include\google/protobuf/descriptor.h(2238): warning C4100: 'location': unreferenced formal parameter
C:\path\to\env\Library\include\google/protobuf/descriptor.h(2238): warning C4100: 'descriptor': unreferenced formal parameter
C:\path\to\env\Library\include\google/protobuf/descriptor.h(2237): warning C4100: 'element_name': unreferenced formal parameter
C:\path\to\env\Library\include\google/protobuf/descriptor.h(2236): warning C4100: 'filename': unreferenced formal parameter
C:\path\to\env\Library\include\google/protobuf/descriptor.h(2246): warning C4100: 'message': unreferenced formal parameter
C:\path\to\env\Library\include\google/protobuf/descriptor.h(2245): warning C4100: 'location': unreferenced formal parameter
C:\path\to\env\Library\include\google/protobuf/descriptor.h(2245): warning C4100: 'descriptor': unreferenced formal parameter
C:\path\to\env\Library\include\google/protobuf/descriptor.h(2244): warning C4100: 'element_name': unreferenced formal parameter
C:\path\to\env\Library\include\google/protobuf/descriptor.h(2243): warning C4100: 'filename': unreferenced formal parameter
C:\path\to\env\Library\include\google/protobuf/map.h(689): warning C4244: 'initializing': conversion from 'uintptr_t' to 'google::protobuf::internal::map_index_t', possible loss of data
C:\path\to\env\Library\include\gazebo-11\gazebo/common/Dem.hh(121): warning C4373: 'gazebo::common::Dem::FillHeightMap': overriding virtual function 'gazebo::common::HeightmapData::FillHeightMap', earlier versions of the compiler did not override when parameters only differed by const/volatile qualifiers
C:\path\to\env\Library\include\gazebo-11\gazebo/common/HeightmapData.hh(55): note: see declaration of 'gazebo::common::HeightmapData::FillHeightMap'

C:\path\to\env\Library\include\boost/mpl/has_xxx.hpp(128): error C2039: 'is_pod': is not a member of 'std'
C:\path\to\env\Library\include\boost/mpl/has_xxx.hpp(128): error C2146: syntax error: missing ';' before identifier 'value'
C:\path\to\env\Library\include\boost/mpl/has_xxx.hpp(128): error C2065: 'value': undeclared identifier
C:\path\to\env\Library\include\boost/mpl/has_xxx.hpp(128): error C2975: 'boost::mpl::aux::has_xxx_impl': invalid template argument 'T'
C:\path\to\env\Library\include\boost/mpl/has_xxx.hpp(108): note: see declaration of 'boost::mpl::aux::has_xxx_impl'
C:\path\to\env\Library\include\boost/mpl/has_xxx.hpp(128): error C2039: 'is_empty': is not a member of 'std'
C:\path\to\env\Library\include\boost/mpl/has_xxx.hpp(128): error C2146: syntax error: missing ';' before identifier 'value'
C:\path\to\env\Library\include\boost/mpl/has_xxx.hpp(128): error C2065: 'value': undeclared identifier
C:\path\to\env\Library\include\boost/mpl/has_xxx.hpp(128): error C2975: 'boost::mpl::aux::has_xxx_impl': invalid template argument 'T'
C:\path\to\env\Library\include\boost/mpl/has_xxx.hpp(108): note: see declaration of 'boost::mpl::aux::has_xxx_impl'
...
```

**Environment**:
- **OS**: Windows 11 Pro
- **ROS Version**: Humble
- **Gazebo Version**: 11.14.0
- **Compiler**: MSVC

I have already installed all necessary dependencies via `vcpkg` and configured the environment variables. Please let me know if additional details are required.

**Steps to Reproduce**:
1. Clone `gazebo_ros2_control` repository.
2. Run `colcon build` in the workspace on Windows.

**Expected Behavior**:
The build should complete successfully without errors.

**Actual Behavior**:
The build fails with the above-mentioned errors.

christophfroehlich, 2024-09-24:
The error messages come already from ros2_control and dependencies like rclcpp_lifecycle. We do not officially support ros2_control compilation on Windows (we don't have CI for that etc..). Were your able to build the ros2_control stack (without gazebo)?

SENAI-GilmarCorreia, 2024-09-24:
Yes, I was able to build ros2_control, ros2_controllers, and gazebo_ros_pkgs. However, some packages may be broken on Windows, such as pid_controller.

christophfroehlich, 2024-09-24:
But there is a syntax error reported in hardware_interface, how can you compile that in ros2_control? 
Is gazebo classic running on Windows at all? Just to avoid running into a dead end with this package here.

SENAI-GilmarCorreia, 2024-09-24:
@christophfroehlich I used the following command:

```
colcon build --symlink install --packages-select hardware_interface --cmake-args -DCMAKE_PREFIX_PATH="C:/boost" -DBOOST_ROOT="C:/boost" -DBoost_NO_SYSTEM_PATHS=ON -DBUILD_TESTING=OFF -DCMAKE_EXPORT_COMPILE_COMMANDS=ON -DCMAKE_WINDOWS_EXPORT_ALL_SYMBOLS=ON -DPython3_EXECUTABLE="C:/Python38/python.exe"
```

And Gazebo is working like a charm (with a good fps). However, two plugins, `gazebo_init` and `gazebo_ray_sensor` from gazebo_ros_pkgs, are not working.

[image]

christophfroehlich, 2024-09-24:
Why you should get syntax errors in the file included in gazebo_ros2_control, if it works if compiled separately? Sorry, I can't help you further here, never used ROS on Windows.
fyi: You will need the gazebo_ros_init plugin to have the sim_time in the ROS world

SENAI-GilmarCorreia, 2024-09-24:
I don't know either 🤣. No problem, thank you for your support. If I find any solution to solve this installation on Windows, I'll share it with you all.

SENAI-GilmarCorreia, 2024-09-25:
Just an update... I was able to minimize some errors using this code in the CMakeLists file:

```cmake
if(WIN32)
  add_compile_definitions(
    # For math constants
    _USE_MATH_DEFINES
    # Minimize Windows namespace collision
    NOMINMAX
    WIN32_LEAN_AND_MEAN
  )
endif()
```

I have tried to split the build and see which errors are related to each plugin. The log from `gazebo_ros2_control` shows the following:

```
1>Checking Build System
  Building Custom Rule C:/path/to/gazebo_ros2_control/CMakeLists.txt
  gazebo_ros2_control_plugin.cpp
C:/path/to/gazebo_ros/include/gazebo_ros/node.hpp(44,1): warning C4275: no DLL interface class 'rclcpp::Node' used as the base for DLL interface class 'gazebo_ros::Node' [C:/path/to/gazebo_ros2_control.vcxproj]
C:/opt/ros/humble/include/rclcpp/rclcpp/node.hpp(77): message : see declaration of 'rclcpp::Node' [C:/path/to/gazebo_ros2_control.vcxproj]
C:/path/to/gazebo_ros/include/gazebo_ros/node.hpp(43): message : see declaration of 'gazebo_ros::Node' [C:/path/to/gazebo_ros2_control.vcxproj]
C:/path/to/controller_manager_msgs/controller_manager_msgs/srv/detail/switch_controller__struct.hpp(143,28): error C2059: syntax error: 'constant' [C:/path/to/gazebo_ros2_control.vcxproj]
C:/path/to/controller_manager_msgs/controller_manager_msgs/srv/detail/switch_controller__struct.hpp(216): message : see reference to instantiation of 'controller_manager_msgs::srv::SwitchController_Request_<ContainerAllocator>' being compiled [C:/path/to/gazebo_ros2_control.vcxproj]
C:/path/to/controller_manager_msgs/controller_manager_msgs/srv/detail/switch_controller__struct.hpp(144,1): error C2238: unexpected token(s) preceding ';' [C:/path/to/gazebo_ros2_control.vcxproj]
C:/path/to/controller_manager_msgs/controller_manager_msgs/srv/detail/switch_controller__struct.hpp(231,66): error C2589: 'constant': invalid token on right side of '::' [C:/path/to/gazebo_ros2_control.vcxproj]
C:/path/to/controller_manager_msgs/controller_manager_msgs/srv/detail/switch_controller__struct.hpp(231): error C2062: unexpected type 'unknown-type' [C:/path/to/gazebo_ros2_control.vcxproj]
C:/path/to/controller_manager_msgs/controller_manager_msgs/srv/detail/switch_controller__builder.hpp(28,3): error C2059: syntax error: ':' [C:/path/to/gazebo_ros2_control.vcxproj]
C:/path/to/controller_manager_msgs/controller_manager_msgs/srv/detail/switch_controller__builder.hpp(44,3): error C2059: syntax error: ':' [C:/path/to/gazebo_ros2_control.vcxproj]
C:/path/to/controller_manager_msgs/controller_manager_msgs/srv/detail/switch_controller__builder.hpp(60,3): error C2059: syntax error: ':' [C:/path/to/gazebo_ros2_control.vcxproj]
C:/path/to/controller_manager_msgs/controller_manager_msgs/srv/detail/switch_controller__builder.hpp(76,3): error C2059: syntax error: ':' [C:/path/to/gazebo_ros2_control.vcxproj]
C:/path/to/controller_manager_msgs/controller_manager_msgs/srv/detail/switch_controller__builder.hpp(92,3): error C2059: syntax error: ':' [C:/path/to/gazebo_ros2_control.vcxproj]
C:/path/to/controller_manager_msgs/controller_manager_msgs/srv/detail/switch_controller__builder.hpp(108,3): error C2059: syntax error: ':' [C:/path/to/gazebo_ros2_control.vcxproj]
C:/path/to/controller_manager_msgs/controller_manager_msgs/srv/detail/switch_controller__builder.hpp(124,3): error C2059: syntax error: ':' [C:/path/to/gazebo_ros2_control.vcxproj]
C:/path/to/controller_manager_msgs/controller_manager_msgs/srv/detail/switch_controller__builder.hpp(140,3): error C2059: syntax error: ':' [C:/path/to/gazebo_ros2_control.vcxproj]
C:/vcpkg/installed/x64-windows/include/google/protobuf/map.h(689,19): warning C4244: 'initializing': conversion from 'uintptr_t' to 'google::protobuf::internal::map_index_t', possible loss of data [C:/path/to/gazebo_ros2_control.vcxproj]
```

And the `gazebo_hardware_plugins` shows these:

```
Building Custom Rule C:/path/to/gazebo_ros2_control/CMakeLists.txt
  gazebo_system.cpp
C:/path/to/angles/include/angles/angles.h(38,1): warning C4005: '_USE_MATH_DEFINES': macro redefinition [C:/path/to/gazebo_hardware_plugins.vcxproj]
C:/path/to/gazebo_system.cpp : message : see previous definition of '_USE_MATH_DEFINES' [C:/path/to/gazebo_hardware_plugins.vcxproj]
C:/path/to/anaconda/envs/gazebo/Library/include/google/protobuf/map.h(689,19): warning C4244: 'initializing': conversion from 'uintptr_t' to 'google::protobuf::internal::map_index_t', possible loss of data [C:/path/to/gazebo_hardware_plugins.vcxproj]
C:/path/to/hardware_interface/types/hardware_interface_return_values.hpp(25,3): error C2143: syntax error: missing '}' before 'constant' [C:/path/to/gazebo_hardware_plugins.vcxproj]
C:/path/to/hardware_interface/types/hardware_interface_return_values.hpp(25,3): error C2059: syntax error: 'constant'
...
```

christophfroehlich, 2024-09-25:
`controller_manager_msgs` are genereated by rosidl, this code is not part of our repos. There must be something different in the cmake file of gazebo_ros2_control compared to ros2_control to make your build fail.

traversaro, 2025-01-10:
I can't remember how I come up with such patch (perhaps I copied it from somewhere else) but on robostack to build gazebo_ros2_control on Windows we used this patch: https://github.com/RoboStack/ros-humble/commit/d7d25b77ca81235d8e3d68d7b7542f43bc4650c0 .