ros-controls/gazebo_ros2_control #383: Can I use like MIT control method in gazebo ros2 control?

lkx8421, 2024-10-09:
In source code I found control methods are   {
    NONE      = 0,
    POSITION  = (1 << 0),
    VELOCITY  = (1 << 1),
    EFFORT    = (1 << 2),
    VELOCITY_PID    = (1 << 3),
    POSITION_PID    = (1 << 4),
  };
How can I use control method like with kp kd postion velocity and effort

christophfroehlich, 2024-10-09:
Please have a look at the [documentation](https://control.ros.org/rolling/doc/gazebo_ros2_control/doc/index.html#using-pid-control-joints)

closed 2024-10-09 by christophfroehlich

lkx8421, 2024-10-21:
But I cannot use position, velocity, effort, kp and kd simultaneous

christophfroehlich, 2024-10-25:
I don't know what the _MIT control method_ is. If you want to apply any learning method - based control strategy (why would one use that in this simulator?): Write your [own plugin](https://control.ros.org/rolling/doc/gazebo_ros2_control/doc/index.html#advanced-custom-gazebo-ros2-control-simulation-plugins).