ros-controls/gazebo_ros2_control #384: Are there support for transmission interfaces?

caioaamaral, 2024-10-17:
**ros2_control** has this concept of [transmissions interfaces](https://github.com/ros-controls/ros2_control/tree/master/transmission_interface), where simple_transmission for instance applies a transmission relation of kind `transmission: joint -> actuator`

**ros2_control_demos** have [an example](https://github.com/ros-controls/ros2_control_demos/blob/master/example_8/hardware/rrbot_transmissions_system_position_only.cpp#L269) of that, where on write operation it computes the corresponding command for an actuator considering its transmission relation.

christophfroehlich, 2024-10-17:
No, those are not integrated in the gazebo_ros2_control plugin.
The transmission interfaces have to be manually included in the hardware components, there is no mechanism to automatically load them at run-time.

caioaamaral, 2024-10-17:
> No, those are not integrated in the gazebo_ros2_control plugin. The transmission interfaces have to be manually included in the hardware components, there is no mechanism to automatically load them at run-time.

could you clarify a bit further? Is there a way to manually include then within gazebo_ros2_control/GazeboSystem, or it would require a different `hardware_interface::SystemInterface`?

christophfroehlich, 2024-10-17:
you have already linked the example_8 from the demos: This is the way to include that in a hardware component.

closed 2024-10-17 by caioaamaral