ros-controls/gazebo_ros2_control #394: Error initializing URDF to resource manager

tomaszkapron, 2024-12-17:
After the change of [`ros2_control`](https://github.com/ros-controls/ros2_control) from version 2.43.1 to 2.44 `gazebo_ros2_control` run into the following error during initialization:

```
[gzserver-1] ?[31m[ERROR 1734368594.176366640] [gazebo_ros2_control]: Error initializing URDF to resource manager! (Load() at ./src/gazebo_ros2_control_plugin.cpp:302)
```
`gazebo_ros2_control` pacakge is on version 4.10, ROS 2 version - humble. The error does not occur with 2.43 version of ros2_control.

I was able to reproduce the error using `gazebo_ros2_control_demos` package using:
```
ros2 launch gazebo_ros2_control_demos cart_example_effort.launch.py
```

christophfroehlich, 2024-12-17:
I built the humble stack now from source, and was able to launch the demo successfully. Same for humble-main binaries. Have you upgraded all upstream ROS packages as well, i.e. `sudo apt-get upgrade`?
```
 $ apt list ros-humble-ros2-control ros-humble-gazebo-ros2-control ros-humble-tinyxml2-vendor ros-humble-rclcpp ros-humble-gazebo-ros ros-humble-gazebo-dev
Listing... Done
ros-humble-gazebo-dev/jammy,now 3.7.0-1jammy.20241125.233927 amd64 [installed,automatic]
ros-humble-gazebo-ros2-control/jammy,now 0.4.10-1jammy.20241128.053639 amd64 [installed]
ros-humble-gazebo-ros/jammy,now 3.7.0-1jammy.20241128.043020 amd64 [installed]
ros-humble-rclcpp/jammy,now 16.0.11-1jammy.20241128.012309 amd64 [installed,automatic]
ros-humble-ros2-control/jammy,now 2.44.0-1jammy.20241128.034910 amd64 [installed]
ros-humble-tinyxml2-vendor/jammy,now 0.7.6-1jammy.20241125.234735 amd64 [installed,automatic]
```

tomaszkapron, 2024-12-20:
I mean it launches successfully and works correctly. But the error still exists there. I have the packages upgraded to same version as pointed in your comment. Can you just confirm you were able to launch demo without the error?

christophfroehlich, 2024-12-21:
ha, you are right :see_no_evil: I'll have a look

christophfroehlich, 2024-12-21:
https://github.com/ros-controls/ros2_control/pull/1960 fixes the error "accidentially": The resource manager does not throw now.

closed 2024-12-21 by christophfroehlich

shr-eyas, 2024-12-30:
How exactly did you fix this? @tomaszkapron 
Thanks.

tomaszkapron, 2024-12-31:
You should probably build `ros2_control` from source until it is released to official apt repositories.

christophfroehlich, 2025-01-03:
It is released already and available on [humble-testing](https://docs.ros.org/en/humble/Installation/Testing.html) distro. We don't know when the next sync will take place.