ros-controls/gazebo_ros2_control #401: Inconsistent joint_states in Gazebo simulation

danielcostanzi18, 2025-01-22:
Hello team!

**Issue Description**
When starting the Gazebo simulation with a robot controlled in position mode, the joint_states topic reports zero effort, while the positions and velocities are consistent. This behavior is observed in the first part of the plot shown in the attached screenshot.

At 12 seconds, switching to an effort-based controller causes the joint_states topic to report non-zero effort values, as expected.

However, at 22 seconds, switching back to the position controller results in an abrupt state change without any transition. Additionally, the effort values remain clamped to the last reported values from the effort-based controller.

**Screenshots**
[image]

**Environment:**
 - OS: Ubuntu 22.04
 - Version: humble
 - I'm running the simulation inside a docker container

christophfroehlich, 2025-01-22:
Afaik: If you use a velocity command interface with gazebo_ros2_control, then the velocity of the physics engine is "overwritten" and no effort is calculated, hence the state interface given back reads zero.

We use `GetForce` to read the effort from the physics engine
https://github.com/ros-controls/gazebo_ros2_control/blob/6a9791ba2eb6da50911eb033d76e900bf0dcb12b/gazebo_ros2_control/src/gazebo_system.cpp#L769
which is "not implemented" acc. do its docstring
https://github.com/gazebosim/gazebo-classic/blob/gazebo11/gazebo/physics/Joint.hh#L292-L322
So I think this is now just undefined behavior.

fyi: gazebo classic reached end of life this January, so don't expect much work to be put in here. Consider updating to ign_ros2_control and gazebo harmonic.

saikishor, 2025-01-22:
@christophfroehlich is it the same for the position command interface?

christophfroehlich, 2025-01-22:
even worse, then everything is broken see https://github.com/ros-controls/gazebo_ros2_control/issues/240

danielcostanzi18, 2025-01-23:
@christophfroehlich thanks a lot for the explanation! I'll close the issue

closed 2025-01-23 by danielcostanzi18